aff3ct/aff3ct #1: segmentation fault when running on armv6

Nixes, 2017-01-17:
Cross compiled (and statically linked) with
```
cmake .. -G"Unix Makefiles" -DCMAKE_CXX_COMPILER=arm-linux-gnueabi-g++ -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=" -Wall -funroll-loops -march=armv6 -static-libgcc -static-libstdc++ -DMULTI_PREC" -DCMAKE_CXX_FLAGS_RELEASE="-Ofast"
```

Output from a Raspberry pi A+
```
./aff3ct --sim-type BFER --cde-type POLAR -K 1024 -N 2048 -m 1.5 -M 2.81 --cde-fb-gen-method GA -t 1

# -------------------------------------------------
# ---- A FAST FORWARD ERROR CORRECTION TOOL >> ----
# -------------------------------------------------
# Parameters:
# * Simulation ------------------------------------
#    ** Type                    = BFER
#    ** SNR min (m)             = -0.432892 dB
#    ** SNR max (M)             = -0.432892 dB
#    ** SNR step (s)            = 0.100000 dB
#    ** Type of bits            = int (32 bits)
#    ** Type of reals           = float (32 bits)
#    ** Inter frame level       = 1
#    ** Seed                    = 0
#    ** Multi-threading (t)     = 1 thread(s)
# * Code ------------------------------------------
#    ** Type                    = POLAR
#    ** Info. bits (K)          = 1024
#    ** Codeword size (N)       = 2048
#    ** Coset approach (c)      = off
#    ** Sigma for code gen.     = adaptative
#    ** Frozen bits gen. method = GA
# * Source ----------------------------------------
#    ** Type                    = RAND
# * Encoder ---------------------------------------
#    ** Type                    = POLAR
#    ** Systematic encoding     = on
# * Modulator -------------------------------------
#    ** Type                    = BPSK
#    ** Bits per symbol         = 1
#    ** Sampling factor         = 1
# * Channel ---------------------------------------
#    ** Type                    = AWGN
#    ** Domain                  = LLR
# * Demodulator -----------------------------------
#    ** Sigma square            = on
#    ** Max type                = unused
# * Decoder ---------------------------------------
#    ** Type (D)                = SC
#    ** Implementation          = FAST
# * Monitor ---------------------------------------
#    ** Frame error count (e)   = 100
#
# The simulation is running...
Segmentation fault
```

OlivierHartmann, 2017-01-17:
Hello Nixes!
Thanks for using Aff3ct.
We never tried to run aff3ct on an armV6.
Can you run with
`cmake .. -G"Unix Makefiles" -DCMAKE_CXX_COMPILER=arm-linux-gnueabi-g++ -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS=" -Wall -funroll-loops -march=armv6 -static-libgcc -static-libstdc++ -DMULTI_PREC"`

And then can you show us the result of a run under gdb or valgrind?

Thank you

Nixes, 2017-01-17:
valgrind output, but it looks like valgrind might be affected by a bug on RPI so I'll try with GDB next
```
valgrind ./aff3ct --sim-type BFER --cde-type POLAR -K 1024 -N 2048 -m 1.5 -M 2.81 --cde-fb-gen-method GA -t 1
==1116== Memcheck, a memory error detector
==1116== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
==1116== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==1116== Command: ./aff3ct --sim-type BFER --cde-type POLAR -K 1024 -N 2048 -m 1.5 -M 2.81 --cde-fb-gen-method GA -t 1
==1116==
disInstr(arm): unhandled instruction: 0xF4A174DD
                 cond=15(0xF) 27:20=74(0x4A) 4:4=1 3:0=13(0xD)
==1116== valgrind: Unrecognised instruction at address 0x4844f2c.
==1116==    at 0x4844F2C: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem.so)
==1116== Your program just tried to execute an instruction that Valgrind
==1116== did not recognise.  There are two possible reasons for this.
==1116== 1. Your program has a bug and erroneously jumped to a non-code
==1116==    location.  If you are running Memcheck and you just saw a
==1116==    warning about a bad jump, it's probably your program's fault.
==1116== 2. The instruction is legitimate but Valgrind doesn't handle it,
==1116==    i.e. it's Valgrind's fault.  If you think this is the case or
==1116==    you are not sure, please let us know and we'll try to fix it.
==1116== Either way, Valgrind will now raise a SIGILL signal which will
==1116== probably kill your program.
==1116==
==1116== Process terminating with default action of signal 4 (SIGILL)
==1116==  Illegal opcode at address 0x4844F2C
==1116==    at 0x4844F2C: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem.so)
==1116==
==1116== HEAP SUMMARY:
==1116==     in use at exit: 0 bytes in 0 blocks
==1116==   total heap usage: 0 allocs, 0 frees, 0 bytes allocated
==1116==
==1116== All heap blocks were freed -- no leaks are possible
==1116==
==1116== For counts of detected and suppressed errors, rerun with: -v
==1116== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Illegal instruction
```

Nixes, 2017-01-17:
gdb output looks more fruitful
```
gdb aff3ct
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from aff3ct...done.
(gdb) run --sim-type BFER --cde-type POLAR -K 1024 -N 2048 -m 1.5 -M 2.81 --cde-fb-gen-method GA -t 1
Starting program: /home/pi/aff3ct_debug/bin/aff3ct --sim-type BFER --cde-type POLAR -K 1024 -N 2048 -m 1.5 -M 2.81 --cde-fb-gen-method GA -t 1
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
# -------------------------------------------------
# ---- A FAST FORWARD ERROR CORRECTION TOOL >> ----
# -------------------------------------------------
# Parameters:
# * Simulation ------------------------------------
#    ** Type                    = BFER
#    ** SNR min (m)             = -0.499823 dB
#    ** SNR max (M)             = -0.499823 dB
#    ** SNR step (s)            = 0.100000 dB
#    ** Type of bits            = int (32 bits)
#    ** Type of reals           = float (32 bits)
#    ** Inter frame level       = 1
#    ** Seed                    = 0
#    ** Multi-threading (t)     = 1 thread(s)
# * Code ------------------------------------------
#    ** Type                    = POLAR
#    ** Info. bits (K)          = 1024
#    ** Codeword size (N)       = 2048
#    ** Coset approach (c)      = off
#    ** Sigma for code gen.     = adaptative
#    ** Frozen bits gen. method = GA
# * Source ----------------------------------------
#    ** Type                    = RAND
# * Encoder ---------------------------------------
#    ** Type                    = POLAR
#    ** Systematic encoding     = on
# * Modulator -------------------------------------
#    ** Type                    = BPSK
#    ** Bits per symbol         = 1
#    ** Sampling factor         = 1
# * Channel ---------------------------------------
#    ** Type                    = AWGN
#    ** Domain                  = LLR
# * Demodulator -----------------------------------
#    ** Sigma square            = on
#    ** Max type                = unused
# * Decoder ---------------------------------------
#    ** Type (D)                = SC
#    ** Implementation          = FAST
# * Monitor ---------------------------------------
#    ** Frame error count (e)   = 100
#
# The simulation is running...

Program received signal SIGSEGV, Segmentation fault.
0x006f28dc in Frozenbits_generator_GA<int>::evaluate (this=0xdc9400)
    at /media/sf_aff3ct/aff3ct/src/Tools/Code/Polar/Frozenbits_generator/Frozenbits_generator_GA.cpp:36
36      /media/sf_aff3ct/aff3ct/src/Tools/Code/Polar/Frozenbits_generator/Frozenbits_generator_GA.cpp: No such file or directory.
```

kouchy, 2017-01-17:
I think that you are compiling without the hardware floating point unit. Did you try these options: `-mfpu=vfp -mfloat-abi=hard` ?

You can also see the discussion here about floating point unit on RPi: http://raspberrypi.stackexchange.com/questions/545/does-the-raspberry-pi-have-hardware-floating-point-support

Nixes, 2017-01-18:
On trying a compile with the above options added I got linking errors as follows.
```
  /usr/lib/gcc-cross/arm-linux-gnueabi/5/../../../../arm-linux-gnueabi/bin/ld:
  error: CMakeFiles/cmTC_a53ee.dir/testCXXCompiler.cxx.o uses VFP register
  arguments, cmTC_a53ee does not

  /usr/lib/gcc-cross/arm-linux-gnueabi/5/../../../../arm-linux-gnueabi/bin/ld:
  failed to merge target specific data of file
  CMakeFiles/cmTC_a53ee.dir/testCXXCompiler.cxx.o

  collect2: error: ld returned 1 exit status

  CMakeFiles/cmTC_a53ee.dir/build.make:97: recipe for target 'cmTC_a53ee'
  failed

  make[1]: *** [cmTC_a53ee] Error 1

  make[1]: Leaving directory
  '/media/sf_aff3ct/aff3ct/build/CMakeFiles/CMakeTmp'

  Makefile:126: recipe for target 'cmTC_a53ee/fast' failed

  make: *** [cmTC_a53ee/fast] Error 2
```

Nixes, 2017-01-18:
I added the suggested options to my build options and then got linker errors in return.
```
  /usr/lib/gcc-cross/arm-linux-gnueabi/5/../../../../arm-linux-gnueabi/bin/ld:
  error: CMakeFiles/cmTC_759a5.dir/testCXXCompiler.cxx.o uses VFP register
  arguments, cmTC_759a5 does not

  /usr/lib/gcc-cross/arm-linux-gnueabi/5/../../../../arm-linux-gnueabi/bin/ld:
  failed to merge target specific data of file
  CMakeFiles/cmTC_759a5.dir/testCXXCompiler.cxx.o

  collect2: error: ld returned 1 exit status

  CMakeFiles/cmTC_759a5.dir/build.make:97: recipe for target 'cmTC_759a5'
  failed

  make[1]: *** [cmTC_759a5] Error 1

  make[1]: Leaving directory
  '/media/sf_aff3ct/aff3ct/build/CMakeFiles/CMakeTmp'

  Makefile:126: recipe for target 'cmTC_759a5/fast' failed

  make: *** [cmTC_759a5/fast] Error 2

```
This is after removing /build and starting fresh.

Nixes, 2017-01-18:
Switching toolchain to arm-linux-gnueabihf was required to prevent linker errors with hardfloat. But I haven't been able to get a compatible binary out of this new toolchain yet. I'll probably have another go at it on the weekend.

kouchy, 2017-01-25:
Any improvement ?

Nixes, 2017-03-03:
I ended up compiling on the pi even though it took quite a few hours. It worked fine this way, It was probably something to do with cross compilation options. Also of note is that you need more than 256mb and less than 512mb of memory to successfully compile it.

closed 2017-03-03 by Nixes

kouchy, 2017-03-03:
Great news. Feel free to ask if you have any questions.