aff3ct/aff3ct #2: Non systematic encoding

SabrAkhi, 2017-04-11:
Error when enabling non systematic encoding with "--enc-no-sys":

(EE) An issue was encountered when building the communication chain.
(EE) aff3ct::simulation::Simulation: something went wrong during the "Decoder<B,Q>" instantiation (tid = 0).
# End of the simulation.
[image]

bonben, 2017-04-11:
Hi @SabrAkhi 
The "FAST" version of the SC decoder is systematic. Therefore the encoder must also be systematic when using this version of the SC decoder.
If you want to generate results for the SC decoder with a systematic encoder, use `--dec-implem NAIVE` to enable the NAIVE (and possibly systematic) version of the SC decoder.

closed 2017-04-11 by SabrAkhi

GitLuc1, 2019-08-05:
Hi, thanks for the great work you do here!

Maybe you can help me with understanding something about your polar code implementation.

Apparently the polar codes used in 5G are non-systematic. aff3ct offers the SCL decoder for non-systematic decoding, thus the CRC check can not be used and the NAIVE implementation is available. Is there any workaround or are you planning on supporting non-systematic codes in 5G?

bonben, 2019-09-02:
Hi @GitLuc1.

Unfortunately, there is no plain and simple solution for your problem at the moment.
As you noted, the `FAST` version of the SCL decoder is only available for systematic encoding.

Nevertheless, the transformation from systematic to non-systematic polar codes is just a reencoding with a special pattern ([10.1109/LCOMM.2011.061611.110862](https://doi-org.docelec.u-bordeaux.fr/10.1109/LCOMM.2011.061611.110862)).

The solution is therefore to concatenate an encoder to existing decoders, either using a [my-project-with-aff3ct](https://github.com/aff3ct/my_project_with_aff3ct) or directly in the aff3ct code, creating an adequate FAST non-systematic SCL decoder.

It is not planned in the near future to integrate FAST non-systematic SCL polar codes, but a contribution in this area would be more than welcome, and we could provide support. 

Regards.