aff3ct/aff3ct #3: Can't run on Windows 10.

kumuds4, 2017-08-01:
Hello,
I did get message from Adrien. It is giving same error. I have MinGw64, g++, cygwin and MS visual C++. I also downloaded gcc and g++ but nothing works.  Right now, have Windows only. I may have eventually Linux .  I have Cmake and aff3ct in my users/document folder as suggested by Adrien. Still could not run. 
To me it seems that the command in ~/.bashrc what is there for gcc an C++ is not in right path. 
I really would like to run Polar codes in AWGN. 

Thank you!
-Kumud

kouchy, 2017-08-01:
Hi Kumud,
AFF3CT compiles on Windows XP, Vista, 7, 8 and 10 with MSVC (from Visual Studio 2015) and with g++.
On Windows, the easiest way to compile AFF3CT is to use GCC, you can follow the steps explained here : [https://github.com/aff3ct/aff3ct/wiki/Compile-AFF3CT-on-Windows](https://github.com/aff3ct/aff3ct/wiki/Compile-AFF3CT-on-Windows).
I advise you to try to compile first with g++ and after to try with Visual Studio. The difference between the two is in the project generation (when you call the _cmake_ command).

We can't help you more if you do not explain exactly what command did you used and what was the result of this command in your terminal.

I think AFF3CT is a good code to simulate Polar codes in AWGN, we have various decoding algorithms (SC, SCAN and SC-List) working at full speed.

kumuds4, 2017-08-02:
> aff3ct/wiki/Compile-AFF3CT-on-Windows.

Adrien,

The following last commands failed highlighted in nay blue. It also can't
find g++.exe build. Perhaps, I already downloaded MinGW64. I have put
MinGW64 and CMAKE in document folder now as your per instructions you wrote
on my query of issues during last week.  However command on GIT hub says
$ echo 'export PATH=$PATH:"C:\Program Files\CMake\bin"' >> ~/.bashrc

These two failed as well.

$ cmake .. -DCMAKE_CXX_COMPILER=g++.exe -DCMAKE_CC_COMPILER=gcc.exe
-DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-Wall -funroll-loops
-DMULTI_PREC -march=native" -DCMAKE_CXX_FLAGS_RELEASE="-Ofast"
$ mingw32-make -j2 (This seems like for Win32? Should it be -j4?)

I feel like CMAKE can't communicate with build command for g++ to get to gcc.

I wish, I could fix all that and run polar codes. You have done great
job. Hope you would be able to help me now I have followed exactly
what you wrote.

Best regards!

-Kumud

 <https://github.com/aff3ct/aff3ct/wiki/Compile-AFF3CT-on-Windows#bonus>

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-02:
> *cell:(804) 564-6670 <(804)%20564-6670> or (804) 564-6692
> <(804)%20564-6692>*
> *Ph./Fax:(804)658-0125 <(804)%20658-0125>*

Adrien,
Sorry to bother. I tried to run. It keeps saying when I try to configure:
CMake Error: The source directory
"C:/Users/kumud/Documents/CMake/share/cmake-3.7/Modules" does not appear to
contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
I did get to g++ and C++ compiler.
C:/Users/kumud/Documents/minGW64/bin/g++.exe
C:/Users/kumud/Documents/CMake/bin
Thank you for help!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kouchy, 2017-08-02:
Hi @kumuds4,
It seems that you are not running cmake command in the right folder. Did you try to execute the command from the build folder ?

kumuds4, 2017-08-03:
I have downloaded latest 3.9.0 CMake.  Still getting following error.

CMake Error in CMakeLists.txt:
Failed to run MSBuild command:

MSBuild.exe

to get the value of VCTargetsPath:

CMake Warning (dev) in CMakeLists.txt:
No cmake_minimum_required command is present. A line of code such as

cmake_minimum_required(VERSION 3.9)

should be added at the top of the file. The version specified may be lower
if you wish to support older CMake versions for this project. For more
information run "cmake --help-policy CMP0000".
This warning is for project developers. Use -Wno-dev to suppress it.

Configuring incomplete, errors occurred!

See also "C:/Users/kumud/Documents/CMake/bin/CMakeFiles/CMakeOutput.log".

Note: I have administrative privileged and have CMake in Program files.
AFF3CT is in user/Document as you suggested.

I see in this latest version there is CMakeLists file. I think, am near to
compile but still unable to do so.

Thank you for help!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-08-03:
Hi Kumuds4 !

What terminal are you using ? Git bash ?

Can you please confirm you are compiling Aff3ct in this directory : "aff3ct/path/build" that should be same than  "~/aff3ct/build". If not please go there with : 
$ cd aff3ct/path/build
and run cmake again : 
$ cmake .. -G"MinGW Makefiles"

How did you installed CMake ? Directly in the Documents folder (as you said in your first and second messages) ? Or in the Program files folder (as you said in your last message) ? Or in the Program files folder and then you copied it in the Documents folder ?
If you have done the last with a copy, please delete this CMake folder in the Documents folder. If you have moved the folder, please delete it and reinstall CMake.
The directory does not matter. What's important is to give to your terminal the good path to find CMake.

Then if you have installed CMake in the directory C:/Users/kumud/Documents/CMake/ , please run 
$ echo 'export PATH=$PATH:"C:\Users\kumud\Documents\CMake\bin"' >> ~/.bashrc

Make sure compiling aff3ct in the right folder, as explained above and tell us the result.

Thank you,

Olivier

kumuds4, 2017-08-03:
Hi,
I put all on issues. However, at this point, am very frustrated. I am going
back to Matlab. At least, it works there. I have wasted several days, have
not slept or barely. I am unable to use, no matter what I do. I am not sure
what else I need to do.
Thank you for help.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kouchy, 2017-08-04:
Hello @kumuds4,

Please notice that we are helping you on our free time too (AFF3CT is Open-source it does not mean that we have to make the support of it).
We understand your frustration it is clear that you tried hard to compile AFF3CT.
We are sorry you did not succeed. We can assure you that we are able to compile AFF3CT on Windows on our side (and other users succeed too). I will recommend you to restart the procedure from the beginning following carefully all the instructions.

I don't know what you wanted to do with AFF3CT, but if you were looking for reference curves you can take a look on our ref database (https://github.com/aff3ct/error_rate_references). And if you want to display these refs you can install PyBER (https://github.com/aff3ct/PyBER) our tool dedicated to the visualization of BER/FER performances.

Best regards.

kouchy, 2017-08-07:
Hi @kumuds4,

I have good news for you. With the new Visual Studio 2017 it's very easy to open a CMake project.
You can see this post for more info: https://blogs.msdn.microsoft.com/vcblog/2016/11/16/cmake-support-in-visual-studio-the-visual-studio-2017-rc-update/

We had some compilation errors with the new version of the C++ compiler (MSVC 1910) so you have to pull or re-clone the last version of AFF3CT from Github.

Hoping it will help.

kumuds4, 2017-08-07:
> com/vcblog/2016/11/16/cmake-support-in-visual-studio-the-
> visual-studio-2017-rc-update/

Hi Adrien,

Thank you so much for updates. I have been grading for my summer classes
and doing some prepping for fall about to start. Now, that I submitted
summer grades, got more time. I went over joint papers of yours in the
documentation. I  uninstalled as you mentioned in the last message. Then,
re-downloaded CMake 3.9.0, Cygwin. I still need to download MINGW64. I went
to the blog, have Visual Studio 2017 on this laptop as well as on the
desktop. In few minutes, am going for AFF3T at GIT. By the way, I mentioned
to Professor S.G. Wilson, the University of Virginia from whom I have
learned a lot of Electrical and Computer Engineering (ECE). I have Math &
Physics background but wanted a change towards ECE. Thus polar coding. I
know LDPC, basics of Turbo. I did send the plot to Professor  Wilson that
you have on the web. He liked it. Now, he has asked me to find out
block-length for the plots that you have on the web. I am sending him the
whole report. Once it starts working, I will do plots, will send at least
one to you. I also have downloaded pyber.
Thank you again for your kindness to help!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D. (Hungarian Academy of Sciences, Budapest,
Hungary)*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-07:
Hi Adrien,
I got Visual MS Visual Studio 2017 running and am able to create a project
file. I was trying to run the Polar codes. I found .sln and .sl2 files for
project. I cloned from GitHub as you wrote the latest aff3ct but can't find
anymore. I think, I overwrote on it. Either I need to create out of
scratch. You have Linux/Unix command for polar codes, Turbo etc. Do you
have one sample for Windows may be for polar codes, would be good to run
yours first to practice?  Then I build form scratch. I may eventually
install back Linux in coming weeks. Right now it is all Windows 10 on
desktop and laptop. Please take your time and send me a sample polar code
project.
Thank you for help! Finally, I have MS Visual C++ running.
-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kouchy, 2017-08-08:
Hi @kumuds4,

I'm not sur if you are doing well. Let me remind you some elementary steps to compile AFF3CT.
1. clone AFF3CT on your laptop, open a terminal and type : `git clone https://github.com/aff3ct/aff3ct.git`
2. go to the AFF3CT directory and clone the sub-modules : `cd aff3ct && git submodule update --init --recursive`
3. open Visual and select the AFF3CT folder (`File > Open > Folder`)
4. build AFF3CT (play button in green)

If you want some examples to launch AFF3CT, you can use the repo I gave you last time (with ref. curves):
- Polar codes:  https://github.com/aff3ct/error_rate_references/blob/master/BFER/Polar/SC/polar_2048_1723_1.0_4.5_32_SC.txt
- Turbo codes: https://github.com/aff3ct/error_rate_references/blob/master/BFER/Turbo/LTE/FNC/LTE_2048_4096_ref.txt
- LDPC codes: https://github.com/aff3ct/error_rate_references/blob/master/BFER/LDPC/BP_LAYERED/SPA/CCSDS_SPA_64_128.txt

If you take a look into these files you will see that the run command is given at the beginning (`./bin/aff3ct ...`).

kumuds4, 2017-08-08:
>    master/BFER/Polar/SC/polar_2048_1723_1.0_4.5_32_SC.txt
>    master/BFER/Turbo/LTE/FNC/LTE_2048_4096_ref.txt
>    master/BFER/LDPC/BP_LAYERED/SPA/CCSDS_SPA_64_128.txt

Adrien,

Thanks so much! I was doing the 1st step I did right. From 2nd onwards it
was more clear in Linux than in Windows. I am planning to put back Linux by
this weekend.
Once plotted, will send you.

Best,

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-08:
Adrien,

It is cloning to this directory, so getting error. I need to put all this
in C:/Users/Kumud/documents I think. It says this Appdata is no a working
directory:
-- Build files have been written to:
C:/Users/kumud/AppData/Local/CMakeBuild/6cc5adb8-cc93-0134-82ab-9507b87758c7/build/x86-Debug

Command line: C:\PROGRAM FILES (X86)\MICROSOFT VISUAL
STUDIO\2017\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe
-G "Visual Studio 15 2017"
-DCMAKE_INSTALL_PREFIX:PATH="C:\Users\kumud\AppData\Local\CMakeBuild\6cc5adb8-cc93-0134-82ab-9507b87758c7\build\install"
-DCMAKE_CONFIGURATION_TYPES="Debug" "C:\Users\kumud\aff3ct"
1> Working directory:
C:\Users\kumud\AppData\Local\CMakeBuild\6cc5adb8-cc93-0134-82ab-9507b87758c7\build\x86-Debug
1> CMake Warning (dev) at C:\Users\kumud\aff3ct\CMakeLists.txt:10 (if):
1>   Policy CMP0054 is not set: Only interpret if() arguments as variables
or
1>   keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
1>   details.  Use the cmake_policy command to set the policy and suppress
this
1>   warning.
1>
1>   Quoted variables like "MSVC" will no longer be dereferenced when the
policy
1>   is set to NEW.  Since the policy is not set the OLD behavior will be
used.
1> This warning is for project developers.  Use -Wno-dev to suppress it.
1>
1> CMake Warning (dev) at C:\Users\kumud\aff3ct\CMakeLists.txt:67 (if):
1>   Policy CMP0054 is not set: Only interpret if() arguments as variables
or
1>   keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
1>   details.  Use the cmake_policy command to set the policy and suppress
this
1>   warning.
1>
1>   Quoted variables like "MSVC" will no longer be dereferenced when the
policy
1>   is set to NEW.  Since the policy is not set the OLD behavior will be
used.
1> This warning is for project developers.  Use -Wno-dev to suppress it.
1>
1> -- MIPP found
1> CMake Warning (dev) at C:\Users\kumud\aff3ct\CMakeLists.txt:208 (if):
1>   Policy CMP0054 is not set: Only interpret if() arguments as variables
or
1>   keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
1>   details.  Use the cmake_policy command to set the policy and suppress
this
1>   warning.
1>
1>   Quoted variables like "MSVC" will no longer be dereferenced when the
policy
1>   is set to NEW.  Since the policy is not set the OLD behavior will be
used.
1> This warning is for project developers.  Use -Wno-dev to suppress it.
1>
1> CMake Warning (dev) at C:\Users\kumud\aff3ct\CMakeLists.txt:211 (elseif):
1>   Policy CMP0054 is not set: Only interpret if() arguments as variables
or
1>   keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
1>   details.  Use the cmake_policy command to set the policy and suppress
this
1>   warning.
1>
1>   Quoted variables like "MSVC" will no longer be dereferenced when the
policy
1>   is set to NEW.  Since the policy is not set the OLD behavior will be
used.
1> This warning is for project developers.  Use -Wno-dev to suppress it.
1>
1> CMake Warning (dev) at C:\Users\kumud\aff3ct\CMakeLists.txt:214 (elseif):
1>   Policy CMP0054 is not set: Only interpret if() arguments as variables
or
1>   keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
1>   details.  Use the cmake_policy command to set the policy and suppress
this
1>   warning.
1>
1>   Quoted variables like "MSVC" will no longer be dereferenced when the
policy
1>   is set to NEW.  Since the policy is not set the OLD behavior will be
used.
1> This warning is for project developers.  Use -Wno-dev to suppress it.
1>
1> -- Configuring done
1> -- Generating done
1> -- Build files have been written to:
C:/Users/kumud/AppData/Local/CMakeBuild/6cc5adb8-cc93-0134-82ab-9507b87758c7/build/x86-Debug
1> Starting CMake target info extraction ...
1> CMake server connection made.
1> Extracted includes paths.
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted global settings.
1> Extracted code model.
1> Collating data ...
1> Target info extraction done.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kouchy, 2017-08-08:
Hello @kumuds4,

There are warnings but no errors here.
Btw, i fixed these warnings yesterday so please get the last version of AFF3CT from Github.

After that you are all good, you can launch the compilation.

kumuds4, 2017-08-08:
Thanks Adrien!

I will redo in few minutes.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-08:
Adrien,
If you take a look into these files you will see that the run command is
given at the beginning (./bin/aff3ct ...).
I tried to run at GIT bash command but says no file to run.
Then I tried in MS studio since it has C++ now with all Cmakelists etc. It
should run there. Maybe I am doing something wrong? Not only that it starts
building AppData folder and aff3ct does not like but runs with error. I
created another folder to build. However, when it runs in the editor, goes
back to AppData folder.
Or should I forget the windows, install Linux, then run on the shell? I
have all latest aff3ct, and this one as well. : cd aff3ct && git submodule
update --init --recursive
what you suggested in the email. Still can't get to run the command that I
really want to see simulation results and curves especially for polar
codes.
I also have PyBer plot installed now.
Thanks for help!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-08:
It says: After I re-downloaded. everything and started running in Visual
C+.
'aff3ct.exe' (Win32): Loaded 'C:\Windows\SysWOW64\bcryptprimitives.dll'.
Cannot find or open the PDB file.
'aff3ct.exe' (Win32): Loaded 'C:\Windows\SysWOW64\sechost.dll'. Cannot find
or open the PDB file.
The thread 0x153c has exited with code 1 (0x1).
The thread 0x3bf8 has exited with code 1 (0x1).
The thread 0x3d44 has exited with code 1 (0x1).
The program '[14172] aff3ct.exe' has exited with code 1 (0x1).
This should be small problem. I will check tonight or tomorrow on system.
 got to teach a class. Thanks fro all help! I think I will get it done with
your help. if I simulate good, ask the professor to say the acknowledgement
in the research report/or paper.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kouchy, 2017-08-08:
It seems to work, you compiled AFF3CT. Now you just have to find the `aff3ct.exe` on your partition and you can launch it from Git bash for example. According to your previous post, AFF3CT binary should be located somewhere like this: `C:/Users/kumud/AppData/Local/CMakeBuild/6cc5adb8-cc93-0134-82ab-9507b87758c7/build/`.

kumuds4, 2017-08-08:
Adrien,
Thanks so much! I will do tonight/late evening when am back. I am
encouraged. I will run from bash.
Sent without spell check, my apologies: it should have been "for" not fro.
Your kindness to help is appreciated. I have learned more.

You have a good evening!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-09:
Adrien,

Am typing ./bin/aff3ct --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51"
--sim-snr-step "0.25" -K "1723" -N "2048" --src-type "AZCW" --chn-type
"AWGN_FAST"
Am I missing some string? or command. I type --help also.

It says nothing exists. I see it's there. Before that I ran aff3ct.exe as
below:
'C:\Users\kumud\AppData\Local\CMakeBuild\6cc5adb8-cc93-0134-82ab-9507b87758c7\build\x86-Debug\bin\Debug\aff3ct.exe'
This ran successfully.

Thank you for help!

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kouchy, 2017-08-09:
Yes it is normal. From Git Bash, go in this directory `C:\Users\kumud\AppData\Local\CMakeBuild\6cc5adb8-cc93-0134-82ab-9507b87758c7\build\x86-Debug\bin\Debug\` where the `aff3ct.exe` binary has been compiled. Next you have to adapt a little bit the command line like this: `./aff3ct.exe --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51"
--sim-snr-step "0.25" -K "1723" -N "2048" --src-type "AZCW" --chn-type
"AWGN_FAST"`. Your are close ;-).

kumuds4, 2017-08-09:
Adrien,
Thanks so much! Still giving an error but I think, I got it. I will try
until done. This weekend am, re-downloading LINUX (UBUNTU). There it's
easier from the root directory.
I have Python also downloaded. Will send plots as soon as I get success.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-11:
Adrien,

Where do you have the following:
conf/cde/awgn_polar_codes.zip
I should have Linux on Sat.Then will run. MS Studio is searching for .pdb
files, says missing. I still couldn't plot. I mean neither on laptop, nor
desktop. I did use GIT bash. I believe it will run on Linux (Ubuntu) fine.
In the mean time, am going over the literature/papers that you on aff3ct
website. BTW, I used same values that you have and plotted in Matlab. I did
change SNR range upon the suggestion of Professor with whom I work. The
plots attached.
Thanks so much for the help! You have a good night!

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 <(804)%20564-6670> or (804) 564-6692
<(804)%20564-6692>*
*Ph./Fax:(804)658-0125 <(804)%20658-0125>*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-08-14:
Hi Adrien,

How are you doing! My Linux install with virtual box did not work. It gives
an error. I have someone who would help to install actual Linux partition
next week not the virtual box on Windows 10. We did try for several hours
yesterday with virtual box. It kept giving an error. Right now I have been
trying again the commands you wrote a few days before on GIT bash. After I
am in the directory for bin where aff3ct.exe  is, I do the next command. It
keeps writing that such directory does not exist on GIT bash. I see for
Linux also you have "go to the root folder". It is same in Windows going to
users/AppData etc. I am not sure why it is not working.
I know you are busy. When you have a minute, please write what else should
I try? BTW, have downloaded your joint papers. I hope I can re-plot, then
go from there. I have told Professor with whom I work that the work you all
do is great. I want to follow on that.

Thank you again for your valuable time to help!
Best,

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

closed 2017-10-03 by kouchy