aff3ct/aff3ct #6: did ./bin o windows.

kumuds4, 2017-09-03:
Hello,
Thank you for all help! I have tried many times. I am trying again. Using Visual Studio 2017. 
I got upto 
'C:\Users\kumud\CMakeBuilds\6cc5adb8-cc93-0134-82ab-9507b87758c7\build\x86-Debug\bin\aff3ct.exe'
gives me on bash
[bashpic.pdf](https://github.com/aff3ct/aff3ct/files/1272659/bashpic.pdf)

However next step when I say this on gitbash for windows: and Adrien's and Couchy's instructions 
.bin/aff3ct.exe --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51" --sim-snr-step "0.25" -K "1723" -N "2048" --src-type "AZCW" --chn-type "AWGN_FAST"
does not do anything. It says no such directory. Picture attached as .pdf file.

OlivierHartmann, 2017-09-04:
Hi kumuds4,

This is Olivier from the Aff3ct's team.

We are really glad to hear that you have successfully run aff3ct! I can see on your screenshot you did it with the first command. However, I don't see the second one's result on your screenshot.
It seems you just do not run it in the right folder. With the first command you gave the aff3ct's absolute path to your computer so it could find it. But with the second command you gave a relative path so the computer tries to find it from the current working directory.

To solve this issue from the information you gave us, I think you just need to run this command first to change the working directory to the aff3ct build directory:
> cd \`C:\Users\kumud\CMakeBuilds\6cc5adb8-cc93-0134-82ab-9507b87758c7\build\x86-Debug\`

and then you will be able to run the second command: 

> ./bin/aff3ct.exe --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51" --sim-snr-step "0.25" -K "1723" -N "2048" --src-type "AZCW" --chn-type "AWGN_FAST"

Don't forget the "/" before the "bin" and after the dot.

Hoping I could help.

Olivier

kumuds4, 2017-09-04:
Hi There Olivier,

It is running. Adrien has been helping me too. I feel great. I have already
downloaded Python but is taking time, so am waiting. Am writing via desktop
and not disturbing laptop for screenshot.I will post once done.
So to make sure: right?
Run PyBER

Launch or re-launch (it is important to re-launch) Git Bash and type:

$ cd ~/aff3ct/plotter/PyBER/

$ python pyBER.py &

Thanks so much!

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-09-05:
Thank you so much Olivier! 
It is still running can't yet get to results has been more than 10 hours. Am patient? doing other things. 
Question: Am I doing right?
Please see screenshot.
[image]

OlivierHartmann, 2017-09-05:
Hi Kumuds!

You got the polar decoder running, bravo! But there are few things you shall work on it.

First, this is normal if it takes time. You are running the simulation really low now at a Frame Error Rate (FER) of 2*10^-8 and it will certainly keep falling a bit. You know, to go to the next simulation point (SNR) enter once ^C  (ctrl + C), and to stop the simulation enter twice really quickly ^C.

Secondly, you are running aff3ct in Debug mode. This is slow and not optimized at all. You shall compile again Aff3ct in Release mode. This will be much faster.

Then, your screenshot is weird. You shall get something like this [https://github.com/aff3ct/aff3ct](url) (scroll down) with one SNR per line. Make sure your terminal is large enough to allow the text to be written on one line only. Else you get what you got. I can see you used the Windows' terminal and not Git bash to run Aff3ct. Normally, text should be written correctly on it but we need to check. For now, what you got is not exploitable at all, you can stop the simulation. *Maybe* in Release mode, display will be better. If not, please try to use Git Bash or I can recommend you to use the standard output redirection: 

> ./bin/aff3ct.exe --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51" --sim-snr-step "0.25" -K "1723" -N "2048" --src-type "AZCW" --chn-type "AWGN_FAST" --sim-pyber "My curve legend" > your_file_name.txt

So you'll get a file called *your_file_name.txt* with your simulation data in your working directory that you can open anytime to see the results (even when it is running, but do not change it). More, on the terminal is displayed only the current processed SNR. This is a file ready to use under PyBER thanks to the **--sim-pyber** option that puts automatically your legend in the file.If you need to change the legend later, change it directly into the file (at the top), don't run again the simulation, but do it one the simulation is over.

*Without* the standard output redirection you just need to copy the results you see in the terminal and to paste them in the file in which you want to save them (don't forget to put a *.txt* file extension). But this works with a good display.

Finally, pyBER. Unfortunately the help has not been updated for this. I am sorry. I'll report it to Adrien. Aff3ct undergoes a lot of changes on these days. We will work on a better documentation after. Don't worry, it won't change nothing for you, normally.

PyBER is now under it's own project: [https://github.com/aff3ct/PyBER](url)

You can install it as you did for Aff3ct with Git bash. First make sure to be at your home:
> cd 

Then you can install pyBER
> git clone https://github.com/aff3ct/PyBER.git
> git submodule update --init --recursive

I think you did already, but if not, download and install python environment: https://repo.continuum.io/archive/Anaconda3-4.2.0-Windows-x86_64.exe

You can now run pyBER:
> cd PyBER
> python pyBER.py &

By default the files you can see in the tree on the left of the pyBER screen are contained in the *data* folder. To see the Aff3ct results with pyBER, put in this folder the files you have created above. If they do not have a *.txt* file extension you won't see them. Much better, if you use Git Bash, you can use the standard output redirection directly into this folder to see the curve progressing in real time:
> ./bin/aff3ct.exe --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51" --sim-snr-step "0.25" -K "1723" -N "2048" --src-type "AZCW" --chn-type "AWGN_FAST" --sim-pyber "My curve legend" > ~/PyBER/data/your_file_name.txt

There are a lot in this post, but to use aff3ct you need to master all of that. I hope I was clear and I helped you. If not, do not hesitate to ask again. But please write us exactly what you have done and what are the results of each command. Thank you.

Olivier

kumuds4, 2017-09-05:
> Windows-x86_64.exe

Olivier,
Thanks so much! The Windows 2017 Visual Studio is not good. It has no build
option. I checked on the web as well. I added incredibuild which is free
but am not sure will work. I need to do in release mode on GIT bash the way
wrote. I will do when am back.
is there Linux like command for release mode that you may send to me,
please.

Then will see whether  I get the results. I got the PyBer installed.

Best regards!
-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-05:
Kumuds,

I think you just need to switch, in the toolbar, from Debug to Release mode. I don't know Visual Studio but it should be easy to do as it is a good IDE.

To install aff3ct in linux like command under Git bash, I think I can't help you more than Adrien did already. I am sorry but I can just tell you to see this again: https://github.com/aff3ct/aff3ct/wiki/Compile-AFF3CT-on-Windows

You may want to know that Windows 10 now can support a Ubuntu kernel to run linux like command. But it would be using a sub-machine gun to kill a fly.

If it takes you too much time, maybe you should ask to your university IT department to help you. But 
you nearly done it! Keep going!

Olivier

kumuds4, 2017-09-05:
> aff3ct/wiki/Compile-AFF3CT-on-Windows

Olivier,

Thanks!

I think, I got it on Visual studio 2017 as well. I will run when am back.
Yes! both of you have helped me. I will ask help from IT if need be but as
you wrote, I am almost there. I will get back with plots.

Best,
-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-09-05:
> *cell:(804) 564-6670 <(804)%20564-6670> or (804) 564-6692
> <(804)%20564-6692>*
> *Ph./Fax:(804)658-0125 <(804)%20658-0125>*

Hi Olievier,

Thanks so much! It is running fast now.

I should work for classwork etc but am so inquisitive with your software. I
still have time till 4:45 p.m. before I leave for the evening class to
teach.

I will post/send as soon as I finished by the time. Otherwise will check
back when back late evening. Thanks to you and Adrien. I leanred. I am
planning to run all of yours first. Then will change rates etc. after
discussing with Professor with whom I work. I have been Applied
Mathematician with Physics background but got into electrical and computer
engineering and am taking lot sof classes myslef plus research. I am so
excited with all success. I download extra MS build tools for Visual studio
2017 from MSDN.

Thank you again! You all have a good evening!
-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

closed 2017-10-03 by kouchy