aff3ct/aff3ct #7: hi_gotendofsimulationsPyber?

kumuds4, 2017-09-06:
Hello there,
Thank you for help! Wow! I got the # end of simulations.  For PyBer I get to CD BER: it goes to the master folder but refuses to plot. Let me send screen shot for the end of simulations and what I am doing for PyBER. I feel great.  I will try again the PyBer Path.

Thanks again!

[polarberinria2.pdf](https://github.com/aff3ct/aff3ct/files/1282011/polarberinria2.pdf)

[polarpybernotyet.pdf](https://github.com/aff3ct/aff3ct/files/1282026/polarpybernotyet.pdf)

kumuds4, 2017-09-07:
Hi everybody,

After posting comments tried again. Got the plots. Thank you so much! This is great!
Olivier thanks for send details where to install PyBER and thanks so much Adrien. 
You have a good night!
[polarsc_scl_inria.pdf](https://github.com/aff3ct/aff3ct/files/1283108/polarsc_scl_inria.pdf)

OlivierHartmann, 2017-09-07:
Hi kumuds,

This is good news!
You can keep the "ctrl" key pressed to select several curves to display on PyBER.

Did you successfully displayed your own results on PyBER? Actually your results are really really weird. How did you get the point at 3.5 dB (Es/N0)? Why did you need to stop it (there is a little cross at the end of the line to show that)?

I am glad to see you finally done it!

kumuds4, 2017-09-11:
Hi Olivier,

Thank you for all help! It is running until end no and I got the plot. I am
trying what you have on the web to "Reproduce scientific results", I am
unable to get now. It says directory does t exist. Also, I want to plot
several curves on one plot and make it in more colors. I am working on it.
Once I get the results, will get back.
Thank you again.

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-09-11:
> *cell:(804) 564-6670 <(804)%20564-6670> or (804) 564-6692
> <(804)%20564-6692>*
> *Ph./Fax:(804)658-0125 <(804)%20658-0125>*

Olivier,

Where do you have this:

The dynamic SC decoder requires an accurate frozen bits generation method
which is not included in *AFF3CT* for now. In this paper we used an
external solution to construct the frozen bits following the Tal & Vardy
method (How to Construct Polar Codes <https://arxiv.org/pdf/1105.6164.pdf>).
To reproduce the results you have to unzip the conf/cde/awgn_polar_codes.zip in
the same folder (conf/cde/).

Thanks!
-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-11:
Hi kumuds,

In order to display several curves on PyBER you can use the "ctrl" keyboard. Keep it pushed and select the different curves you want. Curves colors will change automatically. I don't think you can change them directly. But you can make a right clic on the graph to have some options as the change of the X, Y axis, or the "export" tool. From it you can change your background color (I prefer it in white for documents). Please explore it, I don't know all internal tools.

The polar configuration files are in a compressed zip format because they are big. We call this an archive. To get back the files please see this https://support.microsoft.com/en-us/help/14200/windows-compress-uncompress-zip-files . You'll get then a list of files you can use in Aff3ct.

Finally, we have a polar decoder generator here https://github.com/aff3ct/polar_decoder_gen 
(see all our public projects here https://github.com/aff3ct?tab=repositories). This may help you.
I don't know about the frozen bit generation method, this is Adrien field. He is in vacation this month.

Thank you.

kumuds4, 2017-09-11:
> compress-uncompress-zip-files . You'll get then a list of files you can
> repositories). This may help you.

Hi Olivier,

Thanks! Right now, I was able to change rates for Polar codes, SC and is
running for 4096 with commands you sent.  I am doing as soon as it finishes
simulations to change the background to white. Adrien has written some
instructions for frozen bits' generation, I will read them this
afternoon/evening when am back.

I will post my new results on Github. It feels great! I hope to have enough
success to present at a conference or seminar. At moment, am sending to
Professor Wilson, who is an electrical engineer and I work with him a the
University of Virginia here. I have said to him that I will mention your
name(s) and software in the work and that aff3ct is much better than
Matlab. Once everything ready to go, will post/or send a copy to you. Thank
you for the public projects on decoding.

I am so grateful for all help!

Best regards!

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

closed 2017-10-03 by kouchy