aff3ct/aff3ct #9: Python.exe can't find PyBER

kumuds4, 2017-09-16:
It is not possible to find PyBER. How to fix for Python 3.6.2

OlivierHartmann, 2017-09-18:
Hi kumuds,

I thought you already successfully used PyBER? I don't understand what's wrong?

Can you give us more details about what you have done as the command line you used and the error you got.

kumuds4, 2017-09-18:
Olivier,

Thank you so much! You are great. I have not gone to bed yet was waiting
for a reply from you. I have been taking courses in Electrical and computer
engineering. So, with my Math & Physics background, I can do better
research. They gave a lab-work to do Python programming. I downloaded the
latest anaconda with Python 3.6.2 and that is what was needed for lab work.
It has messed up everything including PyBER and is not letting me do
anything. So I have un-installed all and reinstalled back as per
instructions on GIT hub for aff3ct for Windows. It seems like it is still
there somewhere in the registry. I mean the latest anaconda and python
3.6.2.

I am trying again and re-downloaded Visual studio 2017. Somehow hit the
wrong button instead x-86 release x-86 64  debug. It also gives an error
but was able to fix it. You have anaconda 3 with Python 2.7 and minGW64
also installs that to have in path. I am not sure why PyBER can't find.

On GIT hub am getting errors, is attached. I have been trying since last
Friday almost nonstop.  I was excited to do LDPC and everything was working
for Polar codes. Now, it is not working. I am hopeful it will again once
you see the attachment and guide me through to fix the problem. From now
on, I will install no more for the lab-work on this laptop, only on the
desktop.  The bottom line is not giving me anymore "end of the simulation"
instead I think it keeps running. That means a problem. Please look at the
attachment and let me know. I really like aff3ct software that you all have
created. It produces an excellent plot. The Professor with whom I work, he
also says yours i.e. aff3ct is giving much better results than Matlab.

Thank you again for the help! Best regards!

Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-18:
Kumuds,

I don't see your attached files. As you answering to a bot that paste your message into the github issue ticket here https://github.com/aff3ct/aff3ct/issues/9#issuecomment-330148441
So remember that all you say to us is public and that if you want to attach files you'll have to do it from github directly.

Now I think if you try to run PyBER with python3 in command line, you'll have to run it with:
> python3 pyBER.exe

If it doe not resolve your issue then I don't understand what you mean by "I am not sure why PyBER can't find".

For aff3ct issues please send again your screenshot. Remember that running really low points (low FER/BER) takes time if you have a good error correcting code (as LDPC).

Olivier

kumuds4, 2017-09-18:
> message into the github issue ticket here #9 (comment)

Olivier,

I am doing that. In a minute.
Thanks!
-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-18:
Can you scroll up and screen shot you're command line and the beginning of what aff3ct display please? Thanks

kumuds4, 2017-09-18:
Sending! Thank you!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-18:
still no file...

kumuds4, 2017-09-18:
Sending!

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-18:
Please send it from github directly...

kumuds4, 2017-09-18:
Just posted.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

OlivierHartmann, 2017-09-18:
oh ok sorry.

kumuds4, 2017-09-18:
I posted both the screen shots.

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

closed 2017-10-03 by kouchy