aff3ct/aff3ct #10: Problem with plots_PyBER!

kumuds4, 2017-09-18:
Screenshot of the latest simulation from 
[screnshotnew2.pdf](https://github.com/aff3ct/aff3ct/files/1310037/screnshotnew2.pdf)
the start of GIT.

OlivierHartmann, 2017-09-19:
Kumud,

Your results are really really weird! Can you tell me whether you have done something while aff3ct was running ?

Here is what I get with your same command line:
[image]

It took less than 10 seconds! Remember you strictly won't get the same results than me, because your OS threads management might be different than mine and your computer may does something else next to running aff3ct. But what you get here seems to be a numeric instability. I already got some on big computation servers.

Already last time you show us some results where there were some issues of that kind.

Please can you try again this run but with an AWGN channel and not AWGN_FAST. More use a naive implementation of the decoder instead of a fast one. Normally, these changes are not using SIMD instructions of you CPU. It might resolves your issues.

Here is the command line:
> ./build/bin/aff3ct --sim-type BFER -C POLAR -m 1 -M 4.5 -s 0.25 -K 1723 -N 2048 --src-type AZCW  --chn-type AWGN --dec-implem NAIVE

Show me results. Thank you.

kumuds4, 2017-09-19:
Olivier,
Please see!
I tried to run using command you sent today. It gives errors saying no file or directory. Should it be aff3ct.exe? Then I reran the older version. To me it seems it is still a trouble due to Python versions. I did reinstall all of them again yesterday. Am running on laptop and desktop both. 
If it runs over there, then there is some problem with all different anaconda. I have un-installed Python 3.6 
Let me try again after rebooting the computer and re-cloning aff3ct. Will post again in few minutes. Thanks!
[scrnshotnewltst.docx](https://github.com/aff3ct/aff3ct/files/1314761/scrnshotnewltst.docx)

OlivierHartmann, 2017-09-19:
Sorry Kumuds,

This is my bad, I have seen after sending the comment that I have done an error. Please always read the comments directly on github, we may make some edit.

You don't need to install and reinstall everything each time. It was working fine. The command is:
> ./bin/aff3ct.exe --sim-type BFER -C POLAR -m 1 -M 4.5 -s 0.25 -K 1723 -N 2048 --src-type AZCW --chn-type AWGN --dec-implem NAIVE

Please try to understand the command line. We won't help you later to run commands you want. We try just now to make you able to run aff3ct properly and to correct the bugs. If we give a command to you, you'll certainly have to adapt it to your system. I mean Linux command to Windows command for example, or in function of your working directory. I gave you a command to run aff3ct from my own working directory that's why I called aff3ct with ./build/bin/aff3ct . But you have to adapt it in function of where you call the program. You have compiled aff3ct with Visual Studio and did things we don't used to do. So you have to adapt yourself and take time to understand if you want to succeed with computers. I know, it is a hard science. I hope your futur computer science classes will help you to understand what you do.

Thank you

kumuds4, 2017-09-19:
Olivier,
I just got the message. Thanks! Am trying to install Linux OS or buy one laptop with it in coming days. 
 Windows  10 with Visual studio is good but different.  It has instructions on MSDN website, will try to fix the problem.  This time, I should get the plot. Will post as soon as I have it. Thanks for valuable time.

closed 2017-10-03 by kouchy