aff3ct/aff3ct #11: success_simulated_thanks!

kumuds4, 2017-09-19:
Olivier,
Please see attached. Got success for # end of simulations on desktop. Adrien had instructions for Visual studio for me. However, it is not finding Python. I have Python cloud, Python 35, 36. MinGW64 also installs I believe, checked. I think I need to figure out the correct path. I tried few times. Thank you so much! Here is screen shot showing not plotting yet. 
You all have good evening!

[desktop_succes_nographyet.pdf](https://github.com/aff3ct/aff3ct/files/1315199/desktop_succes_nographyet.pdf)

OlivierHartmann, 2017-09-20:
Kumuds,

I am glad to here it works. However, I don't understand, didn't work the command "_python3 pyBER.py &_" ?
I can see you run it, but what was the result?

kumuds4, 2017-09-20:
Oliver,

Thank you!
I can't get to plot. On Git bash you may see in another post? It says
command not found. I mean PyBER  plot. l think it is problem with Anaconda
& Python version I had to download few days back and program is confused.
Best regards!
Kumud

>
> Kumuds,
>
> I am glad to here it works. However, I don't understand, didn't work the
command "python3 pyBER.py &" ?

closed 2017-10-03 by kouchy