aff3ct/aff3ct #13: hello_plsee!

kumuds4, 2017-09-23:
Thank you for helping! I tried again. IT was not much of help. He knows somewhat of Python? Am fine, learning. Went over Olivier your's recent & other message and that of Adrien's previous messages regarding PATH. It helped. Have Anaconda3 installed with 3.6.1 Python is not letting me install Python 3.6.2. Both similar, OK.
Ran. Sending the screen shot. However, it still can't find Python 3 to help PyBER for the plot.
Anaconda 3 is in Continuum.  
C:\Users\kumud\AppData\Local\Continuum\Anaconda3

Note: With the screenshot, I feel, I am very near to plots. Like you mentioned, it is not able to find Python 3 binary. Am not sure why? It is there. 
Thank you again for all help. In one message is mentioned by you Olivier that Adrien will update for PyBER some help. Would be a great help. I may install Linux by next week or get a laptop with only Linux. You have more help there. 
Nice weekend!
-Kumud

[newrunpyBER_noplot.pdf](https://github.com/aff3ct/aff3ct/files/1326179/newrunpyBER_noplot.pdf)

OlivierHartmann, 2017-09-25:
Hi kumud,

How useful  IT departments always are...

Try to run this:
> PATH=$PATH:/c/Users/kumud/AppData/Local/Continuum/Anaconda3
> cd ~/PyBER
> python3 pyBER &

If you still get nothing, please run these commands:
> echo $PATH
> whereis -b python
and return us the results.

kumuds4, 2017-09-25:
Hi, Olivier,

I am so grateful for sending me commands. I am running in few minutes. Will
get back.

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

kumuds4, 2017-09-25:
Olivier,

It is still searching Python. I see all other in the PATH with CMake when
did the command

echo $PATH

It's getting late, almost 4 A.M. Will resume in the morning. I will get
back. Thank you again for the help!

Best,

-Kumud

-Kumud

****************************************************************
" A Mathematician is a device, turning coffee into Theorems"
-Alfréd Rényi, Budapest, Hungarian Academy of Sciences

****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D.*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804)658-0125*
*Email: ksa2te@virginia.edu <ksa2te@virginia.edu> alias: kumudsa@gmail.com
<kumudsa@gmail.com> *

closed 2017-10-03 by kouchy