aff3ct/aff3ct #14: Success_again!thank_you!

kumuds4, 2017-09-28:
Hello, Olivier,
Thank you for help to search Python. Finally, I just got success, am posting the screen-shot. Is it possible to update a few help for PyBER plotter?  I still have trouble exporting to Matplotlib. I added to PATH the latest PYTHON i.e. 64 bit Python 3.6.2 by using Anaconda3 in the Visual Studio. Thank you for the PATH command. It worked. 
I am working on getting installed Linux. Then, will run all of those you have in documentation/community using Linux command. Then, will try to generate at different rates. I am interested in generating a different set of frozen bits. I need to discuss with Professor I work. Then, will do more simulations.  
Thank you again for your valuable time to help!

[4096_aff3ct_ltstplot.pdf](https://github.com/aff3ct/aff3ct/files/1339040/4096_aff3ct_ltstplot.pdf)

OlivierHartmann, 2017-09-28:
I'm glad to hear it works.
You were welcome.

kumuds4, 2017-09-28:
Thanks for the help! Once did more simulations, will be posting.

Best regards!

-Kumud

closed 2017-10-03 by kouchy