aff3ct/aff3ct #16: Measure throughput performance

danghoaison91, 2018-01-10:
Hello Aff3ct team,

I have a question regarding throughput measurement.

On your instruction page, I saw the result of throughput per thread, but on the code cloned from github, only the BER result is shown. So what is the problem here?

Thank you!

kouchy, 2018-01-10:
Hi @danghoaison91,

We changed that to simplify the actual output of the simulator (only the throughput of the entire multi-threaded simulation is shown by default).

If you want the throughput per thread you can use the "--sim-stats" option which gives you a lot of info.

I hope it helps.
Best regards.

danghoaison91, 2018-01-15:
It works for me, these measurement is very useful.

Thank you.

closed 2018-01-15 by kouchy