aff3ct/aff3ct #17: Polar code performance variation in changing --cde-sigma

codechecker123, 2018-03-07:
Hi, thanks so much for this codebase.  I find it extremely useful.

## Background

I was hoping you could help me understand a performance quirk with polar codes.

I was working on commit `4883036` and at that point there was a command line argument for setting the `--cde-sigma` value.  Through grid searching, I found that `--cde-sigma .87` provided a 2dB (!) improvement over the default `auto` sigma for *half-rate* polar codes with a *QAM modulation order (`mod-bps`) greater than 2*.

This was great, until I also found that setting sigma in this way made is so that many rates other than 1/2 didn't work at all.  

For instance, below, I show a plot with rate = 5/6 and 1/2.  You can see that when sigma is fixed, the 1/2 rate polar code is fantastic, but the 5/6 rate code is catastrophically bad.

I used commands like the following to generate the plot on the old commits and similar commands with the updated CLI arguments on the current master branch to create the other plots in the graph.

```
../build/bin/aff3ct --cde-type POLAR -m 0 -M 16 -s 0.5 -K 2058 -N 4096 --crc-poly 32-GZIP --dec-type ASCL --mod-type QAM --mod-bps 4 --cde-sigma .87
../build/bin/aff3ct --cde-type POLAR -m 0 -M 16 -s 0.5 -K 2058 -N 4096 --crc-poly 32-GZIP --dec-type ASCL --mod-type QAM --mod-bps 4
```

[image]

## Question

I would like to achieve two goals:

- Maintain the superior performance that I had with R=1/2, sigma=.87 using the current version of aff3ct in master.  I am not sure how to do this as `--cde-sigma` appears to have been depreciated.  Is this possible through other means?  You can see from the plot, that the default parameters in master (purple line) do not perform as well as the blue line.

- Not have the decoder be catastrophically bad for R != 1/2.  That is, the BER shouldn't be red line.  I'd settle for the green line.

## Observations

- I only see the improvement (ie the blue-purple SNR gap) when setting sigma for the half rate codes with *bps >= 4*.  For BPSK/QPSK/4-QAM, there is no SNR gap.

- As N grows larger, it seems that more rates (not equal to 1/2) become unusable.  Do you have any intuition for why that might be?

For instance the following converges to a vanishing BER

```
../build/bin/aff3ct --cde-type POLAR -m 5 -M 16 -s 0.5 -K 1600 -N 2000 --crc-poly 32-GZIP --dec-type ASCL --mod-type QAM --mod-bps 4 --cde-sigma .87
# ----------------------------------------------------------------------||--------------------------------||---------------------
#       Bit Error Rate (BER) and Frame Error Rate (FER) depending       ||     Decoder throughput and     ||  Global throughput  
#                    on the Signal Noise Ratio (SNR)                    ||      latency (per thread)      ||  and elapsed time   
# ----------------------------------------------------------------------||--------------------------------||---------------------
# -------|-------|----------|----------|----------|----------|----------||----------|----------|----------||----------|----------
#  Es/N0 | Eb/N0 |      FRA |       BE |       FE |      BER |      FER ||     CTHR |     ITHR |  LATENCY || SIM_CTHR |    ET/RT 
#   (dB) |  (dB) |          |          |          |          |          ||   (Mb/s) |   (Mb/s) |     (us) ||   (Mb/s) | (hhmmss) 
# -------|-------|----------|----------|----------|----------|----------||----------|----------|----------||----------|----------
    9.96 |  5.00 |      131 |    13036 |      131 | 6.35e-02 | 1.00e+00 ||     1.45 |     1.11 |  1411.32 ||     1.39 | 00h00'00  
   10.46 |  5.50 |      138 |     8150 |      126 | 3.77e-02 | 9.13e-01 ||     2.04 |     1.56 |  1005.64 ||     1.96 | 00h00'00  
   10.96 |  6.00 |      318 |     5414 |      124 | 1.09e-02 | 3.90e-01 ||     2.94 |     2.25 |   696.69 ||     4.25 | 00h00'00  
   11.46 |  6.50 |     2060 |     3062 |      105 | 9.48e-04 | 5.10e-02 ||    10.37 |     7.94 |   197.52 ||    21.31 | 00h00'00  
   11.96 |  7.00 |    27101 |     1704 |      100 | 4.01e-05 | 3.69e-03 ||    35.82 |    27.42 |    57.17 ||    67.68 | 00h00'00  
   12.46 |  7.50 |   231875 |      430 |       55 | 1.18e-06 | 2.37e-04 ||    58.96 |    45.14 |    34.73 ||    78.75 | 00h00'06
```

But the following similar code does not converge:

```
../build/bin/aff3ct --cde-type POLAR -m 5 -M 16 -s 0.5 -K 1900 -N 2400 --crc-poly 32-GZIP --dec-type ASCL --mod-type QAM --mod-bps 4 --cde-sigma .87

# ----------------------------------------------------------------------||--------------------------------||---------------------
#       Bit Error Rate (BER) and Frame Error Rate (FER) depending       ||     Decoder throughput and     ||  Global throughput  
#                    on the Signal Noise Ratio (SNR)                    ||      latency (per thread)      ||  and elapsed time   
# ----------------------------------------------------------------------||--------------------------------||---------------------
# -------|-------|----------|----------|----------|----------|----------||----------|----------|----------||----------|----------
#  Es/N0 | Eb/N0 |      FRA |       BE |       FE |      BER |      FER ||     CTHR |     ITHR |  LATENCY || SIM_CTHR |    ET/RT 
#   (dB) |  (dB) |          |          |          |          |          ||   (Mb/s) |   (Mb/s) |     (us) ||   (Mb/s) | (hhmmss) 
# -------|-------|----------|----------|----------|----------|----------||----------|----------|----------||----------|----------
    9.93 |  5.00 |      133 |    23615 |      133 | 9.51e-02 | 1.00e+00 ||     2.48 |     1.13 |  1653.78 ||     2.55 | 00h00'00  
   10.43 |  5.50 |      136 |    21650 |      136 | 8.52e-02 | 1.00e+00 ||     2.44 |     1.11 |  1678.56 ||     3.32 | 00h00'00  
   10.93 |  6.00 |      136 |    18543 |      136 | 7.30e-02 | 1.00e+00 ||     2.27 |     1.04 |  1801.99 ||     3.70 | 00h00'00  
   11.43 |  6.50 |      129 |    12789 |      129 | 5.31e-02 | 1.00e+00 ||     2.57 |     1.17 |  1596.64 ||     3.45 | 00h00'00  
   11.93 |  7.00 |      137 |     8062 |      137 | 3.15e-02 | 1.00e+00 ||     2.51 |     1.14 |  1634.83 ||     3.51 | 00h00'00  
   12.43 |  7.50 |      127 |     4245 |      127 | 1.79e-02 | 1.00e+00 ||     2.81 |     1.28 |  1456.65 ||     3.44 | 00h00'00  
   12.93 |  8.00 |      127 |     2293 |      127 | 9.67e-03 | 1.00e+00 ||     3.23 |     1.47 |  1269.39 ||     3.48 | 00h00'00  
   13.43 |  8.50 |      134 |     1463 |      134 | 5.84e-03 | 1.00e+00 ||     2.82 |     1.29 |  1451.53 ||     3.55 | 00h00'00  
   13.93 |  9.00 |      136 |     1337 |      136 | 5.26e-03 | 1.00e+00 ||     1.88 |     0.86 |  2176.23 ||     3.49 | 00h00'00  
   14.43 |  9.50 |      120 |     1133 |      120 | 5.05e-03 | 1.00e+00 ||     3.39 |     1.55 |  1206.92 ||     2.94 | 00h00'00  
   14.93 | 10.00 |      130 |     1284 |      130 | 5.29e-03 | 1.00e+00 ||     2.05 |     0.94 |  1996.01 ||     3.03 | 00h00'00  
   15.43 | 10.50 |      132 |     1262 |      132 | 5.12e-03 | 1.00e+00 ||     2.77 |     1.26 |  1480.00 ||     3.31 | 00h00'00  
   15.93 | 11.00 |      130 |     1266 |      130 | 5.21e-03 | 1.00e+00 ||     3.15 |     1.44 |  1299.59 ||     3.35 | 00h00'00  
   16.43 | 11.50 |      131 |     1270 |      131 | 5.19e-03 | 1.00e+00 ||     3.26 |     1.49 |  1257.01 ||     3.89 | 00h00'00  
   16.93 | 12.00 |      136 |     1257 |      136 | 4.95e-03 | 1.00e+00 ||     2.63 |     1.20 |  1555.87 ||     3.72 | 00h00'00  
   17.43 | 12.50 |      129 |     1201 |      129 | 4.98e-03 | 1.00e+00 ||     3.06 |     1.39 |  1339.16 ||     3.67 | 00h00'00  
   17.93 | 13.00 |      130 |     1266 |      130 | 5.21e-03 | 1.00e+00 ||     3.32 |     1.52 |  1232.61 ||     3.13 | 00h00'00  
   18.43 | 13.50 |      137 |     1293 |      137 | 5.05e-03 | 1.00e+00 ||     2.32 |     1.06 |  1764.82 ||     3.19 | 00h00'00  
   18.93 | 14.00 |      134 |     1303 |      134 | 5.21e-03 | 1.00e+00 ||     1.34 |     0.61 |  3053.99 ||     2.98 | 00h00'00  
   19.43 | 14.50 |      129 |     1269 |      129 | 5.27e-03 | 1.00e+00 ||     2.31 |     1.05 |  1772.11 ||     3.42 | 00h00'00  
   19.93 | 15.00 |      128 |     1162 |      128 | 4.86e-03 | 1.00e+00 ||     3.14 |     1.43 |  1304.99 ||     3.21 | 00h00'00  
   20.43 | 15.50 |      139 |     1326 |      139 | 5.11e-03 | 1.00e+00 ||     2.24 |     1.02 |  1826.81 ||     3.50 | 00h00'00  
   20.93 | 16.00 |      133 |     1263 |      133 | 5.08e-03 | 1.00e+00 ||     2.85 |     1.30 |  1437.90 ||     3.35 | 00h00'00

```

Any help, pointers, or discussions of the issues involved in my observations would be greatly appreciated!!

madellimac, 2018-03-07:
Dear codechecker123,

thank you for your detailed post.
Everything that you are pointed out is actually expected.
In AFF3CT, when you use the `--cde-sigma` (updated to `--enc-fb-sigma` in the last versions of AFF3CT), the polar code that is used for simulation is generated for a fixed sigma value. If you do not specify this option, a new polar code is generated for each SNR point that you want to simulate.
The decoding performance is very sensitive to the choice of the polar code. Among other things, the code rate, the modulation have to be considered to choose the right polar code. I practice, one often choose it experimentaly by running simulations.

Hoping to have answered your questions,

Best,

Camille

codechecker123, 2018-03-07:
Camile, thanks so much for your quick reply.  

So you are saying there is some sigma value for my 5/6 code that will perform well?  If so, that is great, I am happy to search.  Or is it the case that certain n,k combinations will never perform well regardless of sigma?

Are there any other polar code parameters that I should search over, in addition to sigma, to optimize performance?

Also, can you help me understand why the performance discrepancies only appear for higher-order QAM (bps >= 4).  If I understand that better, I hope to be more efficient at the optimization.

Thanks!

madellimac, 2018-03-07:
Dear codechecker123,

you can definitely try to optimize your polar code given your setup (modulation + code rate).

If you want to further improve your performance, you can try different decoding algorithms, CRC-SCL being probably the best choice to date.

I have to confirm that with our main dev, but I think that the sigma value calculation for the polar code was designed for BPSK, I am not sure that we adapted it to other modulations. This would explain why you see such a shift for high order modulations.

Best,
Camille

codechecker123, 2018-03-07:
Thanks Camille, one more question for you: should I be using https://github.com/aff3ct/polar_decoder_gen to find the best parameters for my code?  If so, how to i use the output `.hpp` files from that tool in the aff3ct simulations?

kouchy, 2018-03-08:
Dear codechecker123,

I confirm what Camille said before, we did not optimize the polar codes (or the frozen bits) for complex modulations.

The repo https://github.com/aff3ct/polar_decoder_gen is not about polar codes generation (or frozen bits generation). It is dedicated to the generation of polar **decoders**.
For more details on the decoders generation, you can read the associated paper "An Efficient, Portable and Generic Library for Successive Cancellation Decoding of Polar Codes" (see our dedicated page for more precise links: http://aff3ct.github.io/publications.html).

Best regards,
Adrien

codechecker123, 2018-03-11:
@kouchy / @madellimac, thanks again for your pointers, but I am afraid I am not having any luck finding a sigma that works.

If I run

```
aff3ct -C POLAR -m 10 -M 15 -s 0.5 -K 384 -N 512 --crc-poly 32-GZIP --dec-type ASCL --mdm-type QAM --mdm-bps 6
```

I get 

```
# -------|-------|----------|----------|----------|----------|----------||----------|----------
#  Es/N0 | Eb/N0 |      FRA |       BE |       FE |      BER |      FER ||  SIM_THR |    ET/RT
#   (dB) |  (dB) |          |          |          |          |          ||   (Mb/s) | (hhmmss)
# -------|-------|----------|----------|----------|----------|----------||----------|----------
   16.15 | 10.00 |      192 |     3384 |      118 | 5.01e-02 | 6.15e-01 ||     0.07 | 00h00'00
   16.65 | 10.50 |      279 |     2909 |      117 | 2.96e-02 | 4.19e-01 ||     1.16 | 00h00'00
   17.15 | 11.00 |      669 |     2310 |      108 | 9.81e-03 | 1.61e-01 ||     2.91 | 00h00'00
   17.65 | 11.50 |     2425 |     2049 |      105 | 2.40e-03 | 4.33e-02 ||     5.98 | 00h00'00
   18.15 | 12.00 |    12813 |     1554 |      101 | 3.45e-04 | 7.88e-03 ||    12.52 | 00h00'00
   18.65 | 12.50 |   156183 |     1392 |      100 | 2.53e-05 | 6.40e-04 ||    15.39 | 00h00'03
   19.15 | 13.00 |   488450 |      240 |       18 | 1.40e-06 | 3.69e-05 ||    15.78 | 00h00'10 x
   19.65 | 13.50 |   489924 |        0 |        0 | 5.80e-09 | 2.04e-06 ||    15.81 | 00h00'10 x
   20.15 | 14.00 |     6049 |        0 |        0 | 4.70e-07 | 1.65e-04 ||    10.45 | 00h00'00 x
```

Which is great.  

But when I manually searched over sigma, no value seems to drive the error rate to zero. For example, I have tried a logarithmic grid on sigma from 0dB to 20dB (0.1dB increments) and the error rate fails to go to zero for any value I have tested.  Here is an example of my search with 10dB increments:

```
aff3ct -C POLAR -m 14 -M 15 -s 0.5 -K 384 -N 512 --crc-poly 32-GZIP --dec-type ASCL --mdm-type QAM --mdm-bps 6 --enc-fb-sigma 1
aff3ct -C POLAR -m 14 -M 15 -s 0.5 -K 384 -N 512 --crc-poly 32-GZIP --dec-type ASCL --mdm-type QAM --mdm-bps 6 --enc-fb-sigma 10
aff3ct -C POLAR -m 14 -M 15 -s 0.5 -K 384 -N 512 --crc-poly 32-GZIP --dec-type ASCL --mdm-type QAM --mdm-bps 6 --enc-fb-sigma 100
```

Am I missing some other argument that I need to pass as part of the search? 

Is it possible to print out a debug from the "frozen bit generator" and then copy over the values to my use case?

Adrien, when you say 

> we did not optimize the polar codes (or the frozen bits) for complex modulations

Is it straightforward for me to do this optimization with the existing code?  Or does this optimization involve code changes and deeper knowledge of polar codes?

Thanks again for your help!

kumuds4, 2018-03-13:
Is there a way to run on Windows this decoder? I don't have yet access to LINUX/UNIX.
Thank you. I may in coming months.

kouchy, 2018-03-14:
Dear @codechecker123,

There was a bug in the code, I fixed it (see https://github.com/aff3ct/aff3ct/commit/cae9471c5f156b73f5c64ee7b55afc9653098ddb). Now it should work.

Yes it is possible to get the frozen bits but you have to modify the code a little bit.
If you are in the adaptive mode you can print out the frozen bits from here : https://github.com/aff3ct/aff3ct/blob/master/src/Module/Codec/Polar/Codec_polar.cpp#L176.
If you are in the static mode you can print out the frozen bits from here : https://github.com/aff3ct/aff3ct/blob/master/src/Module/Codec/Polar/Codec_polar.cpp#L127.

If you want to optimize the frozen bits generation for complex modulations I think you have to understand the Gaussian approximation method. You can refer to the P. Trifonov paper : "Efficient Design and Decoding of Polar Codes" (https://doi.org/10.1109/TCOMM.2012.081512.110872).
In the AFF3CT code, you can find the implementation for the BPSK here : https://github.com/aff3ct/aff3ct/blob/master/src/Tools/Code/Polar/Frozenbits_generator/Frozenbits_generator_GA.cpp#L26-L53.
This code is also used to generate the frozen bits for complex modulations but it is sub-optimal as you reported before.

@kumuds4, your question is not related to this thread, please avoid to mix the topics. AFF3CT can run on Windows, MacOSX and Linux with the same level of features.

Best regards,
Adrien

kumuds4, 2018-03-14:
> There was a bug in the code, I fixed it (see cae9471
> Codec/Polar/Codec_polar.cpp#L176.
> Codec/Polar/Codec_polar.cpp#L127.
> Code/Polar/Frozenbits_generator/Frozenbits_generator_GA.cpp#L26-L53.

Dear Adrien,

Gratefully acknowledged!

****************************************************************
*Wir müssen wissen — wir werden wissen! - David Hilbert*

*We must know — we will know!"The secret of change is to focus on all of
your energy not on fighting the old, but building the new" - Socrates*
****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D. (Applied Math)*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804) 303-5163*
*Email: kumudsa@gmail.com <kumudsa@gmail.com> *
*Alternate email: kumud.altmayer@faculty.umuc.edu
<kumud.atmayer@faculty.umuc.edu>*

codechecker123, 2018-03-20:
@kouchy, thanks for the fix!  Varying sigma works as I would have expected now.

closed 2018-03-20 by codechecker123