aff3ct/aff3ct #19: Simmulation with low N

titelecom, 2018-03-15:
Hi,

I would like to know if is possible run a simulation with low codeword size "N", for example, N =8. The minimum "N" that I was able to run was 64.  I tried to simulate using this command line:

aff3ct --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51" --sim-snr-step "0.25" -K "4" -N "8" --src-type "AZCW" --chn-type AWGN --chn-implem FAST --dec-type "SC" --dec-simd "INTRA" -L "1" --sim-pyber "Fast-SSC"

and I got this message:

"'_decode' function): "'m' has to be equal or greater than 'static_level' ('m' = 3, 'static_level' = 6).""

Thank you

bonben, 2018-03-15:
Hi.

In order to speed up the decoding of the "FAST" version of the polar decoder, some parts of the decoder are compiled statically, but as you saw, the drawback is that very short polar codes cannot be run.

To run the simulation with N = 8, you should run the "NAIVE" version of the polar decoder, by running the following command: 

```
aff3ct --sim-type "BFER" -C "POLAR" -m "1.0" -M "4.51" --sim-snr-step "0.25" -K "4" -N "8" --src-type "AZCW" --chn-type AWGN --chn-implem FAST --dec-type "SC" -L "1" --sim-pyber "Fast-SSC"  --dec-implem "NAIVE"
```

Note that the ``` --dec-simd "INTRA"```, that corresponds to SIMD vectorization, has been removed, and that the NAIVE version of decoder implementation has been selected with ```--dec-implem "NAIVE"```

Nevertheless, if you need to run the "FAST" version of the decoder, you can modify the following line in the source code in order to reduce the level of the statically compiled parts of the tree.

https://github.com/aff3ct/aff3ct/blob/master/src/Module/Decoder/Polar/SC/Decoder_polar_SC_fast_sys.hxx#L28

The value of the ```static_level``` variable can be decreased to 1, for example. Then you could use the command line that you proposed in your original post.

titelecom, 2018-03-15:
thanks, it works well

closed 2018-03-15 by titelecom