aff3ct/aff3ct #21: Channels input by a file

titelecom, 2018-03-16:
Hi,

I would like to know how to input the channel index by a file when I am running the polar code. I tried the following command:

aff3ct --sim-type "BFER" -C "POLAR" -m "1" -M "4" --sim-snr-step "1" -K "128" -N "256" --src-type "AZCW" --chn-type AWGN --chn-implem FAST --dec-type "SCL" --dec-simd "INTRA" -L "8" --dec-polar-nodes "{R0,R0L,R1,REP,REPL}" --sim-pyber "Fast-SCL32" --sim-snr-type "EB" --enc-fb-gen-method "FILE" --enc-fb-awgn-path "src/FZB/8/N256_awgn_s0.500.pc"

And there is the below content in the "N256_awgn_s0.500.pc" file.

 256
awgn
0.500
255 127 191 223 239 247 251 253 254 63 95 159 111 175 207 119 183 215 231 123 187 219 235 125 189 243 221 237 245 126 190 31 222 47 79 55 143 238 87 151 249 103 59 167 91 155 199 107 246 171 203 61 115 93 179 157 109 211 173 250 205 117 227 181 62 94 213 158 15 110 23 121 174 39 185 229 27 71 252 206 118 43 135 217 182 75 51 139 29 214 83 233 45 147 122 77 99 186 53 141 230 163 85 30 241 218 149 195 46 57 101 124 78 7 165 89 234 54 188 142 153 11 86 197 105 220 19 150 242 169 58 13 102 35 113 236 90 166 201 21 67 177 154 37 131 198 106 60 244 14 25 209 69 170 92 22 114 41 133 248 225 202 156 108 178 38 73 49 26 172 210 137 3 70 116 204 81 134 226 42 180 145 5 120 97 74 212 28 50 184 161 138 228 216 193 44 82 9 232 146 76 6 98 240 52 140 162 17 194 84 148 56 33 10 100 164 88 65 196 152 129 104 18 168 112 200 12 176 34 208 224 66 130 20 36 68 24 1 132 40 72 136 48 80 144 96 160 192 2 4 16 0 8 32 64 128

I had unexpected results using this procedure. Am I doing the right procedure? 

Thanks

kouchy, 2018-03-17:
Hi @titelecom,

You are using the right command but I think that your frozen bits are not well generated.
I tried the same command with this code : https://github.com/aff3ct/configuration_files/blob/master/cde/awgn_polar_codes/TV/8/N256_awgn_s0.489.pc and I performed much better :

```
./bin/aff3ct --sim-type "BFER" -C "POLAR" -m "1" -M "4" --sim-snr-step "1" -K "128" -N "256" --src-type "AZCW" --chn-type AWGN --chn-implem FAST --dec-type "SCL" --dec-simd "INTRA" -L "8" --dec-polar-nodes "{R0,R0L,R1,REP,REPL}" --sim-snr-type "EB" --enc-fb-gen-method "FILE" --enc-fb-awgn-path "../conf/cde/awgn_polar_codes/TV/8/N256_awgn_s0.489.pc"
```

I hope it helps.
Best regards

titelecom, 2018-03-19:
Thanks for the answer. Do you know what is means the third line in the file?
I think that is the sigma value. If I am right, it should be an exact value or this value doesn't matter because I wrote the channels in the file?

Best regards.

bonben, 2018-03-19:
Hi @titelecom 

The format of this file is a legacy from other projects and softwares. So you're right, it represents the sigma value. But in aff3ct, it is not read at all, therefore it will not affect (!) your simulations.

Regards.

titelecom, 2018-03-19:
Hi @bonben,

Thanks for the answer. I am trying to compare the AFF3CT implementation with my implementation that I did in MatLab. I used the "Bhattacharyya parameter" to generate the channels index as showed in "A Comparative Study of Polar Code Constructions for the AWGN Channel" paper.

So I pretend to insert the same frozen bits pattern in the AFF3CT and in my code and I expect to see the same performance in BER and FER.

I have more two doubts:

1. The file indicates to encoder and decoder what is the good channels to send the information bits in ascendent order. Am I right?

2. The SC decoder processes the bits in "bit reverse order" or in a regular order? Because it should influence the frozen bits positions.

Best Regards and thanks again to help me.

bonben, 2018-03-19:
1. I don't know what you mean by ascendant. The first channel is the most reliable (see attached file, 7 is the most reliable channel).
[N8_awgn_s0.501.txt](https://github.com/aff3ct/aff3ct/files/1825221/N8_awgn_s0.501.txt)

2. There is no bit reversal in any of our polar encoder & decoder implementations.

> Best Regards and thanks again to help me.
You're welcome

kumuds4, 2018-03-22:
git submodule update --init --recursive
fatal: 'submodule' appears to be a git command, but we were not
able to execute it. Maybe git-submodule is broken?

This is the message, am getting. can't update or init. Need your input/help.

closed 2018-03-30 by kouchy