aff3ct/aff3ct #22: init issue!

kumuds4, 2018-03-22:
At GIT Bash:  git submodule update --init --recursive
fatal: 'submodule' appears to be a git command, but we were not
able to execute it. Maybe git-submodule is broken?

How should I fix this error?

OlivierHartmann, 2018-03-23:
Kumuds,
Please google your issue before posting a new issue here that does not relate directly to AFF3CT.
https://stackoverflow.com/a/14807969/7219905
Thank you.

kumuds4, 2018-03-23:
I did but it did not work, tried also.

****************************************************************
*Wir müssen wissen — wir werden wissen! - David Hilbert*

*We must know — we will know! "It does not COST to be nice & KIND"*
****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D. (Applied Math)*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804) 303-5163*
*Email: kumudsa@gmail.com <kumudsa@gmail.com> *
*Alternate email: kumud.altmayer@faculty.umuc.edu
<kumud.atmayer@faculty.umuc.edu>*

closed 2018-03-30 by kouchy