aff3ct/aff3ct #24: tried again!

kumuds4, 2018-04-09:
Added ninja.exe in Camake.list, still gives following error during compilation of aff3ct. Am trying. 
Severity	Code	Description	Project	File	Line	Suppression State
Error		CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.10/Modules/CMakeTestCXXCompiler.cmake:45 (message):
  The C++ compiler

    "C:/Program Files/CMake/bin/cmake.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/kumud/CMakeBuilds/6cc5adb8-cc93-0134-82ab-9507b87758c7/build/x86-Debug/CMakeFiles/CMakeTmp
    
    Run Build Command:"C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/Ninja/ninja.exe" "cmTC_42a69"
    [1/2] Building CXX object CMakeFiles\cmTC_42a69.dir\testCXXCompiler.cxx.obj
    FAILED: CMakeFiles/cmTC_42a69.dir/testCXXCompiler.cxx.obj 
    C:\PROGRA~1\CMake\bin\cmake.exe     -o CMakeFiles\cmTC_42a69.dir\testCXXCompiler.cxx.obj -c testCXXCompiler.cxx
    CMake Error: The source directory "C:/Users/kumud/CMakeBuilds/6cc5adb8-cc93-0134-82ab-9507b87758c7/build/x86-Debug/CMakeFiles/CMakeTmp/testCXXCompiler.cxx" is a file, not a directory.
    Specify --help for usage, or press the help button on the CMake GUI.
    ninja: build stopped: subcommand failed.
    

  

  CMake will not be able to correctly generate this project.		C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.10/Modules/CMakeTestCXXCompiler.cmake	45

OlivierHartmann, 2018-04-09:
Kumud,
This:
> CMake Error: The source directory "C:/Users/kumud/CMakeBuilds/6cc5adb8-cc93-0134-82ab-9507b87758c7/build/x86-Debug/CMakeFiles/CMakeTmp/testCXXCompiler.cxx" is a file, not a directory.

certainly means that you gave to cmake a wrong path. Please try to check that.

Olivier

kumuds4, 2018-04-09:
Gratefully acknowledged! Will try again this evening!

****************************************************************
*Wir müssen wissen — wir werden wissen! - David Hilbert*

*We must know — we will know! "It does not COST to be nice & KIND"*
****************************************************************
*Best regards!*

*Mrs. Kumud Singh_Altmayer Ph.D. (Applied Math)*

*cell:(804) 564-6670 or (804) 564-6692*
*Ph./Fax:(804) 303-5163*
*Email: kumudsa@gmail.com <kumudsa@gmail.com> *
*Alternate email: kumud.altmayer@faculty.umuc.edu
<kumud.atmayer@faculty.umuc.edu>*

closed 2018-04-12 by aff3ct