aff3ct/aff3ct #25: BSC Channels (Polar codes)

YangZhiyu1994, 2018-05-07:
Hi,
I would like to  know how the performance of SC and SCL decode under the BSC channel. And under BSC channel, which locations to choose as frozen bits are better?
Thanks

OlivierHartmann, 2018-05-07:
Hi YangZhiyu1994,

Thank you for using AFF3CT. Sorry we can't tell you but a good starting point might be around [this paper of Erdal Arıkan](https://arxiv.org/pdf/0807.3917.pdf).

I'm currently integrating BEC and BSC into AFF3CT. I'll come back soon to tell you when it's ready.

Olivier

YangZhiyu1994, 2018-05-07:
All right, thanks a lot.

closed 2018-05-07 by YangZhiyu1994

OlivierHartmann, 2018-05-23:
Hi,
BEC and BSC are now available in AFF3CT but in the **_development_** branch.
You can use them by simply set the channel type argument : --chn-type BSC
But warning, it is of course only available with the OOK modulation :  --mdm-type "OOK"
You can set the noise range with this new matlab style feature : -R "0.1:0.05:0.5,0.01:0.002:0.1"

You have examples in [error rate references repository](https://github.com/aff3ct/error_rate_references/tree/development), in development branch, too.

Think about updating [PyBER](https://github.com/aff3ct/PyBER), too. It has been upgraded to manage those different kind of noise (event probability, EB/N0, ES/N0, ...). Check the display menu to switch from a noise type to another as abscissa.

Olivier