aff3ct/aff3ct #27: How to convert an LDPC parity check matrix to the right format?

julialiao, 2018-06-09:
Hi,

Is there some tools or some guidelines to follow that I can convert my parity check matrix (QC-LDPC) to the *.alist format for the LDPC simulation?

Thank you!

kouchy, 2018-06-09:
Hi @julialiao,

By now there is no tool to convert QC-LDPC to ALIST format in AFF3CT.
But AFF3CT can directly read QC-LDPC format.
Here is a example of AFF3CT QC-LDPC compatible file : https://github.com/aff3ct/configuration_files/blob/master/dec/LDPC/AR4JA_4096_8192.qc.

I hope it helps.

Adrien

julialiao, 2018-06-09:
Hi,

Thanks for the reply. It's great to know that AFF3CT can directly read QC-LDPC format. 

In the example, the first row shows the number of column/row blocks and the submatrix size? 
And what does the last row "1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 0 0 0 0" stand for?

kouchy, 2018-06-09:
The last row stands for the puncturing pattern. You can avoid to put this line if you don't want to puncture your code.

julialiao, 2018-06-09:
I see. Thanks a lot!

closed 2018-06-09 by julialiao