aff3ct/aff3ct #29: SCMA for arbitrary codebooks

klimentyev, 2018-09-20:
I saw the implementation of the SCMA system in your project yesterday. It is very interesting for me. I wanna improve the system. First, use of arbitrary codebooks with size 4x6. In my C code, I calculate factor graph by codebooks: https://github.com/klimentyev/scma_mex/blob/master/src/scma_log_mpa.c#L13
I think in the future we need to extend the code for any size of codebooks (e.g, 8x12, 5x10, etc.).

P.S. In free time, I will try to implement here Log-MPA and Max-Log-MPA algorithms from my project: https://github.com/klimentyev/scma_mex/blob/master/src/scma_log_mpa.c

klimentyev, 2018-09-20:
I'm seeing aff3ct code. I think we can define a structure with codebooks (and re_user) in /src/Tools/Code/SCMA like for LDPC code parameters.
There are many different codebooks 4x6 size with different factor graphs.
I hope this is interesting to you.

OlivierHartmann, 2018-09-21:
Hi klimentyev!

Thank you very much for trying to contribute to AFF3CT. Of course, we would be grateful to see you improve SCMA and happy to have you as a contributor.

It would indeed be a good thing to be able to read codebooks from the files given in the arguments. In your previous request for extraction perhaps, could you give us your view of these codebook files? And if you don't mind, could you modify the Modem_SCMA class with your generic decoder. It would be perfect if it took as input argument the name of the file to be read. I can help you at any step.

Thanks a lot klimentyev.

Olivier

klimentyev, 2018-09-21:
Hi Olivier!

Thanks for your reply. Ok, I give files with 5 codebooks and their
description. Can you open and read .mat files from MATLAB? Also I can
convert files to plain text with arrays.

I checked my old BER results of simulation with your results for Altera
codebooks. They are the same :) I want to see Rayleigh generator in aff3ct
project and add all possible channels for simulation. Now, uplink Rayleigh
channel with diversity is simulated because the channel gains are different
on resource elements for each user. There are 4 Rayleigh channels: DL/UL
with and without diversity. The description from my MATLAB project:
https://github.com/klimentyev/scma/blob/master/README.md#channel-models
It would be good to realize these your project. Your implementation and GCC
7.3 compiling is faster than my project in C and Intel Compiler 2018.

Sincerely,
Vyacheslav Klimentyev

OlivierHartmann, 2018-09-24:
Hi Klimentyev,
Sorry we don't read matlab .mat files. A plain text file is preferable than a binary file, especially for small data quantity.
In the AFF3CT's Rayleigh channel you can simply add an argument to select one of your four models.
We can try to work together on your fork before merging it in the main project repo.
If you don't mind can you develop from the 'development' branch ? There are much more features but, of course, it is not as stable as can be the 'master' branch.
I'll help you to make connections between AFF3CT classes, but I'll let you develop your new features in it and validate them.
Thank you again for wanting to improve AFF3CT with us.
Olivier

klimentyev, 2018-09-24:
Hi Olivier,
The branch from my repo with codebooks and format description: https://github.com/klimentyev/configuration_files/tree/scma_cb_files
It's not PR, because it's only first step. I can send you PDF of papers from README.
In the future, we can extend it to other dimensions (6x4x4). I hope so.

I now recall the principles of OOP and C++ now because wrote code in C for ARM for GNSS signal processing last 5 years.

P.S. Klimentyev is my surname, but name is Vyacheslav :)

klimentyev, 2018-11-24:
We have ability to work with arbitrary regular graph SCMA codebooks for `number_of_users_per_resource = 3` and 3 bps (2 bps per user). Potentially we can work with another codebook size (user bps), SCMA implementation allows it (we need to check the Modem class).
For codebooks with `number_of_users_per_resource > 3` the additional nested loops are required or another data representation. I didn't realize decoders for case > 4 (I added one nested loop).

I think I can add Max-Log-MPA decoder for increase throughput, but later.

OlivierHartmann, 2018-11-26:
Hi Vyacheslav,
Thank you very much for your work.
So you mean you will need to change a lot of things to 4 users ?

OlivierHartmann, 2018-11-26:
Um...... I'll try to think about it. Don't have so much time these days.

klimentyev, 2018-11-26:
> Um...... I'll try to think about it. Don't have so much time these days.

OK. The example of large factor graph in eq. (25): https://ieeexplore.ieee.org/document/7248770
The usage of CS8 can help to reduce MPA complexity due to lower number of projection.
SCMA with `number_of_users_per_resource > 6` isn't good system in practice due to very high complexity of MPA.
There are many different algorithms for large codebooks and factor graphs (sphere decoding, using FFT like as in LDPC decoding, etc.). I don't now details of implementation.

Common codebooks have (4, 6) factor graph and 4...16 codewords.

OlivierHartmann, 2018-12-19:
Hi Vyacheslav,
Sorry didn't had time for now to think about it for now. But I have merged your work. So you can pull the last development version of AFF3CT to use it.
Thank you for your work and for developing AFF3CT with us.
Olivier

klimentyev, 2018-12-20:
Hi Olivier!
Thx for your work. I read in documenttion about restriction in 3 bps for SCMA. I'm closing the issue. Maybe I will make example `number_of_users_per_resource = 4`.
P.S. I'll talk about AFF3CT at local scientific conference in my Saint Petersburg Electrotechnical University in February.

closed 2018-12-20 by klimentyev