aff3ct/aff3ct #31: Code design

Ka-zam, 2018-11-06:
Hi, 

First of all thank you for this very comprehensive tool. Very nice work indeed. 

I have compiled it just fine and ran some of the provided simulations and some of my own. It worked nicely. 

What I'm trying to achieve is to design a code suitable for the following channel:
SNR ~40dB
Slowly, and I mean really slowly on the order of Hz, varying AWGN channel. There might be burst errors. 

Obviously this is strictly BW limited and considering my BW limit (strict physical limitation) and bit rate requirements I see myself needing to use 1024QAM which then lends itself naturally to a half-rate code with 2048QAM. I would probably place these 2048 symbols in a square lattice but remove the symbols furthest from the origon so that I can use well established symbol timing recovery algorithms but at the same time minimize constellation power.

I plan to use 8sps with 0.30 excess bandwidth RRC filtering. All standard stuff. 

En/decoding hardware is a bit restricted so I'm trying to avoid Turbo codes and perhaps use polar code as one of the papers mentioned they are more energy efficient. 

My initial thought was to use an interleaved and scrambled RS code but then I stumbled upon this and became fascinated. So, could you give me some pointers? 

I noticed for instance that if you select POLAR code the interleaving tab disappears in the CLI generator. It's not possible to use interleaving with POLAR codes? 

I also tried generating a .hpp decoder from the polar_decoder_gen repo. That worked fine. Is there a similar tool planned for the encoding step? I guess you just need to supply a symbol map for this to be feasible. 

Thanks!

OlivierHartmann, 2018-11-07:
Hi Ka-zam,

I didn't get everything of your message.

The RS first maps bits to symbols and then encodes them to get the codeword (with symbols and not bits). After modulation and demodulation, the RS decoder tries to correct the got frame to find back the codeword (with symbols), and then unmaps it to get back the bits. In AFF3CT you give the encoder's input and codeword sizes in number of symbols. You can find examples in the `refs` folder.

About interleaving, there is no interleaver in the standard BFER chain. You will have to use the iterative BFER chain (--sim-type BFERI) to find an interleaver between the encoder and the modulator. So there is a decoding loop between the demodulator and the decoder with of course deinterleaving/interleaving steps between. So you can use it with zero loop. The thing behind is that you can use only codes with soft output, as Polars or LDPC. See this image https://github.com/aff3ct/aff3ct/blob/master/doc/images/bfer_bferi.png, and take a look at those refs https://github.com/aff3ct/error_rate_references/tree/master/LDPC/AWGN/CPM

Polar encoder is generic, there's no need to generate one.

Hope this help.

Ka-zam, 2018-11-07:
Thanks, sorry fort being a little confused!

I will definitely look at the /refs and thank for bearing with me!

closed 2018-11-19 by kouchy