aff3ct/aff3ct #32: About compiler performance (throughput)

klimentyev, 2019-01-05:
Hi!

I make presesentation about AFF3CT for our local conference. I provided investigations regarding compiler performance.

My machine: Intel i7 4790k, 16 Gb RAM, Linux Gentoo. I compared GCC 7.3.0, Clang 6.0.1 and Intel Compiler 2018.3.

`CXX_FLAGS="-Ofast -funroll-loops -march=native"` for GCC/Clang
`CXX_FLAGS="-Ofast -funroll-loops -xHost"` for ICC

Yes, I checked results with `-Ofast`. They were correct.

1st experiment:

`$ aff3ct-2.2.0 --sim-type "BFER" -C "UNCODED" -K "12" -m "13" -M "13" -s "15" --chn-type AWGN --chn-implem FAST --mdm-type "SCMA" --mdm-ite "5" --mnt-max-fe "10000" --src-fra "6" --mdm-codebook conf/mod/SCMA/CS5.cb`

GCC/Clang/**ICC**: 1.77, 1.63, **3.76** Mb/s

It's not surprise for me. ICC makes good autovectorization. I saw .dasm for my projects.

2nd experiment:

`aff3ct-2.2.0 -C "POLAR" -K 1723 -N 2048 -m 4.0 -M 4.0 --mnt-max-fe "10000"`

**GCC**/Clang/ICC: **98.7**, 43.8, 57 Mb/s

In this case, MIPP is used. Am I right? Does MIPP use any GCC extensional features?

kouchy, 2019-01-06:
Hi @klimentyev,

It is very difficult to make assumptions on which is the faster compiler, it really depends on the compiler version, and the part of the code you execute.

It does not surprise me that in some cases ICC is faster than GCC and in other cases it is the opposite.

In the first experiment I suppose that ICC is able to optimize better the SCMA demodulate task than GCC and Clang.

In the second experiment, no MIPP is not used and no MIPP does not use any GCC extensional features. Generally, when a task is implemented with MIPP, the throughputs between the different compilers are very similar because the MIPP code is almost assembly code.

If you want to have a more precise idea about which task is faster on which compiler, you can use the `--sim-stats` cmd line argument (https://aff3ct.readthedocs.io/en/v2.2.0/user/simulation/parameters/simulation/simulation.html#sim-stats).

I hope it helps.

klimentyev, 2019-01-06:
Thank you for your answer! I will see dasm for polar decoder and SCMA demodulator sections.
P.S. I also see Intel Vtune profiler results.

closed 2019-01-06 by klimentyev