aff3ct/aff3ct #33: Invalid BCH polynomial

kelvinlayton, 2019-01-16:
Hi,
Firstly, great project!

I tried a BCH code with K=64 and N=127.
`./bin/aff3ct -C "BCH" -K 64 -N 127 -m 1.0 -M 4.0 -s 1.0`
But it generates error
`(EE) In the 'aff3ct/src/Module/Encoder/BCH/Encoder_BCH.cpp' file at line 23 ('Encoder_BCH' function): "'N - K' is different than 'n_rdncy' ('K' = 64, 'N' = 127, 'n_rdncy' = 56, 'N - K' = 63)."`

I think this is a valid combination. Setting T= 10 instead with
`./bin/aff3ct -C "BCH" -T 10 -N 127 -m 1.0 -M 4.0 -s 1` 
automatically selects K=64.

I'm guessing this is a bug?

kouchy, 2019-01-16:
Hi @kelvinlayton,

Yes you're right, this is a bug in AFF3CT. We naively try to guess the value of ``T`` from ``N``, ``K`` and ``m`` (see https://github.com/aff3ct/aff3ct/blob/development/src/Factory/Module/Decoder/BCH/Decoder_BCH.cpp#L74) and this can be wrong.

I will remove this line and put the ``--dec-corr-pow, -T`` as a required argument.

Thank you.

kouchy, 2019-01-16:
Hi again,

I tried to perform the previous modification but it has a negative impact on the Turbo Product Code (TPC) family which uses the BCH by defining only `K` and `N`.

In many cases, the following assumption is true : `T = (N - K) / m` with `m` the Galois field order and `T` the BCH correction power. This is why I prefer to keep the code like it is.

To avoid possible errors, I encourage you to do not give `K` from the command line and only use `N` and `T` when dealing with BCH code family.

I hope it helps.

kelvinlayton, 2019-01-17:
Thanks for looking into this. It would be good to have `T` as required only when code `C` is `"BCH"`, but I assume this is difficult with the current architecture.

Specifying `T` is a fine workaround for my needs, so don't worry too much.

closed 2019-02-11 by kouchy