aff3ct/aff3ct #37: Turbo with 1/2 code rate

igorauad, 2019-01-28:
Hello all,

I was wondering how one could use 1/2 code rate with the Turbo Encoder. The reason why I’m asking is because of line 31 in the following file:

https://github.com/aff3ct/aff3ct/blob/master/src/Module/Encoder/Turbo/Encoder_turbo.cpp#L31

Is it possible to use 1/2 code rate? Am I just misinterpreting this line?

Many thanks

bonben, 2019-01-28:
Hi Igor.

Yes, it is possible to use a 1/2 code rate with the turbo encoder.
Please take a look at the puncturer documentation : 
https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/turbo/puncturer.html

For example, a (4108,2048) turbo code reference is available in our references and in the BER / FER comparator : 
http://aff3ct.github.io/comparator.html?right=null&left=TURBO%2FAWGN%2FBPSK%2FRSC_RSC%2FBCJR%2FLTE%2FTurbo_N4096_K2048_BCJR_i8_ref_LTE.txt

With the corresponding command line : 
```
aff3ct --sim-type BFER -C TURBO -K 2048 -m 0.0 -M 2.0 -s 0.1 --enc-sub-std LTE -i 8 --dec-sf-type LTE_VEC --dec-sub-implem FAST --dec-sub-max MAX --pct-pattern "11,10,01" --crc-implem FAST --crc-poly 24-LTEA --dec-fnc --dec-fnc-ite-m 5
```

Hope it helps :wink:

igorauad, 2019-01-28:
Absolutely helps. Many thanks, @bonben 

I believe you can close this question, as the answer is pretty clear (to use the puncturer). The puncturer removes 3 bits after every 9, meaning it reduces the codeword length to `2/3` of its original length. If code rate is `1/3`, we have a resulting code rate given by `1 / ( 3 * (2/3) ) = 1 / 2`.

Thanks =)

closed 2019-01-29 by kouchy