aff3ct/aff3ct #39: Creating multiple decoders in a program

jianding17, 2019-02-05:
I am using aff3ct as a library based on the script under my_project_with_aff3ct/examples/bootstrap. Single thread works fine, but when I try to create a multi-threading application and distribute multiple decoders to different threads, I always get segmentation fault. 

Is it possible to get multiple decoders work on different threads? If possible, what is the correct way to do it?

kouchy, 2019-02-05:
Hi @jianding17,

The AFF3CT module objects have not been designed to be used simultaneously by multiple threads.
If you want to use multiple threads you will have to allocate multiple module objects.

I hope it helps.

jianding17, 2019-02-05:
Thanks for the fast response. 

I created multiple objects but still always get segmentation faults. I am using the SIMD version of BP_Flooding. Part of the error message is attached below. 
`0x000000000044067b in aff3ct::module::Decoder_LDPC_BP_flooding_inter<int, float, aff3ct::tools::Update_rule_NMS_simd<float, 0> >::_decode_single_ite (msg_chk_to_var=std::vector of length 1824, capacity 1824 = {...}, 
    msg_var_to_chk=std::vector of length 1824, capacity 1824 = {...}, this=0x7ffee8000970)
    at my_project_with_aff3ct/lib/aff3ct/build/install/include/aff3ct-2.2.1/aff3ct/Module/Decoder/LDPC/BP/Flooding/Decoder_LDPC_BP_flooding_inter.hxx:277
277			for (auto v = 0; v < chk_degree; v++)`

kouchy, 2019-02-05:
Oh, it's completely different from what I imagined. It's a bug I think. Actually this code should not be compiled and available to you if you do not have a C++ compiler that support the _dynamic memory allocation for over-aligned data_ feature from the C++17 standard. If you are using the GNU compiler version 8 you can compile AFF3CT with the `-faligned_new` option (as explained here: https://aff3ct.readthedocs.io/en/v2.2.2/user/simulation/parameters/codec/ldpc/decoder.html#dec-implem).

If you do not have a compiler that support the _aligned new_ feature, I suggest you to use those decoders instead:

https://github.com/aff3ct/aff3ct/blob/fcd41ffd04238e314b48ec5d391265e1407975a8/src/Factory/Module/Decoder/LDPC/Decoder_LDPC.cpp#L304-L306

I hope it helps.

jianding17, 2019-02-06:
I replaced the function as you suggested. It works now. Thanks for your help.

closed 2019-02-06 by jianding17

kouchy, 2019-02-06:
I fixed the bug in this commit: https://github.com/aff3ct/aff3ct/commit/0ae46a11af1fb84b5cb93306d492cba1915de7c8.

Btw: if you want to use the full potential of the previous proposed decoder you have to give many frames to the decoder to exploit inter frame SIMD parallelism. This will highly increase the throughput per thread.