aff3ct/aff3ct #40: Polar codes encoder and decoder

Mfn12, 2019-02-10:
Could you please help me to implement the polar codes correctly?
[main.txt](https://github.com/aff3ct/aff3ct/files/2848867/main.txt)

I have written a C++ code to call the polar encoder and decoder functions of AFF3CT. Unfortunately, the decoded bits that I am getting are not the same as the information bits. I have attached the main file of my code.
I got the following
(Information_bits, Decoded_bits):
(1, -2147483648)
(1, -2147483648)
(0, 0)
(1, -2147483648)
(0, 0)
(0, -2147483648)
(0, -2147483648)
(0, 0)
(0, 0)
(1, 0)
(1, 0)
(1, 0)
(1, -2147483648)
(1, 0)
(0, -2147483648)
(1, 0)
(0, -2147483648)
(0, -2147483648)
(1, -2147483648)
(0, -2147483648)
.
.
.

kouchy, 2019-02-10:
Hi @Mfn12,

I think this is because the decoder supposes a systematic encoding.
You should replace `aff3ct::module::Encoder_polar` by `aff3ct::module::Encoder_polar_sys`.

I hope it helps.

Mfn12, 2019-02-12:
Thank you for your reppy.
I have replaced the Encoder_polar by Encoder_polar_sys, and I am still getting the same results mentioned above. Could you please guide me for what can be done ?.

kouchy, 2019-02-12:
Hi @Mfn12,

If you want us to help you more, I recommend you to  make a fork of the [my_project_with_aff3ct](https://github.com/aff3ct/my_project_with_aff3ct) repo and to modify the [bootstrap](https://github.com/aff3ct/my_project_with_aff3ct/tree/master/examples/bootstrap) example as you wish. Then you can show us the code and we will be able to easily compile and test it.

Mfn12, 2019-02-15:
Hello, 
When I replaced in my code the decoder: "Decoder_polar_SCL_fast_sys" by "Decoder_polar_SCL_naive", I got correct result. However, I am still not understanding why I am getting correct results with the naive decoder and incorrect ones when I am using the  fast_sys decoder.

Mfn12, 2019-02-17:
> Hi @Mfn12,
>
> If you want us to help you more, I recommend you to make a fork of the [my_project_with_aff3ct](https://github.com/aff3ct/my_project_with_aff3ct) repo and to modify the [...]

Hello 
I would like to thank you for your help. Actually, I went to the link of "my_project_with_aff3ct", then I did the following:
1- I pressed on the fork icon.
2- I went to the : "examples/bootstrap/src/main.cpp", and I opened the main.cpp file.
3- I wrote my code in the main.cpp file.
Now, I don't know what I should do to let you see my modifications to the main.cpp file.
Could you please let me know what are the next steps ?

Thank you
 
Hi @Mfn12,

If you want us to help you more, I recommend you to make a fork of the my_project_with_aff3ct repo and to modify the bootstrap example as you wish. Then you can show us the code and we will be able to easily compile and test it.

closed 2019-04-12 by kouchy

Aishusiva, 2020-12-07:
@Mfn12 ,
Could you please send the modifications you made in the code to implement polar encoder and decoder.