aff3ct/aff3ct #42: How do I use a custom interleaver with DVB-RCS2 turbocode?

dsorber, 2019-09-30:
I am attempting to simulate using a custom interleaver with a DVB-RCS2 turbocode:

```
./aff3ct -C TURBO_DB --enc-sub-std DVB-RCS2 --dec-type TURBO_DB --itl_type USER --itl_path N_1504_int_LUT.txt -K 1504 -m 1.0 -M 3.0 -s 1 
```
But I get these errors:

> (WW) Unknown argument "--itl_type" at the position 13.
> (WW) Unknown argument "USER" at the position 14.
> (WW) Unknown argument "--itl_path" at the position 15.
> (WW) Unknown argument "N_1504_int_LUT.txt" at the position 16.

I can't seem to figure out the correct syntax. Please help.

ttonnellier, 2019-09-30:
Hello dsorber,
Replace the underscore (_) by the dash (-) in your parameters.
It should work after that.
Thibaud

dsorber, 2019-09-30:
Whoops, simple mistake. Thanks!

closed 2019-09-30 by dsorber