aff3ct/aff3ct #43: Issue with DVB-RCS2 decoder when using aff3ct library

dsorber, 2019-10-03:
I am interested in using the DVB-RCS2 turbo code from the library form of Aff3ct.  I've got the encoder working but I'm having some issues with the decoder.

Figuring that I've done something wrong in my code, I moved to using the bootstrap example from here: https://github.com/aff3ct/my_project_with_aff3ct/blob/master/examples/bootstrap/src/main.cpp. I modified it to use the DVB-RCS2 turbo code:

```
#include <iostream>
#include <memory>
#include <vector>
#include <string>

#include <aff3ct.hpp>
using namespace aff3ct;

template<typename T>
void print_vec(const std::string& label, T& data)
{
	std::cout << label << std::endl;
	for (auto idx = 0; idx < data.size(); ++idx)
	{
		std::cout << data[idx] << " ";
	}
	std::cout << "\n" << std::endl;
}

struct params
{
	int   K         =  408;     // number of information bits
	int   N         = (K * 3);     // codeword size
	int   fe        =  100;     // number of frame errors
	int   seed      =    0;     // PRNG seed for the AWGN channel
	float ebn0_min  =   0.00f; // minimum SNR value
	float ebn0_max  =  10.01f; // maximum SNR value
	float ebn0_step =   1.00f; // SNR step
	float R;                   // code rate (R=K/N)
};
void init_params(params &p);

struct modules
{
	std::unique_ptr<module::Source_random<>>          source;
	// Encoder
	//~ std::unique_ptr<module::Encoder_repetition_sys<>> encoder;
	std::unique_ptr<module::Encoder_turbo_DB<>> 	  encoder;
	
	std::unique_ptr<module::Modem_BPSK<>>             modem;
	
	// Channel
	//~ std::unique_ptr<module::Channel_AWGN_LLR<>>       channel;
	std::unique_ptr<module::Channel_NO<>>          	  channel;
	
	// Decoder
	//~ std::unique_ptr<module::Decoder_repetition_std<>> decoder;
	std::unique_ptr<module::Decoder_turbo_DB<>>    	  decoder;	
	
	std::unique_ptr<module::Monitor_BFER<>>           monitor;
};
void init_modules(const params &p, modules &m);

struct buffers
{
	std::vector<int  > ref_bits;
	std::vector<int  > enc_bits;
	std::vector<float> symbols;
	std::vector<float> noisy_symbols;
	std::vector<float> LLRs;
	std::vector<int  > dec_bits;
};
void init_buffers(const params &p, buffers &b);

struct utils
[... 158 lines cut ...]
	b.symbols       = std::vector<float>(p.N);
	b.noisy_symbols = std::vector<float>(p.N);
	b.LLRs          = std::vector<float>(p.N);
	b.dec_bits      = std::vector<int  >(p.K);
}

void init_utils(const modules &m, utils &u)
{
	// create a sigma noise type
	u.noise = std::unique_ptr<tools::Sigma<>>(new tools::Sigma<>());
	// report the noise values (Es/N0 and Eb/N0)
	u.reporters.push_back(std::unique_ptr<tools::Reporter>(new tools::Reporter_noise<>(*u.noise)));
	// report the bit/frame error rates
	u.reporters.push_back(std::unique_ptr<tools::Reporter>(new tools::Reporter_BFER<>(*m.monitor)));
	// report the simulation throughputs
	u.reporters.push_back(std::unique_ptr<tools::Reporter>(new tools::Reporter_throughput<>(*m.monitor)));
	// create a terminal that will display the collected data from the reporters
	u.terminal = std::unique_ptr<tools::Terminal_std>(new tools::Terminal_std(u.reporters));
}
```
I'm still seeing the same problems with the decoder output.  I also replaced the Channel with the "NO" version (i.e. passthrough) but that doesn't seem to change much.  The output I see (with my added debug) is here: https://pastebin.com/j6J4rdX0.

Can someone help me figure out what I'm doing wrong?  My guess is that I'm doing something wrong with the setup of the decoder, but I'm not sure. 

Any help would be appreciated.

kouchy, 2019-10-04:
Hi @dsorber,

You have to call the `init()` method to build the LUTs after allocating the `Interleaver_core`.
After line 181 (`auto interleaver_core = factory::Interleaver_core::build(itl_params);`) you have to add this line : `interleaver_core->init();` and after line 202 (`auto de_interleaver_core = factory::Interleaver_core::build(itl_params);`) you have to add this line : `de_interleaver_core->init();` then it should work.

For your information you can use the same `Interleaver_core` for the encoder and the decoder.
You can also take a look on the more advance example "Factory" (https://github.com/aff3ct/my_project_with_aff3ct/blob/master/examples/factory/src/main.cpp), it is very easy to change the code type, you just have to replace `Codec_repetition` by `Codec_turbo_DB` in the source code.

I hope it helps.

dsorber, 2019-10-04:
@kouchy Thanks very much! Adding the `init()` calls for the interleaver core fixes the problem I was seeing. I knew there was something simple I was missing.

By the way AFF3CT is awesome!!! I find it very, very useful.  

Thanks again for your help.

closed 2019-10-04 by dsorber

dsorber, 2019-10-16:
@kouchy Thanks, that sounds great.