aff3ct/aff3ct #45: AFF3CT- Unable to install packages

amirajnigam, 2019-11-01:
I am not able to run this command "sudo add-apt-repository ppa:aff3ct/aff3cyt-dev" successfully and facing this error " The repository 'http://ppa.launchpad.net/aff3ct/aff3ct-dev/ubuntu disco Release' does not have a Release file." and I am using Ubuntu version 19.04.

Could you please specify for which Ubuntu version this library will work?

bonben, 2019-11-01:
Hi @amirajnigam. 
AFF3CT apt packages are provided for LTS Ubuntu releases only. To pick up a package from a different release (e.g. 18.04) you should use pinning as it is detailed here : https://help.ubuntu.com/community/PinningHowto.
Let us know if you're experiencing troubles with this method. 
BR.

ibarron9216, 2020-09-22:
Hi, I am also having trouble installing the Ubuntu packages. I am running POP!_OS 20.04 LTS and when I run `sudo add-apt-repository ppa:aff3ct/aff3ct-stable` I get the following error `Error: retrieving gpg key timed out.`. I have already checked that I am able to add other ppas without trouble. Is there any other way to get the Ubuntu packages installed?

bonben, 2020-09-22:
Hi @ibarron9216. 

- We will not officially support other distributions than Ubuntu LTSs. Nevertheless, as POP!_OS seems to be based on Ubuntu, AFF3CT apt packages should be accessible.
- Ubuntu 20.04 AFF3CT packages are not yet deployed, but they will be in a near future. This may solve your problem.
- Other solutions are possible until then : Pinning, as I described earlier, if you really want packages, or other installation options : https://aff3ct.readthedocs.io/en/latest/user/installation/installation/installation.html.

BR.

ibarron9216, 2020-09-22:
Hi @bonben 
Thank you for your quick response. I have opted to compile the library myself and I encountered no issues. However, the very helpful information that you provided is not present in the installation documentation. Do you think it would be a good idea to add the fact that only LTS versions are supported and to mention the Pinning alternative? Once again thanks for the help

bonben, 2020-09-23:
I think it would be a good idea indeed.
It will be added in future versions. 

Thanks for your feedback !