aff3ct/aff3ct #47: Run Time

bencampionmax, 2019-11-19:
Hello,

I just want to start off saying that this toolbox is incredible. A big thank you to you and the rest of your team for making this available. 
Currently, we are running the command line: 
aff3ct -C "POLAR" -K 1723 -N 2048 -m 1.0 -M 4.0 -s 1.0

On a 24-core, 48-thread Hydra Desktop. On the Eb/N0 = 4.00 dB run the ET/RT is at 02h39'15. Is this to be expected or am I compiling and running the code incorrectly?

bencampionmax, 2019-11-19:
Is there a way to run an "optimized" version rather than a debug version?

ttonnellier, 2019-11-20:
Hello,
How did you generate the makefile?
[Check here](https://aff3ct.readthedocs.io/en/latest/user/installation/compilation/compilation.html#id4) for the documentation.
Best

bencampionmax, 2019-11-20:
Hello,

That link is how I generated the makefile. Then I just ran the executable
in the build/bin folder (aff3ct).

Best Regards,
--
Ben Campion
Applications Engineer
MaXentric Technologies, LLC.
737 Pearl Street, Suite 208
La Jolla, CA 92037
Direct: (612) 242-9189
bcampion@maxentric.com <johana@maxentric.com>

ttonnellier, 2019-11-20:
So it should be fine :)
You can use [the comparator](http://aff3ct.github.io/comparator.html) to compare your results

ttonnellier, 2019-11-20:
Wait, no. 
Your results are weird...

With the command `./bin/aff3ct -C "POLAR" -K 1723 -N 2048 -m 1.0 -M 4.0 -s 1.0 -t 1` (same as yours but only on one core), result is almost instantaneous...

bencampionmax, 2019-11-20:
Ok then I will rm -r build and re-do the generate makefile steps. Thank you
for following up.

Best Regards,
--
Ben Campion
Applications Engineer
MaXentric Technologies, LLC.
737 Pearl Street, Suite 208
La Jolla, CA 92037
Direct: (612) 242-9189
bcampion@maxentric.com <johana@maxentric.com>

closed 2019-11-27 by bencampionmax