aff3ct/aff3ct #49: Polar Encoding and Decoding

bencampionmax, 2019-11-23:
Hello,

I was looking through the source code and I wanted to make sure that I am reading the code correctly. The polar encoding labeled "TV" is not using Prof. Vardy's upgrading and degrading method but instead just calling a previously generated codeword based off of various parameters. As for the decoding (SCL), it is also just calling previously generated decoding matrices. 
Is this correct?

bonben, 2019-11-25:
Hi @bencampionmax.

Yes you're right. The previously generated frozen bits set can be found here : https://github.com/aff3ct/configuration_files/tree/master/cde/awgn_polar_codes/TV.

By the way you can also use the Gaussian Approximation generation method with the following argument : `--enc-fb-gen-method GA`. Corresponding documentation can be found here : https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/polar/encoder.html#enc-fb-gen-method.

Regards.

bencampionmax, 2019-11-25:
Thank you for letting me know.

closed 2019-11-25 by bencampionmax

bonben, 2019-11-25:
Hi again @bencampionmax.

I just wanted to make a point of clarification regarding your issue. I misread a bit your message, and I missed the fact that you implied that the generated frozen bits sets had not been created with Pr. Tal and Pr. Vardy's method.

It is not the case. It is really their method that has been used to generate all the sets that are listed here https://github.com/aff3ct/configuration_files/tree/master/cde/awgn_polar_codes/TV.

Regards.