aff3ct/aff3ct #50: Comparing Turbo, LDPC, and Polar Codes

bencampionmax, 2019-12-03:
Hello,

I'm currently trying to compare the performance of polar, turbo, and LDPC with one another with a fixed block length of 1024 and the following code rates: 1/2, 1/16, and 1/256. 
The polar codes seem to have the flexibility to make this happen, but it doesn't look like the turbo and LDPC codes can. Is this right? If so, do you know of any alternative coding suites that could work for my needs?

Thank you for your help.

bencampionmax, 2019-12-11:
Hello,

We are currently working with Prof. Vardy, and we would really appreciate it if we code talk with you about editing your source code over the phone. However, due to the time difference email is fine too.

We are trying to edit the source code. We are trying to add a repetition code as a "sub-encoder" (and decoder) in order to achieve lower rates for the Turbo and LDPC codes. Additionally, we will use the repetition sub-encoder in with Polar Codes to compare the performance of the polar codes without repetition and polar codes with repetition (both of overall equal rate). 

While looking through the code we believed that the easiest way to achieve this was to write this repetition encoding and decoding in the "Channel" to avoid needing to edit everything outside (i.e. Modem and Codec). However, we are having difficulties understanding where in the code the LLRs are calculated. 
We think it is calculated in Channel_AWGN_LLR.cpp
	else
	{
		const auto f_start = (frame_id < 0) ? 0 : frame_id % this->n_frames;
		const auto f_stop  = (frame_id < 0) ? this->n_frames : f_start +1;

		if (frame_id < 0)
			noise_generator->generate(this->noise, this->n->get_noise());
		else
			noise_generator->generate(this->noise.data() + f_start * this->N, this->N, this->n->get_noise());

		for (auto f = f_start; f < f_stop; f++)
			for (auto n = 0; n < this->N; n++)
				Y_N[f * this->N +n] = X_N[f * this->N +n] + this->noise[f * this->N +n];
	}
}

Where can we find noise.data()? Or are we on the wrong path?

kouchy, 2019-12-12:
Hi @bencampionmax,

First thing that comes into my mind is that there is a repetition code already implemented in AFF3CT, why not use/improve this one? Second thing is that you want a sub-encoder, we already managed this specific case with RSC codes to build a turbo encoder/decoder, perhaps you can find some inspiration looking at the turbo encoder/decoder?

For me it is unclear why you want to modify the channel. For what I understand you want to define a repetition code over generic sub-codes, and to do that I don't think that modifying the AWGN channel is a generic and easily reusable approach :-).

In the example you took, `noise` is a `std::vector` of size `N` and it contains the white Gaussian noise. The LLR[n] is the combination of X_N[n] (-1 or 1) + noise[n].

I hope it helps.

bencampionmax, 2019-12-12:
Great! Thank you!

closed 2019-12-12 by bencampionmax