aff3ct/aff3ct #52: AFF3CT TurboCode LTE wrong encoding format

DavideAchilli, 2020-01-13:
### TEST ENVIRONMENT

The version information on AFF3CT are:

aff3ct (Windows 64-bit, MSVC++-1924, NO_INTRINSICS) v2.3.5
Compilation options:
  - Precision:         8/16/32/64-bit
  - Polar bit packing: on
  - Polar bounds:      off
  - Terminal colors:   on
  - Backtrace:         off
  - External strings:  on
  - MPI:               off
  - GSL:               off
  - MKL:               off
  - SystemC:           off

We are testing the library with these parameters:

--sim-type "BFER" -C "TURBO" -e "200" -m "0.0" -M "0.0" -s "0.125" -t 1 --src-type USER --src-path runtest1.dat -K 48 --sim-dbg --sim-dbg-fra 1 --sim-stop-time "1" --enc-sub-poly "{013,015}" --itl-type "LTE"

The "runtest1.dat" file contains:

```
1
48
1 1 1 1 1 1 1 1 0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

```
### ISSUE DESCRIPTION
We compared the encoding and decoding of the AFF3CT library with a currently working implementation.
We expect our 48-bit sequence to be encoded as a 48*3+12=156 bit sequence formed in this way:

- [A] 48 bit of the original content
- [B] 4 tail bits
- [C] 48 bit of parity
- [D] 4 tail bits
- [E] 48 bit of interleaved parity
- [F] 4 tail bits

The AFF3CT library produces the [A], [C] and [E] parts correctly, but then assembles the frame as following:

- [A] 48 bit of the original content
- [G] 3 tail bits
- [C] 48 bit of parity
- [H] 6 tail bits
- [E] 48 bit of interleaved parity
- [I] 3 tail bits

In other words, the 12 tail bits are not distributed evenly 4-4-4, but 3-6-3. The IET site (https://communities.theiet.org/blogs/426/424) confirms "the sequence of tail bits is rearranged and 4 tail bits are attached after each information bit stream".
While this frame format is correctly decoded by AFF3CT itself, this makes it not compatible with the standard frames.

We inspected the code and identified the following method:

```
void Encoder_turbo::_encode(const int8_t *inputData, int8_t *outputData)
{
    // Executes the interleave of "inputData" and saves it in "tempInterleaved".
    // The "tempInterleaved" string is ok (identical to the reference implementation).
    // inputData.........: 111111110000000000000000000000000000000000000000
    // tempInterleaved...: 100000000000011000010000000010000100000000100001
    pi.interleave(inputData, tempInterleaved); // [A]

    // Executes the the encode of "inputData" on "tempData". This function produces
    // on tempData:
    // - [1] the original string: 111111110000000000000000000000000000000000000000
    // - [2] 3 tail bits: 010
    // - [3] itsarity string: 101000111110010111001011100101110010111001011100
    // - [4] 3 tail bits: 001
    // The parity string [3] is ok (identical to the reference implementation).
    // The number of tail bits per string (3) is wrong (should be 4).
    enco_n.encode(inputData, tempData); // [B]

    // Method "enco_n.get_N()" is 102, so it copies the two 48-bit strings (48*2=96)
    // and the two pairs of 3 tail bits (3*2=6, 96+6=102).
    memcpy(outputData, tempData, enco_n.get_N()); // [C]

    // Here it performs the same encoding "enco_n.encode" used before, but using the
    // the interleaved data it generated in "tempInterleaved"
    // This action executes the same code of "enco_n.encode" and produces:
    // - [5] the original interleaved: 100000000000011000010000000010000100000000100001
    // - [6] 3 tail bits.............: 111
    // - [7] its parity string.......: 111100101110000110001001011111011011001011011000
    // - [8] 3 tail bits.............: 001
    // The parity string [7] is ok (identical to the reference implementation).
    enco_i.encode(tempInterleaved, tempData); // [D]

    // Now we have:
    // enco_n.get_N()=102
    // enco_i.get_N()=102
    // enco_i.get_K()=48
    // The following code copies at position 102 of output (i.e. after [1], [2], [3] e [4])
    // the data that begin at "getK", i.e. [6], [7] e [8].
    memcpy(outputData+enco_n.get_N(), tempData+enco_i.get_K(), enco_i.get_N()-enco_i.get_K());  // [E]
}

```
We would expected the encoder to discard the first half of the interleaved block (i.e. the unchanged interleaved data and their tail bits [5] and [6]) to keep only the parity of the interleaved and its tail bits [7] and [8] but instead it takes also the tail bits [6].
Since the tail bits are 3-bits long, taking an extra set of tail bits brings the total back to 12 as expected, but the resulting format is not compatible.
### ANALYSIS OF THE CODE
We examined the code but it seems that the way it is written, it can not code correctly no matter of the external parameters.
Let's suppose we want to use the same code to have tail bit sequences 4-bit long as requested by the specification.
The current code has "enco_n==enco_i", i.e. they are the same instance.
The encode function [B] should produce 48+4+48+4=104 bit instead of 102; thus the memcpy [C] should get 104 instead of 102 from "enco_n.get_N()".
The encode function [D] (which is the same code of [B]) should also produce 104 bits, of which we should keep the last 52 (48-bit interleaved parity and 4 tail bits).
The "memcpy" at [E] begins to copy at "outputData+enco_n.get_N()", that would be correct (outputData+104); however, it takes its input data coded by [D] starting from "enco_i.get_K()", i.e. at 48 and not 52, and it would copy "enco_i.get_N()-enco_i.get_K()", i.e. 104-48=56 and not 52.
The final result would be of 104+56=160 which is not the expected 156.

In other words, the code here implemented, always inserts in the encoded sequence the tail bits of the original interleaved data [6] that is not expected to be part of the final frame.

Best regards
Davide Achilli - DAFOCUS - http://www.dafocus.com/

DavideAchilli, 2020-02-05:
Hello.

We have further investigated this issue.
The four shift registers Xk, Zk, X'k e Z'k (see TS25.222, chapter 4.2.3.2.1) produce three tail bits each that we shall call [A0,A1,A2], [B0,B1,B2], [C0,C1,C2] and [D0,D1,D2].
The AFF3CT library arranges them in this way:

`<systematic> A0 A1 A2 <parity> B0 B1 B2 C0 C1 C2 <interleaved> D0 D1 D2`

However, the arrangment used by LTE is:

`<systematic> A0 B1 C0 D1 <parity> B0 A2 D0 C2 <interleaved> A1 B2 C1 D2`

Best regards
Davide Achilli - DAFOCUS - http://www.dafocus.com/

kouchy, 2020-02-10:
Ok thank you, I will check that ASAP. I don't know if you've seen the documentation but we've explained that here : https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/turbo/encoder.html#enc-sub-no-buff. Perhaps with the `--enc-sub-no-buff` option you will have an ordering closer to the LTE standard.

DavideAchilli, 2020-02-10:
Hello Adrien.

Thanks for your suggestion. However, we did try the --enc-sub-no-buff
option but abandoned it because even the systematic bits in the encoded
part appeared messed up.

Check this out. In red I have evidenced the systematic bits in the input
(U_K) and in the encoded data. This is ok:

WITHOUT PARAMETER --enc-sub-no-buff

# Source_user::generate(int32 U_K[48])
# {OUT} U_K = [1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0]
# Returned status: 0
#
# Encoder_turbo::encode(const int32 U_K[48], int32 X_N[156])
# {IN}  U_K = [1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0]
# {OUT} X_N = [1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 1, 0, 1, 0, 1, 0, 0, 0, 1, 1, 1, 1, 1, 0, 0, 1, 0, 1,
1, 1, 0, 0, 1, 0, 1, 1, 1, 0, 0, 1, 0, 1, 1, 1, 0, 0, 1, 0, 1, 1, 1, 0,
0, 1, 0, 1, 1, 1, 0, 0, 1, 1, 0, 1, 1, 1, 1, 1, 1, 1, 0, 0, 1, 0, 1, 1,
1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 1, 1, 1, 0, 1, 1, 0,
1, 1, 0, 0, 1, 0, 1, 1, 0, 1, 1, 0, 0, 0, 0, 0, 1]

If we activate the --enc-sub-no-buffoption, the first 48 bits of X_N do
not match the first 48 bits of U_K anymore.

WITH PARAMETER --enc-sub-no-buff
   
# Source_user::generate(int32 U_K[48])
# {OUT} U_K = [1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0]
# Returned status: 0
#
# Encoder_turbo_legacy::encode(const int32 U_K[48], int32 X_N[156])
# {IN}  U_K = [1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0]
# {OUT} X_N = [1, 1, 1, 1, 0, 1, 1, 1, 1, 1, 0, 1, 1, 0, 0, 1, 0, 0, 1,
1, 1, 1, 1, 0, 0, 1, 1, 0, 1, 1, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0,
0, 0, 0, 1, 1, 0, 1, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0,
1, 0, 0, 1, 1, 0, 1, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 1, 0, 1, 1, 0,
1, 0, 0, 1, 1, 0, 0, 1, 0, 0, 0, 0, 1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0,
1, 1, 0, 0, 0, 0, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 1, 1, 0, 1, 0, 0,
0, 0, 0, 0, 0, 0, 1, 1, 1, 0, 0, 1, 0, 1, 0, 1, 1]

Thanks and best regards
Davide