aff3ct/aff3ct #53: Issue with LDPC decoders

Zlika, 2020-01-17:
Hi!
I have some problems using the LDPC decoders, the decoded bits are wrong.
I tested the following small test program on both aff3ct-2.3.5 and aff3ct-2.3.5-272-gda6f24b.
The encoded bits are correct (I have the same results using GNU Octave by multiplying ref_bits with G).
For the decoding, I tried both Decoder_LDPC_bit_flipping_OMWBF and Decoder_LDPC_BP_flooding_SPA, but they give me the same (wrong) result:
* with decoder->decode_siho(enc_bits_float, dec_bits), dec_bits is full of zeros,
* with decoder->decode_siso(enc_bits_float, dec_bits), dec_bits is the bitwise inversion of enc_bits (if enc_bits includes bit errors, these bits are also inverted in dec_bits, so no error correction is done even if we use ~dec_bits).
G and H matrices are correct because when I compute the syndrome with GNU Octave (H*enc_bits') I have vector full of zeros.
Could you tell me what is wrong with my test program?
Thanks for your help.

```
int main(int argc, char** argv)
{
	// Read H matrix (AList format)
	std::ifstream h_file;
	h_file.open("AR4JA_r23_k4096n.a");
	Sparse_matrix H = tools::AList::read(h_file);
	h_file.close();

	// Info bits are the first 4096 bits of the encoded codeword
	std::vector<unsigned> read_info_bits_pos(4096);
	std::iota(read_info_bits_pos.begin(), read_info_bits_pos.end(), 0);

	// Read G matrix
	std::ifstream g_file;
	g_file.open("gAR4JA_r23_k4096.txt");
	// Custom function to read a full matrix from a text file
	Sparse_matrix G = read_matrix(g_file, 4096, 7168);
	G = G.transpose();
	g_file.close();

	auto encoder = std::unique_ptr<module::Encoder_LDPC<>>(
						new module::Encoder_LDPC<>(4096, 7168, G));
	// Either decoders give the same (wrong) results
	//auto decoder = std::unique_ptr<module::Decoder_LDPC_bit_flipping_OMWBF<>>(
	//				new module::Decoder_LDPC_bit_flipping_OMWBF<>(4096, 7168, 30, H, read_info_bits_pos));
	auto decoder = std::unique_ptr<module::Decoder_LDPC_BP_flooding_SPA<>>(
							new module::Decoder_LDPC_BP_flooding_SPA<>(4096, 7168, 30, H, read_info_bits_pos));

	// Test vector full of 1s with a few 0s are the beginning
	std::vector<int> ref_bits(4096, 1);
	ref_bits[1] = 0;
	ref_bits[4] = 0;
	ref_bits[5] = 0;
	ref_bits[8] = 0;
	std::vector<int> enc_bits(7168, 0);

	// enc_bits is consistent with the result I have using GNU Octave (enc_bits=mod(ref_bits*G, 2))
	encoder->encode(ref_bits, enc_bits);
	// Convert to float vector for the decoder
	std::vector<float> enc_bits_float(enc_bits.begin(), enc_bits.end());

	// SIHO decoder returns a vector full of 0s
	// SISO decoder returns ~enc_bits
	std::vector<int> dec_bits(4096, 0);
	decoder->decode_siho(enc_bits_float, dec_bits);
	//decoder->decode_siso(enc_bits_float, dec_bits);
}
```

Zlika, 2020-02-10:
I was finally able to make it work: soft-input decoders do not work with 0s/1s but with positive/negative numbers, e.g. mapping 0->+1.0 and 1->-1.0.

closed 2020-02-10 by Zlika

kouchy, 2020-02-10:
Ok good to know and sorry for the latency :-/.

SubtleMuffin, 2020-10-22:
Hi @Zlika , could you please post your working code? Although I changed the mapping to `[0, 1] -> [1, -1]`, it still generates 0s.

Zlika, 2020-10-23:
Hi! It seems you already had an answer to your problem in #87. Anyway, here is a working example I used to understand and test the LDPC decoders:
```cpp
#include <iostream>
#include <fstream>
#include <string>
#include <chrono>

#include <aff3ct.hpp>
using namespace aff3ct;

#define K 4096 // number of information bits
#define N 6144 // codeword size
#define M 1024 // size of submatrix
#define MAX_ITER 10 // max number of iterations for error correction

// Test performance of LDPC decoder.
int main(int argc, char** argv)
{
  // Get the AFF3CT version
  const std::string v = "v" + std::to_string(tools::version_major()) + "." +
                              std::to_string(tools::version_minor()) + "." +
                              std::to_string(tools::version_release());
  std::cout << "AFF3CT library (" << v << ")" << std::endl;

  // Load G matrix
  std::ifstream g_file;
  g_file.open("gAR4JA_r23_k4096.alist");
  Sparse_matrix G = tools::AList::read(g_file);
  g_file.close();

  // Load H matrix
  std::ifstream h_file;
  h_file.open("AR4JA_r23_k4096n.a");
  Sparse_matrix H = tools::AList::read(h_file);
  h_file.close();

  // Info bits = K first bits
  std::vector<unsigned> read_info_bits_pos(K);
  std::iota(read_info_bits_pos.begin(), read_info_bits_pos.end(), 0);

  // Create LDPC encoder and decoder
  auto encoder = std::unique_ptr<module::Encoder_LDPC<>>(
            new module::Encoder_LDPC<>(4*M, N+M, G));
  auto decoder = std::unique_ptr<module::Decoder_LDPC_BP_horizontal_layered_ONMS_inter<>>(
          new module::Decoder_LDPC_BP_horizontal_layered_ONMS_inter<>(K, N+M, MAX_ITER, H, read_info_bits_pos));

  // Generate random reference bits
  std::vector<int> ref_bits(K);
  for (int i = 0; i < K; i++)
  {
    ref_bits[i] = rand() % 2;
  }

  // LDPC encode
  std::vector<int> enc_bits(N+M, 0);
  encoder->encode(ref_bits, enc_bits);

  // Convert hard bit values (0/1) to soft values (1/-1)
  std::vector<float> enc_bits_float(enc_bits.size());
  for (unsigned int i = 0; i < enc_bits.size(); i++)
  {
    enc_bits_float[i] = (enc_bits[i] == 0) ? 1 : -1;
  }

  // Add noise
  const int NB_BITS_ERR = 10; // Nb of bits to flip
  std::vector<float> enc_bits_float_noise(enc_bits_float.begin(),enc_bits_float.end());
  for (int i = 0; i < NB_BITS_ERR; i++)
  {
    int position = rand() % N;
    enc_bits_float_noise[position] = -1*enc_bits_float_noise[position];
  }

  // Puncture code
  std::fill(enc_bits_float_noise.end() - M, enc_bits_float_noise.end(), 0);

  // Count total number of bits in error
  int nb_err_input = 0;
  for (unsigned int i = 0; i < enc_bits_float.size(); i++)
  {
    if (enc_bits_float_noise[i] != enc_bits_float[i])
    {
      nb_err_input++;
    }
  }
  std::cout << "Nb errors at decoder input (including punctured bits) = " << nb_err_input << std::endl;

  std::vector<int> dec_bits(K, 1);
  const int NB_LOOPS = 100000; // number of frames to decode in the test loop
  std::chrono::steady_clock::time_point begin = std::chrono::steady_clock::now();
  for (int i = 0; i < NB_LOOPS; i++)
  {
    // Decode (SIHO = Soft Input / Hard Output)
    decoder->decode_siho(enc_bits_float_noise, dec_bits);
  }
  std::chrono::steady_clock::time_point end = std::chrono::steady_clock::now();
  long dt_us = std::chrono::duration_cast<std::chrono::microseconds> (end - begin).count();
  std::cout << "Test duration for " << NB_LOOPS << " frames decoded = " << dt_us/1.e6 << "s" << std::endl;
  std::cout << "Decoder throughput (info data bits only) = " << ((K*NB_LOOPS)/(dt_us/1.e6))/1.e6 << "Mb/s" << std::endl;
  std::cout << "Decoder throughput (all bits) = " << (((N+256)*NB_LOOPS)/(dt_us/1.e6))/1.e6 << "Mb/s" << std::endl;

  int nb_err = 0;
  for (unsigned int i = 0; i < dec_bits.size(); i++)
  {
    if (dec_bits[i] != enc_bits[i])
    {
      nb_err++;
    }
  }
  std::cout << "Nb errors after decoding=" << nb_err << std::endl;
}
```

SubtleMuffin, 2020-10-23:
@Zlika Thank you!