aff3ct/aff3ct #54: Segfault with LDPC simulation

kc1212, 2020-01-19:
Hi, running aff3ct at 1ceddfc874f38317e83592b506b1fe78dffe44b4 using the following command is giving me a segfault:

```
$ ./aff3ct -C "LDPC" --enc-type LDPC_H --dec-type BIT_FLIPPING --dec-implem WBF --dec-h-path AR4JA_4096_8192.qc  -m 0.0 -M 4.0 -s 1.0
# ----------------------------------------------------
# ---- A FAST FORWARD ERROR CORRECTION TOOLBOX >> ----
# ----------------------------------------------------
# Parameters:
# * Simulation ------------------------------------
#    ** Type                      = BFER
#    ** Type of bits              = int32
#    ** Type of reals             = float32
#    ** Date (UTC)                = 2020-01-19 16:02:55
#    ** Git version               = v2.3.5
#    ** Code type (C)             = LDPC
#    ** Noise range               = 0 -> 4 dB
#    ** Noise type (E)            = EBN0
#    ** Seed                      = 0
#    ** Statistics                = off
#    ** Debug mode                = off
#    ** Multi-threading (t)       = 6 thread(s)
#    ** Coset approach (c)        = no
#    ** Coded monitoring          = no
#    ** Bad frames tracking       = off
#    ** Bad frames replay         = off
#    ** Bit rate                  = 0.4 (2/5)
#    ** Inter frame level         = 1
# * Source ----------------------------------------
#    ** Type                      = RAND
#    ** Implementation            = STD
#    ** Info. bits (K_info)       = 4096
# * Codec -----------------------------------------
#    ** Type                      = LDPC
#    ** Info. bits (K)            = 4096
#    ** Codeword size (N_cw)      = 10240
#    ** Frame size (N)            = 10240
#    ** Code rate                 = 0.4 (2/5)
# * Encoder ---------------------------------------
#    ** Type                      = LDPC_H
#    ** Systematic                = yes
#    ** H matrix path             = AR4JA_4096_8192.qc
#    ** H matrix reordering       = NONE
#    ** G build method            = IDENTITY
# * Decoder ---------------------------------------
#    ** Type (D)                  = BIT_FLIPPING
#    ** Implementation            = WBF
#    ** Systematic                = yes
#    ** H matrix path             = AR4JA_4096_8192.qc
#    ** H matrix reordering       = NONE
#    ** Num. of iterations (i)    = 10
#    ** Stop criterion (syndrome) = on
#    ** Stop criterion depth      = 1
# * Modem -----------------------------------------
#    ** Type                      = BPSK
#    ** Implementation            = STD
#    ** Bits per symbol           = 1
#    ** Sigma square              = on
# * Channel ---------------------------------------
#    ** Type                      = AWGN
#    ** Implementation            = STD
#    ** Complex                   = off
#    ** Add users                 = off
# * Monitor ---------------------------------------
#    ** Lazy reduction            = off
#    ** Frame error count (e)     = 100
#    ** Compute mutual info       = no
# * Terminal --------------------------------------
#    ** Show Sigma                = off
#    ** Enabled                   = yes
#    ** Frequency (ms)            = 500
# 
# The simulation is running...
./aff3ct: line 28: 20209 Segmentation fault      (core dumped) ( $BIN "$@" )
```

Note that the QC file is taken from https://github.com/aff3ct/configuration_files/blob/master/dec/LDPC/AR4JA_4096_8192.qc

Is there a mistake in the command line options I used?

kc1212, 2020-01-19:
compiled aff3ct in debug mode and here's the backtrace, I suppose it failed to make the generator `G` from `H`?
```
Thread 1 "aff3ct-2.3.5" received signal SIGSEGV, Segmentation fault.
0x00007ffff7bd0fe5 in __memmove_avx_unaligned_erms () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff7bd0fe5 in __memmove_avx_unaligned_erms () from /lib64/libc.so.6
#1  0x00005555562370a3 in std::__copy_move<true, true, std::random_access_iterator_tag>::__copy_m<signed char> (__first=0x55555750f1f0 "", __last=0x55555750e9f0 "", __result=0x55555750d9f0 "") at /usr/include/c++/9/bits/stl_algobase.h:386
#2  0x00005555562354f2 in std::__copy_move_a<true, signed char*, signed char*> (__first=0x55555750f1f0 "", __last=0x55555750e9f0 "", __result=0x55555750d9f0 "") at /usr/include/c++/9/bits/stl_algobase.h:404
#3  0x0000555556ee8e8a in std::__copy_move_a2<true, __gnu_cxx::__normal_iterator<signed char*, std::vector<signed char, std::allocator<signed char> > >, __gnu_cxx::__normal_iterator<signed char*, std::vector<signed char, std::allocator<signed char> > > > (__first=0 '\000', __last=0 '\000', __result=0 '\000')
    at /usr/include/c++/9/bits/stl_algobase.h:440
#4  0x0000555556ee7dbd in std::move<__gnu_cxx::__normal_iterator<signed char*, std::vector<signed char, std::allocator<signed char> > >, __gnu_cxx::__normal_iterator<signed char*, std::vector<signed char, std::allocator<signed char> > > > (__first=0 '\000', __last=0 '\000', __result=0 '\000')
    at /usr/include/c++/9/bits/stl_algobase.h:505
#5  0x0000555556ee45b7 in std::vector<signed char, std::allocator<signed char> >::_M_erase (this=0x7ffff5a19010, __first=0 '\000', __last=0 '\000') at /usr/include/c++/9/bits/vector.tcc:190
#6  0x0000555556ee12f3 in std::vector<signed char, std::allocator<signed char> >::erase (this=0x7ffff5a19010, __first=0 '\000', __last=0 '\000') at /usr/include/c++/9/bits/stl_vector.h:1458
#7  0x0000555556ed8504 in aff3ct::tools::Full_matrix<signed char>::self_resize (this=0x7ffffffface0, n_rows=10240, n_cols=4096, o=aff3ct::tools::Matrix::Origin::TOP_RIGHT) at /home/kc1212/tmp/aff3ct/src/Tools/Algo/Matrix/Full_matrix/Full_matrix.cpp:94
#8  0x0000555556f1b921 in aff3ct::tools::LDPC_matrix_handler::transform_H_to_G_identity (H=..., info_bits_pos=std::vector of length 4096, capacity 4096 = {...}) at /home/kc1212/tmp/aff3ct/src/Tools/Code/LDPC/Matrix_handler/LDPC_matrix_handler.cpp:504
#9  0x0000555556f1a778 in aff3ct::tools::LDPC_matrix_handler::transform_H_to_G_identity (H=..., info_bits_pos=std::vector of length 4096, capacity 4096 = {...}) at /home/kc1212/tmp/aff3ct/src/Tools/Code/LDPC/Matrix_handler/LDPC_matrix_handler.cpp:191
#10 0x0000555556d70851 in aff3ct::module::Encoder_LDPC_from_H<int>::Encoder_LDPC_from_H (this=0x5555574f57e0, K=4096, N=10240, _H=..., G_method="IDENTITY", G_save_path="", G_save_path_single_thread=true, n_frames=1) at /home/kc1212/tmp/aff3ct/src/Module/Encoder/LDPC/From_H/Encoder_LDPC_from_H.cpp:28
#11 0x00005555568d664c in aff3ct::factory::Encoder_LDPC::parameters::build<int> (this=0x5555574c7690, G=..., H=...) at /home/kc1212/tmp/aff3ct/src/Factory/Module/Encoder/LDPC/Encoder_LDPC.cpp:120
#12 0x00005555568d6c03 in aff3ct::factory::Encoder_LDPC::build<int> (params=..., G=..., H=...) at /home/kc1212/tmp/aff3ct/src/Factory/Module/Encoder/LDPC/Encoder_LDPC.cpp:142
#13 0x0000555556b5b77c in aff3ct::module::Codec_LDPC<int, float>::Codec_LDPC (this=0x5555574c79c0, enc_params=..., dec_params=..., pct_params=0x5555574a53f0, __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at /home/kc1212/tmp/aff3ct/src/Module/Codec/LDPC/Codec_LDPC.cpp:85
#14 0x00005555560a6d44 in aff3ct::factory::Codec_LDPC::parameters::build<int, float> (this=0x5555574c75e0, crc=0x0) at /home/kc1212/tmp/aff3ct/src/Factory/Module/Codec/LDPC/Codec_LDPC.cpp:136
#15 0x000055555609e3fb in aff3ct::factory::Codec_SIHO::parameters::build<int, float> (this=0x5555574c7680, crc=0x0) at /home/kc1212/tmp/aff3ct/src/Factory/Module/Codec/Codec_SIHO.cpp:38
#16 0x0000555556eaca10 in aff3ct::simulation::BFER_std<int, float, float>::build_codec (this=0x5555574bcec0, tid=0) at /home/kc1212/tmp/aff3ct/src/Simulation/BFER/Standard/BFER_std.cpp:192
#17 0x0000555556eaae14 in aff3ct::simulation::BFER_std<int, float, float>::__build_communication_chain (this=0x5555574bcec0, tid=0) at /home/kc1212/tmp/aff3ct/src/Simulation/BFER/Standard/BFER_std.cpp:51
#18 0x0000555556e4655c in aff3ct::simulation::BFER<int, float, float>::start_thread_build_comm_chain (simu=0x5555574bcec0, tid=0) at /home/kc1212/tmp/aff3ct/src/Simulation/BFER/BFER.cpp:380
#19 0x0000555556e45906 in aff3ct::simulation::BFER<int, float, float>::_build_communication_chain (this=0x5555574bcec0) at /home/kc1212/tmp/aff3ct/src/Simulation/BFER/BFER.cpp:74
#20 0x0000555556ecdb21 in aff3ct::simulation::Simulation::build_communication_chain (this=0x5555574bcec0) at /home/kc1212/tmp/aff3ct/src/Simulation/Simulation.cpp:24
#21 0x0000555556e43e5f in aff3ct::simulation::BFER<int, float, float>::launch (this=0x5555574bcec0) at /home/kc1212/tmp/aff3ct/src/Simulation/BFER/BFER.cpp:87
#22 0x0000555556aab13a in aff3ct::launcher::Launcher::launch (this=0x5555574a15b0) at /home/kc1212/tmp/aff3ct/src/Launcher/Launcher.cpp:212
#23 0x0000555556fe3921 in main (argc=15, argv=0x7fffffffd3e8) at /home/kc1212/tmp/aff3ct/src/main.cpp:232
```

ejanin, 2020-01-20:
I think you try to encode a QC matrix with the LDPC_H encoder which tries to invert all the matrix to get the generator.
You should try to use the LDPC_QC encoder. It's a more efficient way to do this with QC matrix. Or you may have to expend first the matrix (by our own) and save to alist format to use LDPC_H encoder.
https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/ldpc/encoder.html#enc-type

kc1212, 2020-01-20:
Thanks, that solves my issue.

closed 2020-01-20 by kc1212