aff3ct/aff3ct #56: Function is not thread safe

arvindkrovvidi, 2020-02-04:
Hi,
I tried to run the executable in the development version da6f24b. I'm getting an error saying "Source_user_binary is not thread safe". I need the USER_BIN mode to give input as a binary video file.
This user mode is not available in the latest stable version v2.3.5 1ceddfc. Would really appreciate some help with this. 
[image]

kouchy, 2020-02-05:
Hi @arvindkrovvidi,

We are currently working on that specific point on a private branch, it will be released in one or two months.

Until that point you can run your simulation in mono-threaded mode (with the `-t 1` argument, see https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/simulation/simulation.html#sim-threads-t for more info).

arvindkrovvidi, 2020-02-05:
Hi Adrien,

Thank you so much for the quick response. Your suggestion of using
mono-threaded mode did work. A little bit about our work, I'm part of a
research team at FastVDO working on lossy communications. We would like to
view the video after it has been decoded. For this we would need the
current development version code if possible (or at least an API to output
the decoded stream). Can you kindly share the code with us. We would be
happy to provide any fixes or improvements.

Arvind

kouchy, 2020-02-06:
Hi @arvindkrovvidi,

I will published those new features as soon as possible. I will tell you when I will do that.
I don't want to share the actual version of the code, I have things to finish before.

Please be patient :-).

kouchy, 2020-03-23:
Hi @arvindkrovvidi,

I just pushed a new version of AFF3CT (https://github.com/aff3ct/aff3ct/commit/ec40f269c4f2d7907bea23407615068206f96247) supporting the pipeline.
You can find a short example of code using a pipeline here: https://github.com/aff3ct/my_project_with_aff3ct/tree/development/examples/pipeline.

I hope it will help you.

arvindkrovvidi, 2020-03-24:
Hi Adrien,
I really appreciate you going out of the way to create this example for us.
If I can get this example running, it will save us months of work. I'm
following your instructions to compile the example.  However, I've run into
an issue while using cmake. I'm getting an error

CMake Error at CMakeLists.txt:23 (find_package):
  Could not find a configuration file for package "AFF3CT" that is
compatible
  with requested version "2.3.5".

  The following configuration files were considered but not accepted:

my_project_with_aff3ct/examples/pipeline/cmake/Modules/aff3ct-config.cmake,
version: 2.3.4

The CMakeLists,txt file, at line 23, is looking for version 2.3.5 but the
file I was asked to copy from the folder
../../lib/aff3ct/build/lib/cmake/aff3ct-*/*
is version 2.3.4.

Is there another config file I am missing? Could you let me know how to
proceed further.

Sincerely,
Arvind

kouchy, 2020-03-25:
Hi,

It looks like you did not update the version of the AFF3CT library in `my_project_with_aff3ct/lib/aff3ct`.
You have to update the Git submodule like this : `git submodule update --init --recursive`.
After that you have to re-compile AFF3CT and then you have the copy the new CMake config files in `my_project_with_aff3ct/examples/pipeline/cmake/Modules/`.

Then it should work.