aff3ct/aff3ct #57: Issue with PolarDecoding under BEC

JonathanMandelbaum, 2020-02-05:
Hi,
I discovered some problems using the SC and SCL Polar Decoder for a BEC, using the following parameters:
aff3ct -C "POLAR" -K "512" -N "1024" -R "0.3:0.01:0.5" -E "EP" --chn-type "BEC" --mdm-type "OOK" -e "500" --chn-implem "FAST" -D "SC"  --dec-implem "NAIVE"  --enc-fb-gen-method "BEC" 

I compare my SC-Decoder to some results from the literature and from aff3ct. The FER plot from the aff3ct SC Decoder quickly starts drifting off and ends up a round about a factor 10 worse than the others. The same problem occurs with the SCL-Decoder.  
The simulation was tested on aff3ct-2.3.5.

bonben, 2020-02-05:
Hi @JonathanMandelbaum 

AFF3CT polar decoders on BEC channel have only been tested on the SC decoding algorithm.
The results reproduce the FER of the following thesis [10.5075/epfl-thesis-4461](http://dx.doi.org/10.5075/epfl-thesis-4461), Figure 6.4. You can watch the curves on [our comparator](http://aff3ct.github.io/comparator.html?curve0=0ef949b&curve1=b3593c6&curve2=9d30458&curve3=9828f8e&curve4=aa8324d&curve5=551debe&xaxis=EP&yaxes=BER%2CFER).

Though, there may be issues with the particular test case you are using. What is your reference ? 

Regards, 
Mathieu Léonardon.

BeSturm, 2020-02-05:
Hey @bonben 

as far as we see it, the reference is a rate vs FER curve over a BEC(0.5). These values are reproducible, but unfortunately one can only compare 4 different rates on a BEC, all with the same epsilon, i.e. 0.5, since the reference doesn't provide results for different epsilons.

However when comparing the epsilon vs. FER curve from Mori&Tanaka (http://arxiv.org/abs/0901.2207) we encounter a growing difference compared to the aff3ct implementation in the lower epsilon region. However our own implementation and the Matlab Implementation of Vangala et. al (http://polarcodes.info/) seem to fit the reference of Mori&Tanaka. 

Best regards,
Benjamin Sturm

bonben, 2020-02-07:
Hi.

Thanks a lot for this precise information.
There is indeed a problem in our frozen bit generator for the BEC channel.
We identified the problem and fixed it. We will publish it shortly.
I'll let you know.

Thanks a lot for helping us improve AFF3CT !

bonben, 2020-02-08:
Hi, 

The fix is now deployed in the [development branch ](https://github.com/aff3ct/aff3ct/tree/development), see commit [cb3ca94](https://github.com/aff3ct/aff3ct/commit/cb3ca94ae1077024fabab82fa117a7611e6ad051).
The curve that reproduces Mori & Tanaka (http://arxiv.org/abs/0901.2207) has been added to our reference database. The FER we get is in fact slightly lower than the reference. Please let us know if it corresponds to the results you get with your own implementation.

Regards, 
Mathieu.

BeSturm, 2020-02-08:
Hey, 

thank you very much for your work! 
The curves correspond now with our own implementation. 
The slightly better performance is caused by the handling of erasures (LLR=0) at a decision step of the SC decoder. If in these uncertain situations the bit is decided to 0 or 1 randomly, the slightly better curve is obtained. But if no random decision is made and instead the decoding is aborted and a frame error declared, the resulting curve matches the reference.

Regards,
Benjamin

closed 2020-02-09 by kouchy