aff3ct/aff3ct #58: AVX2 version slower than SSE4.2 version

Zlika, 2020-02-11:
Hi!
I tested the performance of LDPC decoders on my Intel Core i7-7700HQ @ 2.80GHz CPU. As this CPU supports AVX2 instruction set, I used the AVX2 binary version of AFF3CT for my tests. I then switched to the SSE4.2 version to check the difference, and I was surprised by the results: the BP-HL LDPC decoder can process up to 25Mb/s of data (on a single thread) with the AVX2 version, whereas it performs at up to 32Mb/s with the SSE4.2 version (+28%) !?!
I'm not familiar with the SIMD instruction sets, so I googled a bit and found this article that "may" be relevant: https://software.intel.com/en-us/articles/avoiding-avx-sse-transition-penalties .
Do you think this is a "bug" of aff3ct (or MIPP)?
Regards,
Thomas

kouchy, 2020-02-17:
Hi @Zlika,

Yes it is possible to have this behavior sometime, the SIMD strategy used in the BP-HL decoder is an inter-frame strategy. Depending on the frame size this strategy can stress the caches/RAM of the CPU and will affect negatively the throughput. This phenomena is amplified when using larger SIMD instruction sets and this is probably why you have higher performance with SSE ISA.

I don't think this is due to the AVX/SSE penalties because we do not switch from SSE to AVX in MIPP.

If you want to know more about this phenomena, we wrote a paper on turbo decoders based on the same SIMD strategy and the throughput performance decrease is explained : https://hal.archives-ouvertes.fr/hal-01363980/document (see fig. 3, https://doi.org/10.1109/ISTC.2016.7593092).

Zlika, 2020-02-17:
Thank you very much for your answer.

closed 2020-02-17 by Zlika