aff3ct/aff3ct #60: The parameter "--enc-fb-awgn-path" don't work

lukegood, 2020-03-26:
Hi!
I have doenloaded the stable version AFF3CT-2.3.5. When I use "aff3ct-2.3.5 -C "POLAR" -K 4 -N 8 -m 2.0 -M 2.0 --enc-fb-gen-method FILE --enc-fb-awgn-path D:\1.ascll", there is an error “stoi”. The content of "1.ascll" is: 
awgn
0.435999
7 6 5 3 4 2 1 0
Is my command incorrect? How can I solve the problem?
Thank you in advance for your help!
[image]
[image]

bonben, 2020-03-26:
Hi !
Your file is ill-formed, see [documentation](https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/polar/encoder.html#enc-fb-awgn-path). The first line should specify the codeword length (N).
```
8
awgn
0.435999
7 6 5 3 4 2 1 0
```
Please note also that for N < 64, the default `FAST` implem of the SC decoder won't work. You should in this case use the `NAIVE` implementation with the following argument: `--dec-implem NAIVE`.

Regards, 
Mathieu.

lukegood, 2020-03-26:
> Hi !
> Your file is ill-formed, see [documentation](https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/polar/encoder.html#enc-fb-awgn-path). The first line should specify the [...]

Thanks for your help.

closed 2020-03-26 by lukegood