aff3ct/aff3ct #61: Problem about using the parameter "--dec-implem NAIVE" and "--enc-no-sys"(Polar codes)

lukegood, 2020-04-01:
Hi,
I use "--enc-fb-gen-method FILE" to construct Polar codes. When I use "--dec-implem NAIVE --enc-no-sys" as following
```
aff3ct-2.3.5 -C "POLAR" -m "2" -M "2" -K "1024" -N "2048" --enc-fb-gen-method "FILE" --enc-fb-awgn-path "D:\2\51.txt" --dec-type SCL --dec-lists 32 --dec-implem NAIVE --enc-no-sys
```
the FER is 6.78e-5.
[image]
When I don't use them
```
aff3ct-2.3.5 -C "POLAR" -m "2" -M "2" -K "1024" -N "2048" --enc-fb-gen-method "FILE" --enc-fb-awgn-path "D:\2\51.txt" --dec-type SCL --dec-lists 32
``` 
the FER is  2.31e-4.
[image]
**We know that systematic polar encoding usually has the same FER performance as non-systematic polar codes.** So is this because I didn't choose the parameters correctly or because my file has problems? 
My FILE is
```
2048
awgn
1
255 383 443 445 446 447 463 471 475 477 478 479 487 490 491 492 493 494 495 497 498 499 500 501 502 503 504 505 506 507 508 509 510 511 631 635 637 638 639 671 687 695 697 698 699 700 701 702 703 718 719 723 725 726 727 729 730 731 732 733 734 735 739 741 742 743 745 746 747 748 749 750 751 753 754 755 756 757 758 759 760 761 762 763 764 765 766 767 799 811 813 814 815 819 821 822 823 825 826 827 828 829 830 831 839 843 845 846 847 851 852 853 854 855 856 857 858 859 860 861 862 863 866 867 868 869 870 871 872 873 874 875 876 877 878 879 880 881 882 883 884 885 886 887 888 889 890 891 892 893 894 895 903 906 907 908 909 910 911 913 914 915 916 917 918 919 920 921 922 923 924 925 926 927 929 930 931 932 933 934 935 936 937 938 939 940 941 942 943 944 945 946 947 948 949 950 951 952 953 954 955 956 957 958 959 961 962 963 964 965 966 967 968 969 970 971 972 973 974 975 976 977 978 979 980 981 982 983 984 985 986 987 988 989 990 991 992 993 994 995 996 997 998 999 1000 1001 1002 1003 1004 1005 1006 1007 1008 1009 1010 1011 1012 1013 1014 1015 1016 1017 1018 1019 1020 1021 1022 1023 1087 1119 1133 1134 1135 1139 1141 1142 1143 1145 1146 1147 1148 1149 1150 1151 1179 1181 1182 1183 1191 1195 1197 1198 1199 1203 1205 1206 1207 1208 1209 1210 1211 1212 1213 1214 1215 1223 1227 1228 1229 1230 1231 1233 1234 1235 1236 1237 1238 1239 1240 1241 1242 1243 1244 1245 1246 1247 1249 1250 1251 1252 1253 1254 1255 1256 1257 1258 1259 1260 1261 1262 1263 1264 1265 1266 1267 1268 1269 1270 1271 1272 1273 1274 1275 1276 1277 1278 1279 1295 1303 1307 1309 1310 1311 1319 1321 1322 1323 1324 1325 1326 1327 1329 1330 1331 1332 1333 1334 1335 1336 1337 1338 1339 1340 1341 1342 1343 1347 1349 1350 1351 1353 1354 1355 1356 1357 1358 1359 1361 1362 1363 1364 1365 1366 1367 1368 1369 1370 1371 1372 1373 1374 1375 1377 1378 1379 1380 1381 1382 1383 1384 1385 1386 1387 1388 1389 1390 1391 1392 1393 1394 1395 1396 1397 1398 1399 1400 1401 1402 1403 1404 1405 1406 1407 1411 1413 1414 1415 1417 1418 1419 1420 1421 1422 1423 1425 1426 1427 1428 1429 1430 1431 1432 1433 1434 1435 1436 1437 1438 1439 1441 1442 1443 1444 1445 1446 1447 1448 1449 1450 1451 1452 1453 1454 1455 1456 1457 1458 1459 1460 1461 1462 1463 1464 1465 1466 1467 1468 1469 1470 1471 1473 1474 1475 1476 1477 1478 1479 1480 1481 1482 1483 1484 1485 1486 1487 1488 1489 1490 1491 1492 1493 1494 1495 1496 1497 1498 1499 1500 1501 1502 1503 1504 1505 1506 1507 1508 1509 1510 1511 1512 1513 1514 1515 1516 1517 1518 1519 1520 1521 1522 1523 1524 1525 1526 1527 1528 1529 1530 1531 1532 1533 1534 1535 1551 1557 1558 1559 1561 1562 1563 1564 1565 1566 1567 1571 1573 1574 1575 1577 1578 1579 1580 1581 1582 1583 1585 1586 1587 1588 1589 1590 1591 1592 1593 1594 1595 1596 1597 1598 1599 1603 1605 1606 1607 1609 1610 1611 1612 1613 1614 1615 1617 1618 1619 1620 1621 1622 1623 1624 1625 1626 1627 1628 1629 1630 1631 1633 1634 1635 1636 1637 1638 1639 1640 1641 1642 1643 1644 1645 1646 1647 1648 1649 1650 1651 1652 1653 1654 1655 1656 1657 1658 1659 1660 1661 1662 1663 1667 1669 1670 1671 1673 1674 1675 1676 1677 1678 1679 1681 1682 1683 1684 1685 1686 1687 1688 1689 1690 1691 1692 1693 1694 1695 1697 1698 1699 1700 1701 1702 1703 1704 1705 1706 1707 1708 1709 1710 1711 1712 1713 1714 1715 1716 1717 1718 1719 1720 1721 1722 1723 1724 1725 1726 1727 1729 1730 1731 1732 1733 1734 1735 1736 1737 1738 1739 1740 1741 1742 1743 1744 1745 1746 1747 1748 1749 1750 1751 1752 1753 1754 1755 1756 1757 1758 1759 1760 1761 1762 1763 1764 1765 1766 1767 1768 1769 1770 1771 1772 1773 1774 1775 1776 1777 1778 1779 1780 1781 1782 1783 1784 1785 1786 1787 1788 1789 1790 1791 1795 1797 1798 1799 1801 1802 1803 1804 1805 1806 1807 1809 1810 1811 1812 1813 1814 1815 1816 1817 1818 1819 1820 1821 1822 1823 1825 1826 1827 1828 1829 1830 1831 1832 1833 1834 1835 1836 1837 1838 1839 1840 1841 1842 1843 1844 1845 1846 1847 1848 1849 1850 1851 1852 1853 1854 1855 1857 1858 1859 1860 1861 1862 1863 1864 1865 1866 1867 1868 1869 1870 1871 1872 1873 1874 1875 1876 1877 1878 1879 1880 1881 1882 1883 1884 1885 1886 1887 1888 1889 1890 1891 1892 1893 1894 1895 1896 1897 1898 1899 1900 1901 1902 1903 1904 1905 1906 1907 1908 1909 1910 1911 1912 1913 1914 1915 1916 1917 1918 1919 1921 1922 1923 1924 1925 1926 1927 1928 1929 1930 1931 1932 1933 1934 1935 1936 1937 1938 1939 1940 1941 1942 1943 1944 1945 1946 1947 1948 1949 1950 1951 1952 1953 1954 1955 1956 1957 1958 1959 1960 1961 1962 1963 1964 1965 1966 1967 1968 1969 1970 1971 1972 1973 1974 1975 1976 1977 1978 1979 1980 1981 1982 1983 1984 1985 1986 1987 1988 1989 1990 1991 1992 1993 1994 1995 1996 1997 1998 1999 2000 2001 2002 2003 2004 2005 2006 2007 2008 2009 2010 2011 2012 2013 2014 2015 2016 2017 2018 2019 2020 2021 2022 2023 2024 2025 2026 2027 2028 2029 2030 2031 2032 2033 2034 2035 2036 2037 2038 2039 2040 2041 2042 2043 2044 2045 2046 2047 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 256 257 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 277 278 279 280 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 300 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 320 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 340 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 360 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 380 381 382 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 400 401 402 403 404 405 406 407 408 409 410 411 412 413 414 415 416 417 418 419 420 421 422 423 424 425 426 427 428 429 430 431 432 433 434 435 436 437 438 439 440 441 442 444 448 449 450 451 452 453 454 455 456 457 458 459 460 461 462 464 465 466 467 468 469 470 472 473 474 476 480 481 482 483 484 485 486 488 489 496 512 513 514 515 516 517 518 519 520 521 522 523 524 525 526 527 528 529 530 531 532 533 534 535 536 537 538 539 540 541 542 543 544 545 546 547 548 549 550 551 552 553 554 555 556 557 558 559 560 561 562 563 564 565 566 567 568 569 570 571 572 573 574 575 576 577 578 579 580 581 582 583 584 585 586 587 588 589 590 591 592 593 594 595 596 597 598 599 600 601 602 603 604 605 606 607 608 609 610 611 612 613 614 615 616 617 618 619 620 621 622 623 624 625 626 627 628 629 630 632 633 634 636 640 641 642 643 644 645 646 647 648 649 650 651 652 653 654 655 656 657 658 659 660 661 662 663 664 665 666 667 668 669 670 672 673 674 675 676 677 678 679 680 681 682 683 684 685 686 688 689 690 691 692 693 694 696 704 705 706 707 708 709 710 711 712 713 714 715 716 717 720 721 722 724 728 736 737 738 740 744 752 768 769 770 771 772 773 774 775 776 777 778 779 780 781 782 783 784 785 786 787 788 789 790 791 792 793 794 795 796 797 798 800 801 802 803 804 805 806 807 808 809 810 812 816 817 818 820 824 832 833 834 835 836 837 838 840 841 842 844 8 [... cut ...] 48 1549 1550 1552 1553 1554 1555 1556 1560 1568 1569 1570 1572 1576 1584 1600 1601 1602 1604 1608 1616 1632 1664 1665 1666 1668 1672 1680 1696 1728 1792 1793 1794 1796 1800 1808 1824 1856 1920 
```
I just want to construct the Polar code with K/N=0.5. The first 1024 channels in this file are the channels that will be selected to transmit information bits. They are arranged in ascending order of the serial number. It is not the reason leads to the wrong result.
Thank you in advance for your help!

kouchy, 2020-04-01:
Hi @lukegood,

The problem you encounter is due to too much aggressive optimizations made in the decoding polar tree, and more specifically to the Single Parity Check (SPC) nodes. Sometime using this optimization is transparent, sometime it is not (in your case clearly not). We explained this phenomena in this paper: _Fast and Flexible Software Polar List Decoders_ (check _section 3.2.2 Tree pruning strategy,_ [DOI](https://doi.org/10.1007/s11265-018-1430-3), [PDF](https://hal.inria.fr/hal-01987848/document)).

To fix the problem you need to disable big SPC nodes with the `--dec-polar-nodes` parameter like this:

    aff3ct-2.3.5 -C "POLAR" -m "2" -M "2" -K "1024" -N "2048" --enc-fb-gen-method "FILE" --enc-fb-awgn-path "D:\2\51.txt" --dec-type SCL --dec-lists 32 --dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"

`SPC_4` limits the size of the SPC nodes to 4 LLRs maximum.

Here is my simulation result:

    # ---------------------||------------------------------------------------------||---------------------
    #  Signal Noise Ratio  ||   Bit Error Rate (BER) and Frame Error Rate (FER)    ||  Global throughput  
    #         (SNR)        ||                                                      ||  and elapsed time   
    # ---------------------||------------------------------------------------------||---------------------
    # ----------|----------||----------|----------|----------|----------|----------||----------|----------
    #     Es/N0 |    Eb/N0 ||      FRA |       BE |       FE |      BER |      FER ||  SIM_THR |    ET/RT 
    #      (dB) |     (dB) ||          |          |          |          |          ||   (Mb/s) | (hhmmss) 
    # ----------|----------||----------|----------|----------|----------|----------||----------|----------
          -1.01 |     2.00 ||  1368753 |     2090 |      100 | 1.49e-06 | 7.31e-05 ||   28.891 | 00h00'48   
    # End of the simulation.

This is very close to the FER of the `NAIVE` version.

I hope it helps.

lukegood, 2020-04-02:
**@kouchy** Thank you so much for the quick response.
But sometimes when I use some other files without  the `--dec-implem NAIVE` parameter, the FER is close to the `NAIVE` version. 
So am I right in understanding that problem may occur without  `--dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"` because of too much aggressive optimizations and the result will always be right with `--dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"`?

kouchy, 2020-04-02:
Yes, if you use the `--dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"` you will always have the optimal decoding performance of the SCL.

bonben, 2020-04-02:
To develop kouchy's response, we never noticed any performance degradation on the SCL algorithm using `--dec-polar-nodes"{R0,R0L,R1,REP,REPL,SPC_4}"`. However, strictly speaking, R1 and SPC might, in some rare cases, degrade performance since their use modifies the SCL decoding algorithm. To disable them: `--dec-polar-nodes"{R0,R0L,R1_1,REP,REPL}"` `.

lukegood, 2020-04-03:
> Yes, if you use the `--dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"` you will always have the optimal decoding performance of the SCL.

Thanks for your help！

lukegood, 2020-04-03:
> To develop @kouchy's response, we never noticed any performance degradation on the SCL algorithm using `--dec-polar-nodes"{R0,R0L,R1,REP,REPL,SPC_4}"`. However, strictly speaking, R1 and SPC nodes might, in some rare cases, degrade performance since their use modifies the SCL decoding algorithm. To disable them: `--dec-polar-nodes"{R0,R0L,R1_1,REP,REPL}"` `.

Thank you for your help. @bonben 
You mean that the SCL decoder always gets the right performance with `--dec-polar-nodes"{R0,R0L,R1_1,REP,REPL}"` . Am I right?

bonben, 2020-04-03:
That's right !

lukegood, 2020-04-04:
> That's right !
Thank you for your help!

closed 2020-04-04 by lukegood