aff3ct/aff3ct #62: Problem about --dec-polar-nodes parameter

Whitee1, 2020-04-05:
Hi, I have a problem that when I use a command with 
`--dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"`
```
aff3ct-2.3.5 -C POLAR -m 2 -M 2 -K 1040 -N 2048 --enc-fb-gen-method TV --dec-type SCL --dec-lists 32 --dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}" --crc-type "16-CCITT"
```
the result is 
```
The simulation is running...
# -----------------------||-----------------------------------------------------------||-----------------------
#   Signal Noise Ratio   ||      Bit Error Rate (BER) and Frame Error Rate (FER)      ||   Global throughput
#          (SNR)         ||                                                           ||   and elapsed time
# -----------------------||-----------------------------------------------------------||-----------------------
# -----------|-----------||-----------|-----------|-----------|-----------|-----------||-----------|-----------
#      Es/N0 |     Eb/N0 ||       FRA |        BE |        FE |       BER |       FER ||   SIM_THR |     ET/RT
#       (dB) |      (dB) ||           |           |           |           |           ||    (Mb/s) |  (hhmmss)
# -----------|-----------||-----------|-----------|-----------|-----------|-----------||-----------|-----------
       -1.01 |      2.00 ||    209325 |       191 |         6 | 8.91e-007 | 2.87e-005 ||     1.147 |  00h03'06  x
# End of the simulation.
```
which is much better than the command that removes 
`--dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}"`
```
aff3ct-2.3.5 -C POLAR -m 2 -M 2 -K 1040 -N 2048 --enc-fb-gen-method TV --dec-type SCL --dec-lists 32 --crc-type "16-CCITT"
```

```
 The simulation is running...
# -----------------------||-----------------------------------------------------------||-----------------------
#   Signal Noise Ratio   ||      Bit Error Rate (BER) and Frame Error Rate (FER)      ||   Global throughput
#          (SNR)         ||                                                           ||   and elapsed time
# -----------------------||-----------------------------------------------------------||-----------------------
# -----------|-----------||-----------|-----------|-----------|-----------|-----------||-----------|-----------
#      Es/N0 |     Eb/N0 ||       FRA |        BE |        FE |       BER |       FER ||   SIM_THR |     ET/RT
#       (dB) |      (dB) ||           |           |           |           |           ||    (Mb/s) |  (hhmmss)
# -----------|-----------||-----------|-----------|-----------|-----------|-----------||-----------|-----------
       -1.01 |      2.00 ||    207278 |       715 |        29 | 3.37e-006 | 1.40e-004 ||     1.314 |  00h02'41  x
# End of the simulation.
```
So what can I do to solve this problem? I notice [https://github.com/aff3ct/aff3ct/issues/61](https://github.com/aff3ct/aff3ct/issues/61). Is it the same problem? 

Thanks for help!

kouchy, 2020-04-06:
Hi @Whitee1,

Yes it is normal, by default when you do not use the `--dec-polar-nodes` parameters, the tree cuts are configured like this : `{R0,R0L,R1,REP,REPL,SPC}`.

So all the SPC nodes are considered and this can lead to performance loss.

Whitee1, 2020-04-06:
@kouchy Thank you!

closed 2020-04-06 by Whitee1