aff3ct/aff3ct #63: Problem“mipp::mul<int32_t> (SSE2) is undefined!”with aff3ct

HongweiHou, 2020-04-14:
I install AFF3CT by following instructions:
```
git clone --recursive https://github.com/aff3ct/aff3ct.git
mkdir aff3ct/build
cd aff3ct/build
cmake .. -DCMAKE_BUILD_TYPE="Release"
make -j4
```
and to reproduce the throughput performance in paper "Fast and Flexible Software Polar List Decoders",  I use the command line parameters as follows:
`./aff3ct -p "32" --sim-type "BFER" -C "POLAR" -m "0.0" -M "2.6" -s "0.1" -e "200" --enc-fb-gen-method "GA" -K "1723" -N "2048" --src-type "RAND" --src-implem "FAST" --chn-type "AWGN" --chn-implem "FAST" --dec-type "ASCL_MEM" --dec-simd "INTRA" --crc-type "32-GZIP" --crc-implem "FAST" --dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}" -L "8"`
But some errors occurred during the running of the program:
```
(EE) mipp::mul<int32_t> (SSE2) is undefined!
(EE) Backtrace:
(EE) ./aff3ct-2.3.5(+0xe23809) [0x7f28e0823809]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct5tools17PRNG_MT19937_simd4seedEN4mipp3RegIiEE+0x42) [0x7f28e0823c72]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct5tools17PRNG_MT19937_simdC1Ev+0xd6) [0x7f28e0823e36]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct6module18Source_random_fastIsEC1Eiii+0x5d) [0x7f28e07ad6ed]
(EE) ./aff3ct-2.3.5(_ZNK6aff3ct7factory6Source10parameters5buildIsEEPNS_6module6SourceIT_EEv+0x184) [0x7f28e041cb04]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation8BFER_stdIsfsE12build_sourceEi+0x49) [0x7f28e07efaa9]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation8BFER_stdIsfsE27__build_communication_chainEi+0x4a) [0x7f28e07f73ea]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation4BFERIsfsE29start_thread_build_comm_chainEPS2_i+0x32) [0x7f28e07d4932]
(EE) /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xbd66f) [0x7f28df32d66f]
(EE) /lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f28dea976db]
(EE) /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f28de7b188f]
(EE) mipp::mul<int32_t> (SSE2) is undefined!
(EE) Backtrace:
(EE) ./aff3ct-2.3.5(+0xe23809) [0x7f28e0823809]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct5tools17PRNG_MT19937_simd4seedEN4mipp3RegIiEE+0x42) [0x7f28e0823c72]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct5tools17PRNG_MT19937_simdC1Ev+0xd6) [0x7f28e0823e36]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct6module18Source_random_fastIsEC1Eiii+0x5d) [0x7f28e07ad6ed]
(EE) ./aff3ct-2.3.5(_ZNK6aff3ct7factory6Source10parameters5buildIsEEPNS_6module6SourceIT_EEv+0x184) [0x7f28e041cb04]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation8BFER_stdIsfsE12build_sourceEi+0x49) [0x7f28e07efaa9]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation8BFER_stdIsfsE27__build_communication_chainEi+0x4a) [0x7f28e07f73ea]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation4BFERIsfsE29start_thread_build_comm_chainEPS2_i+0x32) [0x7f28e07d4932]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation4BFERIsfsE26_build_communication_chainEv+0x166) [0x7f28e07d4ce6]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation10Simulation25build_communication_chainEv+0x17) [0x7f28e08091d7]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct10simulation4BFERIsfsE6launchEv+0x39) [0x7f28e07cb489]
(EE) ./aff3ct-2.3.5(_ZN6aff3ct8launcher8Launcher6launchEv+0x114) [0x7f28e04ac4d4]
(EE) ./aff3ct-2.3.5(main+0x1d9) [0x7f28dfeda349]
(EE) /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f28de6b1b97]
(EE) ./aff3ct-2.3.5(_start+0x2a) [0x7f28dfeda46a]
```
Would it be convenient for you to answer the questions?Thank you for your attention to this matter and I look forward to hearing from you soon.

With thanks and best regards.

bonben, 2020-04-14:
Hi @HongweiHou .

Thanks for your interest in our work.

You might find the answer to your question here : https://aff3ct.readthedocs.io/en/latest/user/installation/compilation/compilation.html#specific-options

Your issue is related to the support of SIMD instruction by your CPU. Using this command line for cmake should solve your problem. 
```
cmake .. -DCMAKE_BUILD_TYPE="Release" -DCMAKE_CXX_FLAGS="-funroll-loops -march=native"
```
Let us know if it does not.

Regards.

HongweiHou, 2020-04-14:
> Hi @HongweiHou .
>
> Thanks for your interest in our work.
>
> You might find the answer to your question here : [...]

It works well. But when I use the command line parameters as follows:
`./aff3ct -p "32" --sim-type "BFER" -C "POLAR" -m "0.0" -M "2.6" -s "0.1" -e "200" --enc-fb-gen-method "GA" -K "1723" -N "2048" --src-type "RAND" --src-implem "FAST" --chn-type "AWGN" --chn-implem "FAST" --dec-type "ASCL_MEM" --dec-simd "INTRA" --crc-type "32-GZIP" --crc-implem "FAST" --dec-polar-nodes "{R0,R0L,R1,REP,REPL,SPC_4}" -L "8" -t "1"`
I can't reproduce the throughput performance in paper "Fast and Flexible Software Polar List Decoders", Would it be convenient for you to tell me how to reproduce it? Is there an error in my command line parameters?

With thanks and best regards.

bonben, 2020-04-14:
Few things : 

- The measured throughput if you use this command line correspond to the throughput of the whole chain. If you want the throughput of the decoder only, add the `--sim-stats` argument that will detail the throughputs and latencies of each module.
- To achieve best performance, use quantified version of the decoder : `-p "8"`. In this case, to avoid performance issues, only REP nodes whose size is inferior to 8 should be used: `--dec-polar-nodes "{R0,R0L,R1,REP_2-8,REPL,SPC_4}"`
- In the paper, the number of information bits is 1723. In the command line, `K` correspond to the number of non-frozen bits, CRC bits included. Therefore, as you use a 32-bit long CRC, `K` must be set to 1755: `-K 1755`
- The SNR values for which results are provided are 3.5, 4.0 and 4.5 dB : `-m "3.5" -M "4.5" -s "0.5"`

Therefore it should be good with the following command line : 
```
./aff3ct -p "8" --sim-type "BFER" -C "POLAR" -m "3.5" -M "4.5" -s "0.5" -e "200" --enc-fb-gen-method "GA" -K "1755" -N "2048" --src-type "RAND" --src-implem "FAST" --chn-type "AWGN" --chn-implem "FAST" --dec-type "ASCL_MEM" --dec-simd "INTRA" --crc-type "32-GZIP" --crc-implem "FAST" --dec-polar-nodes "{R0,R0L,R1,REP_2-8,REPL,SPC_4}" -L "8" -t "1" --sim-stats
```

Regards.

HongweiHou, 2020-04-14:
> Few things :
>
> * The measured throughput if you use this command line correspond to the throughput of the whole chain. If you want the throughput of the decoder only, add the `--sim-stats` argument [...]

Thank you for your warm answers. I used the command line you provided and got the decoder throughput performance I wanted. But I still have some problems: On my machine, the decoder throughput performance still does not reach the ~425Mbps given in the paper.
The result is following:
3.5dB   113.04Mbps(Average), 1.82Mbps(Min), 170.86Mbps(Max);
4.0dB   130.96Mbps(Average), 1.64Mbps(Min), 150.71Mbps(Max);
4.5dB   124.48Mbps(Average), 1.64Mbps(Min), 141.57Mbps(Max);
The relevant parameters of the test machine are as follows:
1: CPU Model : Intel Xeon Gold 6154;
2: Number of Physical Cores: 72(but only one used);
3: Main Frequency : 3.00GHz;
4: Turbo: Disabled;
Is the poor throughput performance I get due to my machine configuration? Is the poor throughput performance I get due to my machine configuration?

With thanks and best regards.

bonben, 2020-04-15:
First of all, in the case of a N=2048 polar code, you may obtain better results by compiling with avx2 and not avx512 : replace `-march=native` by `-mavx2` and give it an other try.

Then, it is weird that you get so little difference between the SNR points. Could you provide the full output of your simulation ? Please use the code block function of markdown to format this output (https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks).

bonben, 2020-04-15:
And also the output of `aff3ct -v` !

kouchy, 2020-04-20:
Hi @HongweiHou,

I would like to complete the @bonben's answer.
First **reproducibility is something very important for us** and even more for published results. So we are very concerned by your issue.

The result you want to reproduce is available in the AFF3CT BER/FER comparator here: http://aff3ct.github.io/comparator.html?curve0=85e4f9e.
This page will give you **the command line** we used in the paper as well as **the corresponding paper**.

I tried to reproduced the performance results on the original machine. Here are the specs:
- Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
- Turbo Boost: Enabled
- Max freq: 3.9 GHz

The machine runs on Ubuntu 16.04 LTS and I compiled with GCC (v5.4, `-O3 -funroll-loops -march=avx2`).

`aff3ct --version` gives me:

    aff3ct (Linux 64-bit, g++-5.4, AVX2) v2.3.5-385-g1bbacbc
    Compilation options:
      - Precision:         8/16/32/64-bit
      - Polar bit packing: on
      - Polar bounds:      off
      - Terminal colors:   on
      - Backtrace:         on
      - External strings:  on
      - MPI:               off
      - GSL:               off
      - MKL:               off
      - SystemC:           off
    Copyright (c) 2016-2019 - MIT license.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

As you can see I used the last development version of AFF3CT at this time (https://github.com/aff3ct/aff3ct/tree/1bbacbcfc3bb4b88f109e0c3032cc07713585c8d).

I took the command given by the AFF3CT BER/FER comparator and I added `-t 1 --sim-stats` to be on 1 thread and show the statistics of the decoder:

    aff3ct -p 8 --sim-type BFER -C POLAR -m 3.5 -M 4.51 -s 0.5 -e 1000000 --enc-fb-gen-method GA -K 1755 -N 2048 --src-type RAND --src-implem FAST --chn-type AWGN --chn-implem FAST --qnt-bits 6 --qnt-dec 1 --qnt-implem FAST --dec-type ASCL --dec-simd INTRA --crc-type 32-GZIP --crc-implem FAST --dec-polar-nodes "{R0,R0L,R1,REP_2-8,REPL,SPC_4}" -L 32 -t 1 --sim-stats

I ran each SNR point one minute to stabilize the CPU frequency and here is my simulation result:

    # ----------------------------------------------------
    # ---- A FAST FORWARD ERROR CORRECTION TOOLBOX >> ----
    # ----------------------------------------------------
    # Parameters:
    # * Simulation ------------------------------------
    #    ** Type                     = BFER
    #    ** Type of bits             = int8
    #    ** Type of reals            = float32
    #    ** Type of quant. reals     = int8
    #    ** Date (UTC)               = 2020-04-20 09:17:36
    #    ** Git version              = v2.3.5-385-g1bbacbc
    #    ** Code type (C)            = POLAR
    #    ** Noise range              = 3.5 -> 4.5 dB
    #    ** Noise type (E)           = EBN0
    #    ** Seed                     = 0
    #    ** Statistics               = on
    #    ** Debug mode               = off
    #    ** Multi-threading (t)      = 1 thread(s)
    #    ** Coset approach (c)       = no
    #    ** Coded monitoring         = no
    #    ** Bad frames tracking      = off
    #    ** Bad frames replay        = off
    #    ** Clone allocations        = off
    #    ** Sequence multithreading  = off
    #    ** Bit rate                 = 0.841309 (1723/2048)
    #    ** Inter frame level        = 1
    # * Source ----------------------------------------
    #    ** Type                     = RAND
    #    ** Implementation           = FAST
[... 123 lines cut ...]
    #        Statistics for the given task       ||       Basic statistics       ||       Measured throughput      ||        Measured latency        
    #     ('*' = any, '-' = same as previous)    ||          on the task         ||   considering the last socket  ||   considering the last socket  
    # -------------------------------------------||------------------------------||--------------------------------||--------------------------------
    # -------------|-------------------|---------||----------|----------|--------||----------|----------|----------||----------|----------|----------
    #       MODULE |              TASK |   TIMER ||    CALLS |     TIME |   PERC ||  AVERAGE |  MINIMUM |  MAXIMUM ||  AVERAGE |  MINIMUM |  MAXIMUM 
    #              |                   |         ||          |      (s) |    (%) ||   (Mb/s) |   (Mb/s) |   (Mb/s) ||     (us) |     (us) |     (us) 
    # -------------|-------------------|---------||----------|----------|--------||----------|----------|----------||----------|----------|----------
    #      Encoder |            encode |       * ||  1904750 |    32.27 |  54.99 ||   120.88 |     3.57 |   122.60 ||    16.94 |    16.70 |   572.95
    #      Channel |         add_noise |       * ||  1904750 |    14.76 |  25.16 ||   264.20 |     3.73 |   278.19 ||     7.75 |     7.36 |   549.40
    #      Decoder |       decode_siho |       * ||  1904750 |     7.68 |  13.09 ||   435.19 |     1.34 |   474.45 ||     4.03 |     3.70 |  1312.50
    #          CRC |             build |       * ||  1904750 |     1.62 |   2.76 ||  2062.64 |     3.65 |  2196.50 ||     0.85 |     0.80 |   480.50
    #        Modem |          modulate |       * ||  1904750 |     0.56 |   0.96 ||  6914.97 |    36.88 |  9941.75 ||     0.30 |     0.21 |    55.54
    #       Source |          generate |       * ||  1904750 |     0.55 |   0.93 ||  6007.64 |     3.60 |  7004.06 ||     0.29 |     0.25 |   478.98
[... cut ...]
    #        TOTAL |                 * |       * ||  1904750 |    58.69 | 100.00 ||     0.03 |     0.00 |     0.03 ||    30.81 |    29.60 |  3579.44
    #
    # End of the simulation.

This time **I was able to reproduce the expected throughput performances** but sometime regressions can occur and it is very hard to track (even if we try to do it of course). So in that case you can refer to the AFF3CT version we used in the BER/FER comparator (cf. http://aff3ct.github.io/comparator.html?curve0=85e4f9e) and compile this specific AFF3CT version.

I hope it helps.

HongweiHou, 2020-05-11:
I think the throughput performance I get is due to my machine model and configuration. I have another question now: Why is the difference in throughput performance between 16-bit fixed point and 8-bit fixed point in Table 2 so small? And sometimes the throughput of 16-bit fixed point is higher than 8-bit fixed point?

kouchy, 2020-05-12:
The difference is very small for 3.5 and 4.0 dB, I will not try to interpret those values to much. But it is true that we noticed some slowdowns in 8-bit for the SCL decoder, it depends on the codeword size. For big codeword sizes the vectorized part of the SCL can take advantage of the SIMD units and the decoder will be faster than 16-bit and 32-bit implems. For short codeword sizes the sequential part of the SCL is mainly driving the throughput performance and sequential 8-bit operations are slower than 16-bit and 32-bit operation on x86 architectures.

HongweiHou, 2020-05-12:
> The difference is very small for 3.5 and 4.0 dB, I will not try to interpret those values to much. But it is true that we noticed some slowdowns in 8-bit for the SCL decoder, it depends on the [...]

Thank you for your warm answers.

With thanks and best regards.

kouchy, 2020-05-12:
You're welcome. I close this issue.

closed 2020-05-12 by kouchy