aff3ct/aff3ct #64: Possible problem with Turbo_DB encoder

tbresnan, 2020-04-19:
I was comparing the Turbo_DB encoder(type DVB-RCS2) outputs to another encoder I'm using. I found that the parity bits did not match between the two.  The code below is the relevant encoder code using  aff3ct. K=512,  1/2 rate code

    aff3ct::factory::Interleaver_core::parameters m_itl_params;

    // Custom ARP Interleaver
    m_itl_params.size = K/2;
    m_itl_params.type = "USER";
    m_itl_params.path = "../itl_files/custom_itl.txt";

    // Create and init interleaver core
    auto p_itl_core = aff3ct::factory::Interleaver_core::build(m_itl_params);
    p_itl_core->init();
    auto p_interleaver_b_8 = aff3ct::factory::Interleaver::build<B_8>(*p_itl_core);

    aff3ct::factory::Encoder_RSC_DB::parameters m_sub_enc_params;

    // Set constituent encoder parameters (DVB-RCS2)
    m_sub_enc_params.K = K;
    m_sub_enc_params.N_cw = m_sub_enc_params.K * 2;
    m_sub_enc_params.standard = "DVB-RCS2";

    // Create the encoder
    auto p_enc_n = aff3ct::factory::Encoder_RSC_DB::build<B_8>(m_sub_enc_params);

    // Make the Turbo DB  Encoder
    aff3ct::factory::Encoder_turbo_DB::parameters m_enc_params;
    m_enc_params.K = K;
    m_enc_params.N_cw = m_enc_params.K * 3;

    auto p_encoder = aff3ct::factory::Encoder_turbo_DB::build<B_8>(m_enc_params,
                                                                 *p_interleaver_b_8,
                                                                 *p_enc_n);
    // Make puncturer
    aff3ct::factory::Puncturer_turbo_DB::parameters m_punc_params;
    m_punc_params.K = K;
    m_punc_params.N = m_punc_params.K*2;
    m_punc_params.N_cw = m_punc_params.K*3;
    m_punc_params.type = "TURBO_DB";
    auto p_puncturer = aff3ct::factory::Puncturer_turbo_DB::build<B_8 ,Q_8>(m_punc_params);

    // Buffers to hold input and output
    std::vector<B_8> data(m_punc_params.K, 0);
    std::vector<B_8> m_codeword(p_puncturer->get_N_cw(),0);
    std::vector<Q_8> cw_punc(p_puncturer->get_N());

    // Set a few bits for testing
    data[2] = 1;
    data[3] = 1;
    data[4] = 1;
    data[13] = 1;
    data[37] = 1;

    // Encode the data
    p_encoder->encode(data, m_codeword);

    // Puncture the codeword
    p_puncturer->puncture(m_codeword, cw_punc);
I think the aff3ct stage 1 bit level permutation is not correct. It looks like bit swapping is done at j mod 2 = 0 rather than j mod 2 = 1 where j is natural order input couple index.

**Reference DVB-RCS2:**
The natural order index is defined as j=0,...N-1 and i is the interleaved order index.
Stage 1 permutation:
if j mod 2 = 1, interchange Aj and Bj, i.e. let (Aj,Bj) = (Bj,Aj)

This is the code in question:

File:  \aff3ct\src\Module\Encoder\Turbo_DB\Encoder_turbo_DB.cpp
75:    for (**auto i = 0**; i < this->K; i+=4)
76:       std::swap(U_K_cpy[frame_id * this->K +i], U_K_cpy[frame_id * this->K + i +1]);

If I change line 75 to start at i =2 shown below the aff3ct version parity bits match the other encoder I'm comparing it with. 

File:  \aff3ct\src\Module\Encoder\Turbo_DB\Encoder_turbo_DB.cpp
75:    for (**auto i = 2**; i < this->K; i+=4)
76:       std::swap(U_K_cpy[frame_id * this->K +i], U_K_cpy[frame_id * this->K + i +1]);

Is this a problem or am I missing some subtlety?

ttonnellier, 2020-04-20:
Dear tbresnan,

The DVB-RCS2 specification [can be found here.](https://www.etsi.org/deliver/etsi_en/301500_301599/30154502/01.02.01_60/en_30154502v010201p.pdf)
On page 133, it says: `if j mod 2 = 0, interchange Aj  and Bj, i.e. the internal order is changed to (Bj, Aj)` 
And according to the previous paragraph, indexes start at 0.

Best, 
Thibaud

tbresnan, 2020-04-20:
Hi Thibaud,
It looks like there was a change between V1.1.1 and V1.2.1 which is the subtlety I was not aware of. The other encoder I was using is based on V.1.1.1. I'm going to update my other encoder to support either version. Thanks for looking at this.
Much appreciated,
Tim

closed 2020-04-20 by tbresnan

ttonnellier, 2020-04-20:
With pleasure :)
Good luck! 
If you find anything else weird, don't hesitate!
Thibaud

Divya916947, 2021-03-16:
@tbresnan 
Could you please tell me how you found the size of the interleaver?