aff3ct/aff3ct #65: 'N_cw ' has to be greater than 0

kckjn97, 2020-05-01:
I ran LDPC with the following command and an error occurred.
N_cw was not delivered.

1.
./aff3ct -C "LDPC" -N 1024 -K 512 -M 3 -m 1
=>
(EE) In the '/Users/kckjn97/aff3ct/src/Module/Codec/Codec.hxx' file at line 44 ('Codec' function): "'N_cw' has to be greater than 0 ('N_cw' = 0)."
# End of the simulation.

2.
./aff3ct -C "LDPC" -N 1024 -K 512 -M 3 -m 1 -N_cw 1024
=>
(WW) Unknown argument "-N_cw" at the position 11.
(WW) Unknown argument "1024" at the position 12.
# The simulation is running...
(EE) In the '/Users/kckjn97/aff3ct/src/Module/Codec/Codec.hxx' file at line 44 ('Codec' function): "'N_cw' has to be greater than 0 ('N_cw' = 0)."
# End of the simulation.

When I ran same command with POLAR, it worked fine.

kckjn97, 2020-05-01:
$./aff3ct -C "LDPC" -N 1024 -K 512 -M 3 -m 1
```
# ----------------------------------------------------
# ---- A FAST FORWARD ERROR CORRECTION TOOLBOX >> ----
# ----------------------------------------------------
# Parameters:
# * Simulation ------------------------------------
#    ** Type                      = BFER
#    ** Type of bits              = int32
#    ** Type of reals             = float32
#    ** Date (UTC)                = 2020-05-01 10:36:38
#    ** Git version               = v2.3.5
#    ** Code type (C)             = LDPC
#    ** Noise range               = 1 -> 3 dB
#    ** Noise type (E)            = EBN0
#    ** Seed                      = 0
#    ** Statistics                = off
#    ** Debug mode                = off
#    ** Multi-threading (t)       = 8 thread(s)
#    ** Coset approach (c)        = no
#    ** Coded monitoring          = no
#    ** Bad frames tracking       = off
#    ** Bad frames replay         = off
#    ** Bit rate                  = 0.5 (1/2)
#    ** Inter frame level         = 1
# * Source ----------------------------------------
#    ** Type                      = AZCW
#    ** Implementation            = STD
#    ** Info. bits (K_info)       = 512
# * Codec -----------------------------------------
#    ** Type                      = LDPC
#    ** Info. bits (K)            = 512
#    ** Codeword size (N_cw)      = 0
#    ** Frame size (N)            = 1024
#    ** Code rate                 = 0.5 (1/2)
# * Encoder ---------------------------------------
#    ** Type                      = AZCW
#    ** Systematic                = yes
# * Decoder ---------------------------------------
#    ** Type (D)                  = BP_FLOODING
#    ** Implementation            = SPA
#    ** Systematic                = yes
#    ** Num. of iterations (i)    = 10
#    ** Stop criterion (syndrome) = on
#    ** Stop criterion depth      = 1
# * Modem -----------------------------------------
#    ** Type                      = BPSK
#    ** Implementation            = STD
#    ** Bits per symbol           = 1
#    ** Sigma square              = on
# * Channel ---------------------------------------
#    ** Type                      = AWGN
#    ** Implementation            = STD
#    ** Complex                   = off
#    ** Add users                 = off
# * Monitor ---------------------------------------
#    ** Lazy reduction            = off
#    ** Frame error count (e)     = 100
#    ** Compute mutual info       = no
# * Terminal --------------------------------------
#    ** Show Sigma                = off
#    ** Enabled                   = yes
#    ** Frequency (ms)            = 500
#
# The simulation is running...
(EE) In the '/Users/kckjn97/aff3ct/src/Module/Codec/Codec.hxx' file at line 44 ('Codec' function): "'N_cw' has to be greater than 0 ('N_cw' = 0)."
# End of the simulation.
```

ttonnellier, 2020-05-01:
You have to provide a parity check matrix:
https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/ldpc/decoder.html

Otherwise, you can use the DVB-S2 encoder, since matrices of this standard are included: https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/ldpc/encoder.html#enc-type