aff3ct/aff3ct #66: The chn file dumped by --sim-err-trk

zqcawzs, 2020-05-04:
Hi,

First, aff3c is awesome!

I have a question about the applied noise from the channel dumped by --sim-err-trk. I can't see any usable information except some garbled text. 

Thanks!

kouchy, 2020-05-05:
Hi @zqcawzs,

Thx for your interest in AFF3CT :-).

This is because the noise is dumped **in binary**. This is mandatory to have the exact 32-bit float number values.

The format of the binary file is the following:
  - **first value** is a **32-bit unsigned integer** representing the number of _frames_ contained in the file (_F_),
  - **second value** in a **32-bit signed integer** presenting the size of one _frame_ (_N_),
  - **the next values** are all **32-bit floating point numbers** representing the noise added to the modulation symbols. There is _F_ x _N_ 32-bit floats.

I hope it helps.

zqcawzs, 2020-05-05:
Thanks! That helps.

closed 2020-05-05 by zqcawzs