aff3ct/aff3ct #67: Question about the AWGN channel

zqcawzs, 2020-05-05:
Hi,

I want to check how many errors in one frame before decoding, so I remove the warning in the code to make aff3ct support zero decoding iteration. I use this to dump the error statistic when EbN0 is 3.90 for (1723, 2048) LDPC code：
`./aff3ct-2.3.3 --sim-type BFER -C LDPC  --dec-h-path conf/dec/LDPC/10GBPS-ETHERNET_1723_2048.alist --src-type AZCW --src-implem FAST -K 1723 -N 2048 -m 3.90 -M 3.90 -s 0.2 -n 10000 --dec-type BP_FLOODING --dec-implem OMS --dec-off 1 -i 0 -p 16 --qnt-bits 4 --qnt-dec 0 --mnt-err-hist 0 --mnt-err-hist-path ./hist_OMS --sim-crit-nostop  -c --sim-coded`
In order to verify that, I use the awgn function in Matlab to add noise in all 0 frame. The code is like:
```
ferror = fopen( 'error_num.txt','w' );
error_dis = zeros(100,1);
for i_ldpc_frames = 1 : 10000
    x = (zeros(1, 2048))';
    EbNoCoded = 3.9 + 10*log10(1723/2048);
    x_e = pskmod(x,2);
    rxSig = awgn(x_e,EbNoCoded,'measured');
    y = pskdemod(rxSig,2);
    error_num = sum(y ~= x);
    error_dis(error_num) = error_dis(error_num)+1;
end
for i = 1: 1 : 100
    fprintf(ferror, '%d; %d\n', i, error_dis(i));
end
fclose(ferror);
```

The result shows that Matlab will generate more errors than aff3ct. The attached pic shows the distribution of the errors.
[image]

So, what's the difference between the AWGN functions in aff3ct and Matlab?

Thanks!

kouchy, 2020-05-12:
I have to check that but it is hard to believe that the distribution of the noise is wrong in AFF3CT because we reproduced a lot of state-of-the-art BER/FER performances from various sources.

zqcawzs, 2020-05-13:
I agree with you. I am not sure that the way I get the error distribution from aff3ct is correct or not. I remove the warning in /aff3ct/src/Module/Decoder/LDPC/BP/Decoder_LDPC_BP.cpp,
```
if (n_ite <= 0)
        {
                std::stringstream message;
                message << "'n_ite' has to be greater than 0 ('n_ite' = " << n_ite << ").";
                throw tools::invalid_argument(__FILE__, __LINE__, __func__, message.str());
        }
```
and set the iteration number to 0. 
What do you think about this method to get the error distribution?

Thanks.

zqcawzs, 2020-05-24:
> I have to check that but it is hard to believe that the distribution of the noise is wrong in AFF3CT because we reproduced a lot of state-of-the-art BER/FER performances from various sources.

Hi, any ideas about this?

kouchy, 2020-05-26:
Hi @zqcawzs,

Perhaps you can check that sigma is the same than in MATLAB with the `--ter-sigma` parameter.
I'm also not entirely sure that performing 0 iteration is completely safe. If I were you I will replace the following code : https://github.com/aff3ct/aff3ct/blob/v2.3.3/src/Module/Decoder/LDPC/BP/Flooding/Decoder_LDPC_BP_flooding.hxx#L136-L158 by

    tools::hard_decide(Y_N, V_N, this->N);

Let me know if it changes something.

zqcawzs, 2020-05-27:
> Hi @zqcawzs,
>
> Perhaps you can check that sigma is the same than in MATLAB with the `--ter-sigma` parameter.
> I'm also not entirely sure that performing 0 iteration is completely safe. If I were you I [...]

Hi,

I used this command this time.
```
./aff3ct-2.3.3 --sim-type BFER -C UNCODED -D NONE --dec-implem HARD_DECISION --src-type AZCW --src-implem FAST -K 2048 -m 3.15 -M 3.15 -s 0.2  -p 16 --mnt-err-hist 0 --mnt-err-hist-path ./hist_HARD_10 --sim-crit-nostop -c  -n 1000000 -e 1000000 -E EBN0 --ter-sigma --sim-crit-nostop -c
```
The result seems pretty close to the matlab outputs.  So apparently, the 0 iteration is not a good way to get the error numbers.

Thanks for the help.

kouchy, 2020-05-27:
Ok good to ear! I also think this way is better.

What I don't understand is why do you use the `-c` or `--sim-coset` (https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/simulation/simulation.html?#sim-coset-c) parameter?

zqcawzs, 2020-05-27:
> Ok good to ear! I also think this way is better.
> 
> What I don't understand is why do you use the `-c` or `--sim-coset` (https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/simulation/simulation.html?#sim-coset-c) parameter?

I thought -c should be used if I choose the AZCW to generate information bits.. Will this cause some system mistakes?

kouchy, 2020-05-27:
It will not cause system mistakes. I think in your case it will change nothing to remove the coset. This parameter is useful if you are playing with AZCW and a modulation with a memory effect like the CPM for instance. In your case I think you want a basic BPSK with no memory effect, so you can remove the `-c` parameter I guess.

zqcawzs, 2020-05-27:
> It will not cause system mistakes. I think in your case it will change nothing to remove the coset. This parameter is useful if you are playing with AZCW and a modulation with a memory effect like [...]

All right, thanks.

closed 2020-05-27 by zqcawzs