aff3ct/aff3ct #68: Building for android and iOS

fmiceli24, 2020-05-17:
Hi,

I was wondering if there are any limitations to building aff3ct for Android or iOS.

Also any pointers as to how to do this would be great.

Thanks!

kouchy, 2020-05-18:
Hi @fmiceli24,

I don't think that there are any limitations to build AFF3CT for Android or iOS as there is no other dependencies than a C++11 compatible compiler. However we never did that, you may encounter some issues we did not anticipate... We are very interested by your feedback.

fmiceli24, 2020-05-18:
Thanks for answering.

Will post my results once I get time to do so.

Cheers!

elkhishen, 2021-03-02:
Hi @fmiceli24 , Do you have any updates regarding building aff3ct for android?

fmiceli24, 2021-03-03:
Hi. Have not proceeded with this since we found Siamese FEC simpler to
build under all of the platforms we support (Windows/OSX/Linux/Android/iOS).

If anyone gets it to work we would be super interested in testing it.