aff3ct/aff3ct #69: How to use libaff3ct-2.3.5.so with multithread when DVB-RCS2 Turbo_DB decoding

guokr321, 2020-06-11:
Hi,
Using libaff3ct-2.3.5.so to build a program under  Linux vbox 5.0.0-36-generic #39~18.04.1-Ubuntu SMP Tue Nov 12 11:09:50 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux:
    g++ -I. -I../include -I/usr/local/include/aff3ct-2.3.5/aff3ct -I/usr/local/include/aff3ct-2.3.5/MIPP -I/usr/local/include/aff3ct-2.3.5/rang  -g -DWall -D_REENTRANT -o rlecli rle_cli.o rle.o phy.o comm_log.o comm_sock.o comm_util.o crc.o comm_queue.o turbo_db.o -lpthread -lrt -lm -ldl -laff3ct.

When using multithread run the following code：
      turbo_modules.modem  ->modulate    (turbo_buffers.enc_bits,      turbo_buffers.symbols      );
      
      turbo_modules.channel->add_noise   (turbo_buffers.symbols,       turbo_buffers.noisy_symbols);

      turbo_modules.modem  ->demodulate  (turbo_buffers.noisy_symbols, turbo_buffers.LLRs         );
 
      turbo_modules.decoder->decode_siho (turbo_buffers.LLRs,          turbo_buffers.dec_bits     );
      
     // reset the monitor for the next SNR
      turbo_modules.monitor->reset();
      turbo_utils.terminal->reset();
after the program has run for several minutes, segment fault occurs, and core file can't be seen with gdb.
But using only one thread to run the above code, the result is OK. 

Thanks.

kouchy, 2020-06-12:
Hi @guokr321,

The `v2.3.5` of AFF3CT does not contain the tools to easily manage the multi-threading by yourself.
I'm not surprised that the code fails if multiple threads are running the same instance of a module.
A module like an encoder or a decoder is not made to be run by multiple threads. If you want to do that you will need to create as many instances as your multi-threading level.

However, in the development branch of AFF3CT (https://github.com/aff3ct/aff3ct/tree/development) we recently introduced new tools like the `sequence` and the `pipeline` to simplify the multi-threading management.
You can see some examples of code here:
  - sequence : https://github.com/aff3ct/my_project_with_aff3ct/blob/development/examples/sequence/src/main.cpp
  - pipeline : https://github.com/aff3ct/my_project_with_aff3ct/blob/development/examples/pipeline/src/main.cpp

The prerequisites to understand these codes is to know the AFF3CT notions of tasks and tasks binding. If you are not familiar with the AFF3CT tasks, I encourage you to read this documentation page: https://aff3ct.readthedocs.io/en/development/user/library/examples.html.

I hope it helps.

guokr321, 2020-06-15:
> Hi @guokr321,
>
> The `v2.3.5` of AFF3CT does not contain the tools to easily manage the multi-threading by yourself.
> I'm not surprised that the code fails if multiple threads are running the same [...]

Thanks. It helpls a lot.

closed 2020-10-17 by kouchy