aff3ct/aff3ct #74: Encode/Decode BCH codes

fender9669, 2020-07-10:
Hi,

I want to use aff3ct as a library and make a simple BCH encoder/decoder. I used bootstrap as a guideline but when I call  m.decoder -> decode_siho(b.LLRs, b.dec_bits) I get a segmentation fault. I am trying to use the Decoder_BCH_fast. I debugged the code with gdb and it seems that the problem is the "alpha_to" vector. When the constructor is called, "alpha_to" vector has the correct number of elements, but when it reaches the _decode method suddenly it has 35139647296722 elements.

Are you aware of this bug and do you know a fix for it?

Best regards,
Vlad

kouchy, 2020-07-13:
Hi,

It is hard to help you without your source code.
Can you fork the `my_project_with_aff3ct` and share your source code with us?

And also, this issue should be posted here: https://github.com/aff3ct/my_project_with_aff3ct/issues.

Best regards.

fender9669, 2020-07-13:
Hi,

I forked my_project_with_aff3ct here: https://github.com/fender9669/my_project_with_aff3ct/blob/master/examples/bootstrap/src/main.cpp

Best regards,
Vlad

kouchy, 2020-07-13:
I close this issue. I answered here: https://github.com/aff3ct/my_project_with_aff3ct/issues/3#issuecomment-657422992.

closed 2020-07-13 by kouchy