aff3ct/aff3ct #84: Polar SCL decoder overflow

SubtleMuffin, 2020-10-20:
I have put up a small end-to-end script using the AFF3CT lib, specifically using the systematic encoder and SCL decoder. It seems that the position where "1" should appear, overflowed with the max of 32-bit integer, i.e., -2147483648.

Here is the script --

```c++
#include "aff3ct.hpp"
#include "math.h"

#define old_api

using namespace aff3ct;

template <typename T> auto print_vector(std::vector<T> &vec) {
  std::for_each(vec.begin(), vec.end(),
                [&](const auto i) { std::cout << i << ' '; });
  std::cout << std::endl;
}

template <typename T> auto regulate(std::vector<T> &vec) {
  std::for_each(vec.begin(), vec.end(), [&](auto &i) { i = (i == 0 ? 0 : 1); });
}

auto generate_frozen_bits(const int k, const int n, const float snr_max) {
  // calculate constants
  auto r = static_cast<float>(k * 1.0 / n);
  const auto esn0 = tools::ebn0_to_esn0(snr_max, r);
  const auto sigma = tools::esn0_to_sigma(esn0);

  // set noise
  auto noise = std::unique_ptr<tools::Sigma<>>(new tools::Sigma<>());
#ifdef old_api
  tools::Frozenbits_generator_GA frozen_bits_generator(k, n);
  noise->set_noise(sigma, snr_max, esn0);
#else
  tools::Frozenbits_generator_GA_Arikan frozen_bits_generator(k, n);
  const auto ebn0 = tools::esn0_to_ebn0(esn0);
  noise->set_values(sigma, ebn0, esn0);
#endif

  // generate frozen bits
  std::vector<bool> frozen_bits(n);
  frozen_bits_generator.set_noise(*noise);
  frozen_bits_generator.generate(frozen_bits);
  return frozen_bits;
}

auto polar_encode_sys(const int k, const int n,
                      const std::vector<bool> &frozen_bits,
                      const std::vector<int> &info_bits) {
  // populate vector
  std::vector<int> encoded_bits(n);
  // encode
  module::Encoder_polar_sys<int> polar_encoder(k, n, frozen_bits);
  polar_encoder.encode(info_bits, encoded_bits);
  return encoded_bits;
}

auto polar_decode_sys_scl(const int k, const int n,
                          const std::vector<bool> &frozen_bits,
                          const std::vector<float> &received, const int l = 8) {
  // populate vector
  std::vector<int> decoded_bits(k);
  // decode
  module::Decoder_polar_SCL_MEM_fast_sys<int> polar_decoder(k, n, l,
                                                            frozen_bits);
  polar_decoder.decode_siho(received, decoded_bits);
  //   regulate(decoded_bits);
  return decoded_bits;
}

int main() {
  int k = 10;
  int n = 16;
  float snr_max = 6;

  // generate frozen bits
  auto frozen_bits = generate_frozen_bits(k, n, snr_max);

  // generate information bits
  auto source =
      std::unique_ptr<module::Source_random<>>(new module::Source_random<>(k));
  std::vector<int> info_bits(k);
  source->generate(info_bits);

  // polar encode
  auto encoded_bits = polar_encode_sys(k, n, frozen_bits, info_bits);

  // generate gaussian noise
  const float mean = 0.0;
  const float stddev = std::sqrt(std::pow(10, -snr_max / 10.0));
  std::normal_distribution<float> dist(mean, stddev);
  std::default_random_engine generator;

  // awgn channel
  std::vector<float> received;
  std::transform(
      encoded_bits.begin(), encoded_bits.end(), std::back_inserter(received),
      [&](const int i) -> float { return -i * 2 + 1 + dist(generator); });

  // polar decode
  auto decoded_bits = polar_decode_sys_scl(k, n, frozen_bits, received, 8);

  // print vectors
  std::cout << "--- info bits         ---" << std::endl;
  print_vector(info_bits);
  std::cout << "--- decoded bits      ---" << std::endl;
  print_vector(decoded_bits);

  // check for errors
  std::cout << "--- error positions   ---" << std::endl;
  for (auto i = 0; i < k; i++) {
    if (decoded_bits[i] != info_bits[i]) {
      std::cout << i << ' ';
    }
  }
}
```

And here are the outputs,
```
--- info bits         ---
0 1 1 0 1 1 1 1 1 1 
--- decoded bits      ---
0 -2147483648 -2147483648 0 -2147483648 -2147483648 -2147483648 -2147483648 -2147483648 -2147483648 
--- error positions   ---
1 2 4 5 6 7 8 9 ⏎
```

What could go wrong?

bonben, 2020-10-20:
Hi @SubtleMuffin, nothing actually goes wrong :wink: 

See https://github.com/aff3ct/my_project_with_aff3ct/issues/4#issuecomment-691913784.

SubtleMuffin, 2020-10-20:
Hi @bonben , thank you for your timely reply! Sorry.. I should have dig the issues deeper...

closed 2020-10-20 by SubtleMuffin