aff3ct/aff3ct #85: aff3ct fails to install as a submodule

jdemel, 2020-10-22:
Hi,

I try to build my own Python bindings for aff3ct with pybind11. The examples in [aff3ct example project](https://github.com/aff3ct/my_project_with_aff3ct) suggest to link statically against libaff3ct. Also, they suggest to not install aff3ct but just copy the generated cmake.

In order to make the installation of my bindings easier, I'd like to include aff3ct as a submodule. My workflow would look smth like this:
```
git checkout --recursive URL
mkdir build
cd build
cmake ..
make -j[some_value]
make install
```
The last step in this procedure causes issues.

```
CMake Error at lib/aff3ct/cmake_install.cmake:64 (file):
  file INSTALL cannot find
  "{source_prefix}/pybind11_aff3ct/build/lib/cmake/aff3ct-2.3.5/aff3ct-config-version.cmake":
  No such file or directory.
Call Stack (most recent call first):
  cmake_install.cmake:63 (include)
```
I assume this is the case because of these lines in CMake:
https://github.com/aff3ct/aff3ct/blob/1ceddfc874f38317e83592b506b1fe78dffe44b4/CMakeLists.txt#L556-L558

I left the aff3ct library in the subfolder `lib/aff3ct` like in the example project. aff3ct CMake expects a file in my build folder at:
`lib/cmake/aff3ct-2.3.5/aff3ct-config-version.cmake` but it ends up in `lib/aff3ct/lib/cmake/aff3ct-2.3.5/aff3ct-config-version.cmake`.

It would be sufficient to change `CMAKE_BINARY_DIR` to `CMAKE_CURRENT_BINARY_DIR` to fix this issue.

kouchy, 2020-10-22:
Hi @jdemel,

Yes you're right! This is a mistake in the `CMakeLists.txt`! I will accept the PR.

Just out of curiosity, did you plan to release your binding in `pybind11`?
I ask the question because you are not alone trying to wrap AFF3CT for Python.
@SubtleMuffin is working on a binding of the polar encoder and decoder with `Cython` (see PR #83).

jdemel, 2020-10-22:
I've seen #83. As far as I can tell, this is a different approach from what I do. I decided to go with pybind11 because it seems to integrate better with CMake. While the result is quite similar I assume these are different approaches.

I might release my pybind11 bindings. I stumbled over quite a few issues that I want to consider first.
It turned out that it is not as simple as expected to implement those bindings. At least I had hoped to bind the encoder and decoder classes directly. So far, all codes I used, polar, turbo, LDPC, required some wrapper classes.
If I'm not mistaken, aff3ct expects a `Codec` object to be created that holds all configuration and encoder and decoder objects. I want to wrap encoders and decoders directly. In that case encapsulation breaks quite frequently. e.g. classes only store references to parameters. I'd like to learn more about how aff3ct is supposed to be used before I release anything.

kouchy, 2020-10-22:
Ok I understand. To be honest I don't know which approach is the best. But I'm convinced that there is a huge interest using AFF3CT from Python. But if you're right and if it can be fully integrated with CMake, I think it is a good point.

In AFF3CT, the encoders and the decoders are regrouped in codecs yes. The codecs classes have been made to simplify the usage of encoders and decoders that work together. Your understanding of the codecs is correct. I understand that you only want to wrap the encoder and the decoder. Depending of the code, you will need to wrap more classes that just the encoder and the decoder. For instance, for polar codes you will need to manage the frozen bits, for LDPC codes you will need to manage G and H matrices and for the turbo code you will need to wrap the interleaver.

kouchy, 2020-10-23:
I close this issue as you resolved it. Do not hesitate to open other issues if you encounter new problems for the Python binding.

PR #86 (2020-10-22): cmake: Enable aff3ct include as submodule
If you want to include aff3ct as a submodule in your project, you might
run into issues during installation. This commit resolves this issue.

Fixes #85
commit 97f0a2f3 (2020-10-22): cmake: Enable aff3ct include as submodule

If you want to include aff3ct as a submodule in your project, you might
run into issues during installation. This commit resolves this issue.

Fixes #85
closed 2020-10-23 by kouchy